Dunedan/mbp-2016-linux #2: Can screen brightness be modified?

berglh, 2017-02-05:
This patch fixed the gmux control of brightness on the MBP 11,5. Maybe this patch is already upstream now? https://bugzilla.kernel.org/show_bug.cgi?id=105051

Dunedan, 2017-02-06:
I don't have a MacBook Pro with AMD graphics, so I can't test if that's still an issue. If you or somebody else can, I'd be happy to include it into the notes.

roadrunner2, 2017-02-20:
Thanks @berglh! That patch indeed fixed the issue on MacBookPro13,3. I've updated my instructions accordingly.

Dunedan, 2017-02-24:
Thanks @berglh. I updated my instructions as well.
Would be still quite good if somebody could put effort into getting that patch upstream. Volunteers?

Dunedan, 2017-07-13:
Closing this as it's properly documented in the README.

closed 2017-07-13 by Dunedan