Dunedan/mbp-2016-linux #4: iSight webcam driver possible fix

berglh, 2017-02-05:
Not sure if this is relevant. It may be possible to get this script to work with extracting the new firmware from OS X and creating a driver to suit: https://github.com/patjak/bcwc_pcie

Dunedan, 2017-02-06:
Thanks for that hint. I'm not sure if that's applicable for the MacBook Pros from 2016, as they access the iSight webcam through USB (via the iBridge) and not via PCIe, but I'll definitely check it out.

Dunedan, 2017-03-17:
So, got some new insights: `bcwc_pcie` is not relevant here, because as mentioned the webcam is connected via USB.

I got it to work with the `uvcvideo` kernel module from upstream kernel!

There is still a major caveat: I need to initialize the webcam first in another operating system (in my case a Windows VM), before it's working properly under Linux (same behavior which we used to initialize the Touch Bar, before we figured out how the initialization works).

Log without initialization:
```
[   10.992553] uvcvideo: Found UVC 1.50 device iBridge (05ac:8600)
[   10.995097] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[   10.995847] uvcvideo: Failed to query (129) UVC probe control : -32 (exp. 34).
[   10.995848] uvcvideo: Failed to initialize the device (-5).
[   10.995872] usbcore: registered new interface driver uvcvideo
[   10.995874] USB Video Class driver (1.1.1)
```

Log with initialization:
```
[ 1163.411581] uvcvideo: Found UVC 1.50 device iBridge (05ac:8600)
[ 1163.414648] uvcvideo 1-3:1.0: Entity type for entity Camera 1 was not initialized!
[ 1163.414688] usbcore: registered new interface driver uvcvideo
[ 1163.414689] USB Video Class driver (1.1.1)
```

mhkabir, 2017-03-17:
Woohoo!

Dunedan, 2017-03-18:
So, the webcam is working now. :tada: 

For now it's just a matter of setting the correct option for `uvcvideo`. For instructions see: https://github.com/Dunedan/mbp-2016-linux#facetime-hd-camera

I also already submitted a patch for the Linux kernel to get it working without manually setting the option in further kernel versions.

closed 2017-03-18 by Dunedan