Dunedan/mbp-2016-linux #6: Intel graphics on MBP 13,3

mhkabir, 2017-03-12:
This might be interesting : http://lukeluo.blogspot.in/2014/04/mac-book-pro-113-linux-customization-2.html

> Apple does not "expect" other OS running in rmbp.  Its EFI firmware will power down some hardware if it realize you are not booting into OS X.  At this time, the Intel Iris GPU won't show up.

Looking into how I can test this. I'm running refind, which directly hooks into the kernel stub loader, and thus doesn't use GRUB.

mhkabir, 2017-03-12:
The reason I pursue this is to achieve better battery life, since running the AMD graphics all the time is heavy on power consumption. 

Any other tips on improving battery life or pointers on power hogs which might be fixable would be appreciated!

mhkabir, 2017-03-12:
Moar resources : 
`. http://unix.stackexchange.com/questions/193425/enabling-intel-iris-pro-syslinux-tails-system-macbook-pro-15-retina-late-2013
2. https://marc.info/?l=grub-deavel&m=141586614924917&w=2
3. https://github.com/0xbb/gpu-switch

mhkabir, 2017-03-12:
Trying to boot from refind with `spoof_osx_version 10.12` causes boot to hang up at the place where it tries to load graphics drivers. So we are _potentially_ on the right track. 

@roadrunner2 you might want to have a look at this too, since you have a 13,3. Let me know if you have any ideas or are able to debug why it hangs.

mhkabir, 2017-03-13:
Intel graphics is now powered on : 
```
$ lspci | grep "VGA"
00:02.0 VGA compatible controller : Intel Corporation HD Graphics 530 (rev 06)
01:00.0 VGA compatible controller : Advanced Micro Devices, Inc. (AMD/ATI) Baffin [Radeon RX 460] (rev c7)
```
@roadrunner2 @Dunedan it still can't bring up Xorg, do you have any tips from how you forced the usage of `intel` as documented here : ? I already tried booting with `i915.modeset=0` but it doesn't help.

mhkabir, 2017-03-13:
Already have this : 
```
cat /etc/X11/xorg.conf.d/20-intel.conf 
Section "Device"
   Identifier  "Intel Graphics"
   Driver      "intel"
   BusID "PCI:0:2:0"
   Option      "TearFree"    "true"
   Option      "AccelMethod"  "glamor"
EndSection
```

mhkabir, 2017-03-13:
Xorg log while trying to run intel graphics : 
```
[    25.643] 
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    25.643] X Protocol Version 11, Revision 0
[    25.643] Build Operating System: Linux 4.4.0-45-generic x86_64 Ubuntu
[    25.643] Current Operating System: Linux Kabirs-MacBookPro 4.10.1-041001-generic #201702260735 SMP Sun Feb 26 12:36:48 UTC 2017 x86_64
[    25.643] Kernel command line: \vmlinuz-4.10.1-041001-generic cryptdevice=/dev/nvme0n1p5:vgcrypt:allow-discards root=/dev/mapper/vgcrypt-root rw single intremap=nosid initrd=\initramfs-4.10.1-041001-generic.img
[    25.643] Build Date: 02 November 2016  10:06:10PM
[    25.643] xorg-server 2:1.18.4-0ubuntu0.2 (For technical support please see http://www.ubuntu.com/support) 
[    25.643] Current version of pixman: 0.33.6
[    25.643] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    25.643] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    25.643] (==) Log file: "/var/log/Xorg.0.log", Time: Mon Mar 13 17:49:11 2017
[    25.648] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    25.648] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    25.649] (==) No Layout section.  Using the first Screen section.
[    25.649] (==) No screen section available. Using defaults.
[    25.649] (**) |-->Screen "Default Screen Section" (0)
[    25.649] (**) |   |-->Monitor "<default monitor>"
[    25.649] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    25.649] (**) |   |-->Device "Intel Graphics"
[    25.649] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    25.649] (==) Automatically adding devices
[    25.649] (==) Automatically enabling devices
[    25.649] (==) Automatically adding GPU devices
[    25.649] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    25.650] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    25.650] 	Entry deleted from font path.
[    25.650] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    25.650] 	Entry deleted from font path.
[    25.650] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    25.650] 	Entry deleted from font path.
[    25.650] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    25.650] 	Entry deleted from font path.
[    25.650] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    25.650] 	Entry deleted from font path.
[    25.650] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[    25.650] (==) ModulePath set to "/usr/lib/x86_64-linux-gnu/xorg/extra-modules,/usr/lib/xorg/extra-modules,/usr/lib/xorg/modules"
[    25.650] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    25.650] (II) Loader magic: 0x556b77a05dc0
[    25.650] (II) Module ABI versions:
[    25.650] 	X.Org ANSI C Emulation: 0.4
[    25.650] 	X.Org Video Driver: 20.0
[    25.650] 	X.Org XInput driver : 22.1
[    25.650] 	X.Org Server Extension : 9.0
[    25.650] (++) using VT number 7

[    25.650] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    25.651] (II) xfree86: Adding drm device (/dev/dri/card0)
[    25.653] (--) PCI: (0:0:2:0) 8086:191b:106b:015f rev 6, Mem @ 0x81000000/16777216, 0xa0000000/268435456, I/O @ 0x00005000/64
[    25.653] (--) PCI:*(0:1:0:0) 1002:67ef:106b:0166 rev 199, Mem @ 0xb0000000/268435456, 0xc0000000/2097152, 0x82600000/262144, I/O @ 0x00004000/256, BIOS @ 0x????????/131072
[    25.653] (II) LoadModule: "glx"
[    25.653] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    25.658] (II) Module glx: vendor="X.Org Foundation"
[    25.658] 	compiled for 1.18.4, module version = 1.0.0
[    25.658] 	ABI class: X.Org Server Extension, version 9.0
[    25.658] (==) AIGLX enabled
[    25.658] (II) LoadModule: "intel"
[    25.658] (II) Loading /usr/lib/xorg/modules/drivers/intel_drv.so
[    25.659] (II) Module intel: vendor="X.Org Foundation"
[    25.659] 	compiled for 1.18.4, module version = 2.99.917
[    25.659] 	Module class: X.Org Video Driver
[    25.659] 	ABI class: X.Org Video Driver, version 20.0
[    25.659] (II) intel: Driver for Intel(R) Integrated Graphics Chipsets:
	i810, i810-dc100, i810e, i815, i830M, 845G, 854, 852GM/855GM, 865G,
	915G, E7221 (i915), 915GM, 945G, 945GM, 945GME, Pineview GM,
	Pineview G, 965G, G35, 965Q, 946GZ, 965GM, 965GME/GLE, G33, Q35, Q33,
	GM45, 4 Series, G45/G43, Q45/Q43, G41, B43
[    25.660] (II) intel: Driver for Intel(R) HD Graphics: 2000-6000
[    25.660] (II) intel: Driver for Intel(R) Iris(TM) Graphics: 5100, 6100
[    25.660] (II) intel: Driver for Intel(R) Iris(TM) Pro Graphics: 5200, 6200, P6300
[    25.671] (II) intel(0): Using Kernel Mode Setting driver: i915, version 1.6.0 20161121
[    25.671] (II) intel(0): SNA compiled: xserver-xorg-video-intel 2:2.99.917+git20160325-1ubuntu1.2 (Timo Aaltonen <tjaalton@debian.org>)
[    25.671] (II) intel(0): SNA compiled for use with valgrind
```

mhkabir, 2017-03-13:
@Dunedan @roadrunner2 Aaaand I have it working! 

Instructions : 
1. Make sure the EFI powers on the IGP : 

> 
> By default the Intel GPU gets switched off by the MacBook Pro EFI if you boot anything but OS X. So to use the Intel GPU, you need to trick the EFI by using the "apple_set_os" hack either with:
> 
> rEFInd version 0.10.0 or above (recommended): http://www.rodsbooks.com/refind
> 
> Recent versions of rEFInd have the "apple_set_os" hack built-in. You can enable it by setting the spoof_osx_version option in your refind.conf.
> 
> or
> 
> apple_set_os.efi: https://github.com/0xbb/apple_set_os.efi
> or
> 
> a patched GRUB:
> https://lists.gnu.org/archive/html/grub-devel/2013-12/msg00442.html
> https://wiki.archlinux.org/index.php/MacBookPro11,x#Getting_the_integrated_intel_card_to_work_on_11.2C3
> or
> 
> a patched Kernel: https://www.marc.info/?l=grub-deavel&m=141586614924917&w=2
> Otherwise you will end up with a powered-down integrated graphics card and a black screen.

2. Create a Xorg config for the Intel GPU : 
```
$ cat /etc/X11/xorg.conf.d/20-intel.conf 
Section "Device"
	Identifier "Intel Graphics"
	Driver "intel"
	BusID "PCI:0:2:0"
	Option "TearFree" "true"
	Option "AccelMethod" "glamor"
EndSection
```
3. Clone gpu-switch and run it (from Linux) : 
```
git clone https://github.com/0xbb/gpu-switch
cd gpu-switch
./gpu-switch -i # needs root
```
4. Reboot system
```
sudo reboot
```
You should have a working Intel card now with proper HiDPI detection!

mhkabir, 2017-03-13:
I can't boot into macOS (boot hangs) after running `gpu-switch -i`, but it works again after running `gpu_switch -d`. Opened a refind discussion here https://sourceforge.net/p/refind/discussion/general/thread/84a55c48/ on whether we can do it from the bootloader automatically.

mhkabir, 2017-03-13:
The core issue here is that card switching isn't handled correctly by the kernel yet. We need proper `vgaswitcheroo` support for new Retina models to avoid the `gpu-switch` stuff and be able to dynamically switch between GPUs. 

https://www.phoronix.com/scan.php?page=news_item&px=Apple-GMUX-VGA-Switcher-4.6

Dunedan, 2017-03-13:
Congrats for getting it to work. As I've a MacBookPro 13,2 without a separate AMD GPU I won't be able to benefit from your findings. Anyway, as this aims to be a complete guide I've of course already included a link to your instructions into the README: https://github.com/Dunedan/mbp-2016-linux/#intel

> I can't boot into macOS (boot hangs) after running gpu-switch -i, but it works again after running gpu_switch -d.

As already mentioned on the refind-mailing list: All the `gpu-switch` utility does is to set a NVRAM variable, like I did for disabling the auto-boot feature (https://github.com/Dunedan/mbp-2016-linux/#disable-auto-boot). So you wouldn't necessary need it as a separate tool. That said: Can't you simply for now run it on shutdown with `gpu-switch -i`? That's still just a hacky workaround, but maybe better than nothing in the short term.

What I'd really like to see is the mentioned kernel- and/or grub-patch being applied upstream, as that would it make way more easier for the average user to use it.
@mhkabir Do you have in interest in pushing for that?

Dunedan, 2017-03-14:
I had another idea in the meanwhile:
The change `gpuswitch` does is a "non volatile" one, so it's permanent across reboots. What you could do instead is to make it a volatile one, so a reboot would cause it to be erased again.

You should be able to easily achieve that by changing the `\x07` to a `\x06` in the following line: https://github.com/0xbb/gpu-switch/blob/master/gpu-switch#L46

For reference from the [UEFI standard](http://www.uefi.org/sites/default/files/resources/UEFI%20Spec%202_6.pdf) chapter 7.2:
> //*******************************************************
> #define EFI_VARIABLE_NON_VOLATILE				0x00000001
> #define EFI_VARIABLE_BOOTSERVICE_ACCESS		0x00000002
> #define EFI_VARIABLE_RUNTIME_ACCESS			0x00000004
> #define EFI_VARIABLE_HARDWARE_ERROR_RECORD	0x00000008 

and:

>  If the `EFI_VARIABLE_NON_VOLATILE` attribute is not set, the firmware stores the variable in
normal memory and it is not maintained across a power cycle.

roadrunner2, 2017-04-13:
@mhkabir Excellent work! I played with this a bit, using the [apple_set_os.efi](https://github.com/0xbb/apple_set_os.efi) which works quite nicely. Turning off the dGPU saves quite a bit of power (dGPU only: 17.2W, dGPU+iGPU and using iGPU: 16.2W, iGPU only: 11.1W).

However, I did run into several issues. First, when both drivers are loaded, Wayland won't start, i.e. we fall back to plain old X; but this appears to be a Wayland/Mutter issue. I was able to fix this by adding `MUTTER_ALOW_HYBRID_GPUS=1` to `/etc/environment`. Second, if I power down the dGPU under X I often get long hangs as X apparently still tries to access the dGPU (despite the iGPU being the main one) - probably need to figure out a way to tell X to ignore the dGPU altogether. I also tried blacklisting the `amdgpu` module, but in this setup `vga_switcheroo` doesn't create the sysfs entries and hence the dGPU can't (easily) be turned off.

One big issue still remaining for me is that the console is garbled: booting with no `quiet` option I first see readable (though very small) output, but then it switches modes apparently after which the console output is completely garbled.

@Dunedan You can't change the flags - doing so just results in an EINVAL.

mhkabir, 2017-04-13:
I wasn't able to get power consumption down that low :( What are you using to measure power draw?

I also have this interrupt, gpe07, which keeps firing continuously. That was definitely not helping power consumption, so I disabled it till a proper fix can be found. Any ideas on what's gpe07? 

Regarding garbled text, it is fixed for me on kernel 4.11, but that kernel also  seems to break the keyboard driver startup. I need to enter my encryption keys before the OS can be loaded, and post kernel 4.11, the keyboard does not come up before the encryption prompt from lvm. If I use an e.g USB keyboard, I can get in, and the SPI driver comes up afterwards (might also have something to do with my rc.local configuration because it reloads applespi a couple of time to try and improve chances of touchpad in it. )

roadrunner2, 2017-04-13:
I measured the power consumption in several ways, but all of course with the power unplugged: `cat /sys/class/power_supply/BAT0/current_now` (together with `/voltage_now`), via `powertop`, and via the Gnome "Power Statistics" utility. I did this on a completely idle laptop with display at 50% brightness.

Regarding GPE 07, that appears to be from the embedded controller (see the `dsdt.dsl`) - I haven't noticed anything about that, but then again I wasn't looking - where do you see the fact that it's being fired a lot?

It's good to hear 4.11 will fix the garbled text - look forward to trying it. As for the loading of the applespi driver, I also need to enter the encryption key, and I originally had to figure out the list of kernel modules that are required so they could be added to the initrd (see the [keyboard.conf](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#file-keyboard-conf) on my gist). It may be that either an additional driver is needed on 4.11, or even that one was renamed - try looking at list of loaded modules for anything with `spi` or `lpss` in the name.

Dunedan, 2017-04-14:
> Regarding garbled text, it is fixed for me on kernel 4.11, but that kernel also seems to break the keyboard driver startup.

For me keyboard initialization during initrd works the same way with 4.11 as it did with 4.10, although I don't try to reload during that phase to get the touchpad working.

mhkabir, 2017-04-23:
@roadrunner2 I was checking interrupts in this manner : 
``` grep . -r /sys/firmware/acpi/interrupts/``` : 

```
kabir@Kabirs-MacBookPro:~$ grep . -r /sys/firmware/acpi/interrupts/
/sys/firmware/acpi/interrupts/ff_gbl_lock:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe15:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe43:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe71:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe05:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe33:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/gpe6D:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/gpe61:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe23:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe51:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe13:       0     STS invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe41:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe03:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe31:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/gpe6B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe21:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe78:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe11:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_pwr_btn:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/ff_slp_btn:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe68:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe01:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_pmtimer:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe58:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe48:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe76:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe38:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe66:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe28:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe56:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe18:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe46:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe74:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe08:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/sci:   33414
/sys/firmware/acpi/interrupts/gpe36:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe64:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe26:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/error:       0
/sys/firmware/acpi/interrupts/gpe54:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe16:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe44:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe72:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe06:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_rt_clk:       0         disabled     unmasked
[... 50 lines cut ...]
/sys/firmware/acpi/interrupts/gpe47:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1A:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe75:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe09:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe37:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0A:       0         invalid      unmasked
[... cut ...]
/sys/firmware/acpi/interrupts/gpe35:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6F:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe63:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe25:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe_all:   33419
/sys/firmware/acpi/interrupts/gpe53:       0         invalid      unmasked

```
Can you cross-test on yours and let me know your findings?

Dunedan, 2017-04-27:
What's the status of GPU-switching right now guys? Any new insights?

All help I can provide is that the MacBookPro13,3 apparently contains a Renesas R4F2113XLG microcontroller, which is referred as supported in https://github.com/torvalds/linux/blob/master/drivers/platform/x86/apple-gmux.c . What I can imagine causes it not to work is the following line:
https://github.com/torvalds/linux/blob/5924bbecd0267d87c24110cbe2041b5075173a25/include/linux/apple-gmux.h#L23

So `apple-gmux` only works when a ACPI device called `APP000B` is present. My MacBookPro13,2 obviously doesn't have that, as I have only IGP, but what I have are `APP000D` (keyboard & touchpad) and `APP0001` (?). Maybe you have another one which isn't `APP000B`. Or it's something completely different. :smile:

Anyway I think there is still a pretty good chance to get `vgaswitcheroo` to work without too much effort.

1005088h, 2017-12-31:
I am running Ubuntu with kernel 4.15-rc5 on MacBookPro14,3. Tried to use iGPU, however I gets flickering screen before login session, where I only can choose wayland or xorg. 
I followed the instructions:
1.
**mkdir /boot/efi/EFI/custom
cp apple_set_os.efi /boot/efi/EFI/custom**
add 
**search --no-floppy --set=root --label EFI
chainloader (${root})/EFI/custom/apple_set_os.efi
boot**
to **/etc/grub.d/40_custom**

2.
create /usr/share/X11/xorg.conf.d/20-intel.conf
**Section "Device"
	Identifier "Intel Graphics"
	Driver "intel"
	BusID "PCI:0:2:0"
	Option "TearFree" "true"
	Option "AccelMethod" "glamor"
EndSection**

3.
**git clone https://github.com/0xbb/gpu-switch
cd gpu-switch
./gpu-switch -i 
sudo reboot**

Result is the same when I tried compile apple_set_os.efi as well as download from git.
After I booted into high sierra then I can boot into Ubuntu again. But still the iGPU is not showing.
Did I miss anything?

TheoAndersen, 2018-03-23:
I can confirm that this works as well on a MacBookPro11,5. Used the Refind method, and did a ```blacklist radeon``` to blacklist the radeon integrated card.

arno01, 2018-05-08:
Thanks guys! I decided to write a small write-up if anyone would benefit, I'd be glad. :-)

# Steps to activate and use Intel HD Graphics on MacBookPro in Linux

> Tested on my ``MacBookPro14,3`` in Ubuntu 18.04 LTS (Bionic Beaver)

So it turns out that the Intel IGD (Integrated Graphics) must be unlocked from EFI (right in the Grub and right before selecting Ubuntu) and also activated via the efivars interface in advance.

Doing just one thing, won't help.

## 1. Build apple_set_os.efi

> I used docker to build the binary, you can follow the README there to use ``make`` command directly.

```
git clone https://github.com/0xbb/apple_set_os.efi.git
cd apple_set_os.efi/
docker build -t apple_set_os . && docker run --rm -it -v $(pwd):/build apple_set_os
```

## 2. Copy it to your EFI partition

```
sudo cp apple_set_os.efi /boot/efi/EFI/
```

## 3. Add a new menu entry to your Grub boot loader

Add the following to ``/etc/grub.d/40_custom`` file:

```
menuentry "Unlock Intel IGD" {
  search --no-floppy --set=root --file /EFI/apple_set_os.efi
  chainloader /EFI/apple_set_os.efi
  boot
}
```

Make sure your ``/etc/default/grub`` file contains these lines,
including the ``GRUB_HIDDEN_TIMEOUT`` must be commented:

```
GRUB_DEFAULT=0
# GRUB_HIDDEN_TIMEOUT=0
# GRUB_HIDDEN_TIMEOUT_QUIET=true
GRUB_TIMEOUT=10
GRUB_CMDLINE_LINUX="modprobe.blacklist=amdgpu acpi_backlight=intel_backlight"
```

> Notice that I am blacklisting the ``amdgpu`` here, otherwise it won't boot. Probably since ``i915`` cannot be loaded at the same time with the ``amdgpu``.
> I would recommend adding the custom entries to a ``/etc/grub.d/40_custom`` file rather than modifying the ``GRUB_CMDLINE_LINUX``.

Run ``sudo update-grub``.

## 4. Switch to the integrated GPU

> The ``gpu-switch`` is a shell script which basically flips the bit in ``/sys/firmware/efi/efivars/gpu-power-prefs-fa4ce28d-b62f-4c99-9cc3-6815686e30f9`` file.

```
$ git clone https://github.com/0xbb/gpu-switch
$ cd gpu-switch/
$ sudo ./gpu-switch -i
```

> To swtich back, just run ``sudo ./gpu-switch -d``.

## 5. Reboot your system

Now that you have activated integrated graphic card (Intel IGD), prepared the Grub boot loader, you can reboot.

IMPORTANT: before selecting ``Ubuntu`` at the Grub loader screen, first you need to select and load ``Unlock Intel IGD`` menu entry! Otherwise you will see a flickering screen and only forced Power Down will help.

Once you are in the system, you can see that you've got your Intel IGD activated:

```
$ lspci | grep "VGA"
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 460] (rev c0)

$ glxinfo | grep "OpenGL renderer"
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 

$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x49 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel
```

And since the HiDPI got properly recognized, so you can switch back to ``100%`` scale in the Displays options.

Dunedan, 2018-05-19:
@arno01 Thanks for the write up. :+1:  With this procedure, can you then switch between both GPUs at run time or are you limited to the iGPU?

arno01, 2018-05-19:
@Dunedan I haven't figured that yet. I think, with Nvidia that would have been easier than with AMD https://wiki.archlinux.org/index.php/PRIME#PRIME_GPU_offloading

myrgy, 2018-07-30:
Hi guys, I was fighting with nvidia on m6 macbook 11,3. To turn it off I wrote a small tool. https://github.com/myrgy/app-nv-xrun/blob/master/README.md
In general This tool allows to control apple muxer.

roadrunner2, 2018-08-26:
To expand on what @arno01 wrote, I have been using pretty the same setup with one notable difference:
> Notice that I am blacklisting the amdgpu here, otherwise it won't boot. Probably since i915 cannot be loaded at the same time with the amdgpu.

The problem with this approach is that the dGPU stays enabled and consumes a significant amount of power. And for me the system boots just fine with both drivers loaded (MBP13,3), so it doesn't look like there's a fundamental issue with that.

Instead I disable the dGPU after logging in with
```
echo OFF | sudo tee /sys/kernel/debug/vgaswitcheroo/switch
```
This gives me a nice 5 or 6W power savings (see also [my comment above](#issuecomment-293777940)).

The main drawback I've found with this is that shutdown will usually hang (the amdgpu driver appears to be unhappy that the dGPU is powered down, though powering it back up again first doesn't seem to help much either). Also, there are some issues trying to disable the dGPU before logging in, which I haven't been able to nail down so far.

mzogithub, 2018-11-15:
Hi, thanks for the overview. I used the modified Ubuntu 18.04 for MacBook (Linux 4.15.0-22) ISO provided at https://nixaid.com/linux-on-macbookpro/ to install Ubuntu on a MacBookPro 13,3, 15inches, touch bar, Radeon 460. I'm triple booting MacOS High Sierra, Ubuntu and Win 10.

After fixing some EFI problems such as Win10 not booting, now the three OS's boot fine (I use rEFInd). However, Ubuntu has the whole screen blanking by going to either orange (solid, no stripes or artifacts) or grey or black anywhere after a few seconds to a few minutes of normal use. This happens both when booting from live USB or from installed version (same behavior). The only way out is to hard-restart. This also happens during installation, in fact, I had to run the installation a few times until the screen lasted enough before hanging to complete the installation.

The blank screen kicks in at a random time (no patterns). I suspect this is related to the dual graphics cards or AMD drivers. I tried to disable screensaver and display power management but didn't fix the issue.

I thought this was related to "the black screen issue"  discussed at https://github.com/Dunedan/mbp-2016-linux/issues/73#issuecomment-422397681 but after updating the kernel to the latest version (tried 4.19.2 and 4.20-rc2) I reproduced the actual black screen issue  (black screen immediately after boot), and that looked a different problem as my screen hangs at least 30 seconds or up to 15 minutes after boot and goes either black, orange or grey. So with the latest kernels, I had a black screen at boot and when I went back to 4.15.0-22, the problem described above returned.

Additionally, I also had this same problem when I booted from a non-modified live USB ISO  downladed directly from Ubuntu website.

 Any help would be much appreciated. Thanks.

renatolfc, 2020-01-20:
@roadrunner2 for the past couple of days, I've been testing the following approach:
 1- Blacklist the `amdgpu` module, as suggested by @arno01 
 2- Manually loading the `amdgpu` module so that I can turn the GPU off prior to starting X
 I've written the following systemd unit, which seems to be working fine so far:

```
cat /etc/systemd/system/macbook-quirks.service                                                                   
[Unit]
Description=Perform some tweaks to allow sleeping and better power management
Before=display-manager.service

[Service]
Type=oneshot
ExecStart=/bin/modprobe amdgpu
ExecStart=/bin/sh -c "echo OFF > /sys/kernel/debug/vgaswitcheroo/switch"
RemainAfterExit=yes
TimeoutSec=0

[Install]
WantedBy=multi-user.target
```

Turning the DGPU back on will cause an X restart, but apart from that, it seems to be working fine.

I'm still missing backlight control, though. Right now, I only have two levels: off, and max brightness. The same is true for the keyboard backlight.

roadrunner2, 2020-01-22:
@renatolfc Interesting: this allows you to disable the dGPU before logging in? I'll play around with blacklisting amdgpu and later loading it explictly then.

As far as your backlight issues go, that's really weird - both backlights have a full range for me. What do your `max_brightness` levels look like?
```
grep . /sys/class/backlight/*/max_brightness /sys/class/leds/spi::kbd_backlight/max_brightness
```
Assuming they're larger than 1, what happens when you write values manually to the `brightness` attributes? E.g.
```
echo 250 | sudo tee /sys/class/backlight/gmux_backlight/brightness
```

renatolfc, 2020-01-22:
@roadrunner2, that's correct. Right after logging into X, these are the contents of `/sys/kernel/debug/vgaswitcheroo/switch`:

```
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :Off:0000:01:00.0
```

The key in the unit file above is the entry `Before=display-manager.service`, which makes sure this gets executed before the display manager is loaded (in my case, XDM). In case it helps, `dmesg | grep amdgpu` gives this:

```
[   13.301882] [drm] amdgpu kernel modesetting enabled.
[   13.302056] amdgpu 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 0: 0xb0000000 -> 0xbfffffff
[   13.302058] amdgpu 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 2: 0xc0000000 -> 0xc01fffff
[   13.302059] amdgpu 0000:01:00.0: remove_conflicting_pci_framebuffers: bar 5: 0x82600000 -> 0x8263ffff
[   13.303503] amdgpu 0000:01:00.0: BAR 2: releasing [mem 0xc0000000-0xc01fffff 64bit pref]
[   13.303506] amdgpu 0000:01:00.0: BAR 0: releasing [mem 0xb0000000-0xbfffffff 64bit pref]
[   13.303514] amdgpu 0000:01:00.0: BAR 0: assigned [mem 0xb0000000-0xbfffffff 64bit pref]
[   13.303522] amdgpu 0000:01:00.0: BAR 2: assigned [mem 0xc0000000-0xc01fffff 64bit pref]
[   13.303539] amdgpu 0000:01:00.0: VRAM: 4096M 0x0000000000000000 - 0x00000000FFFFFFFF (4096M used)
[   13.303540] amdgpu 0000:01:00.0: GART: 256M 0x000000FF00000000 - 0x000000FF0FFFFFFF
[   13.319266] [drm] amdgpu: 4096M of VRAM memory ready
[   13.319270] [drm] amdgpu: 4096M of GTT memory ready.
[   13.369557] amdgpu: [powerplay] hwmgr_sw_init smu backed is polaris10_smu
[   14.519599] [drm:retrieve_link_cap [amdgpu]] *ERROR* retrieve_link_cap: Read dpcd data failed.
[   14.536106] [drm:dc_link_detect [amdgpu]] *ERROR* No EDID read.
[   14.700454] amdgpu 0000:01:00.0: fb1: amdgpudrmfb frame buffer device
[   14.704175] [drm] Initialized amdgpu 3.35.0 20150101 for 0000:01:00.0 on minor 1
[   14.712083] amdgpu: switched off
```

So, the GPU module is loaded and the GPU is switched off just fine.

With regards to the backlight, it appears writing directly to `/sys/class/backlight/gmux_backlight/brightness` seems to do the trick. `light`, `xbacklight`, and the xfce power manager can't seem to manage that, though. Thanks for that!

cristianmiranda, 2020-04-25:
Hi guys! Thank you for the amazing job documenting everything you've tried, very helpful! :smile: 
I was able to get Intel integrated GPU powered on, blacklist `amdgpu` and run `gpu-switch -i`

The problem is that dGPU (ATI Radeon) is still powered and consuming A LOT of resources. I've seen you mentioned that the card should be loaded and powered off. So I did: `sudo modprobe amdgpu`, and for turning it off (here comes the tricky part, I don't have vgaswitcheroo) I tried to power off the PCI device, like so:

```bash
➜ lspci| grep "VGA"                                       
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X] (rev c0)

# Notice how 01:00.0 is the ID for the PCI device

➜ sudo cat /sys/bus/pci/devices/0000:01:00.0/power/control 
on

sudo su
➜ echo OFF > /sys/bus/pci/devices/0000:01:00.0/power/control
echo: write error: invalid argument
```
I get the same error using 'OFF','1','0' (although for turning off it should be 1)

---
### Questions:
After manually loading amdgpu, resources consumption started to decrease, also the laptop temperature went down as well. Is this even possible or it's just me?. I'm not sure how the driver manages the eGPU, but now it seems to be in an idle state. Maybe?
 
Just to make sure that Intel is the card providing graphics, I run:
```bash
➜ glxinfo | grep "OpenGL renderer"          
OpenGL renderer string: Mesa Intel(R) HD Graphics 530 (SKL GT2)
```
What's the correct way of turning the eGPU off in my case?

Here's the output of `gpu-manager` (if it's of any use):
```bash
➜ sudo gpu-manager                                     
[sudo] password for cmiranda: 
last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /run/u-d-c-nvidia-was-loaded file
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/5.6.0-1007-oem/updates/dkms
Looking for amdgpu modules in /lib/modules/5.6.0-1007-oem/updates/dkms
Is nvidia loaded? no
Was nvidia unloaded? no
Is nvidia blacklisted? no
Is intel loaded? yes
Is radeon loaded? no
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? yes
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? no
Is nvidia kernel module available? no
Is amdgpu kernel module available? no
Vendor/Device Id: 8086:191b
BusID "PCI:0@0:2:0"
Is boot vga? yes
Vendor/Device Id: 1002:67ef
BusID "PCI:1@0:0:0"
Is boot vga? no
Found "/dev/dri/card1", driven by "amdgpu"
output 0:
	card1-eDP-2
Number of connected outputs for /dev/dri/card1: 1
Skipping "/dev/dri/card1", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "i915"
Skipping "/dev/dri/card1", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "i915"
Skipping "/dev/dri/card1", driven by "amdgpu"
Found "/dev/dri/card0", driven by "i915"
output 0:
	card0-eDP-1
Number of connected outputs for /dev/dri/card0: 1
Does it require offloading? yes
last cards number = 2
Has amd? yes
Has intel? yes
Has nvidia? no
How many cards? 2
Has the system changed? No
Intel IGP detected
Desktop system detected
or laptop with open drivers
Nothing to do

~ took 2s 
➜ sudo cat /var/lib/ubuntu-drivers-common/last_gfx_boot
8086:191b;0000:00:02:0;1
1002:67ef;0000:01:00:0;0
```

Thanks! :smiley:

roadrunner2, 2020-04-29:
> The problem is that dGPU (ATI Radeon) is still powered and consuming A LOT of resources. I've seen you mentioned that the card should be loaded and powered off. So I did: `sudo modprobe amdgpu`, and [...]
[snip]

This won't work - the GPU needs to be properly powered down by the driver. Why don't you have vgaswitcheroo? Because that's what you need, and without it you can't power down the dGPU properly. If you don't have vgaswitcheroo because you're running kernel 5.6, then downgrade to kernel 5.5.

> After manually loading amdgpu, resources consumption started to decrease, also the laptop temperature went down as well. Is this even possible or it's just me?. I'm not sure how the driver manages the eGPU, but now it seems to be in an idle state. Maybe?

Right: when powered on, the GPU is completely on with all clocks at max; when initializing, the driver with then lower the clocks and put various parts of the GPU into power-saving mode (very roughly).

cristianmiranda, 2020-04-29:
> This won't work - the GPU needs to be properly powered down by the driver. Why don't you have vgaswitcheroo? Because that's what you need, and without it you can't power down the dGPU properly. If you don't have vgaswitcheroo because you're running kernel 5.6, then downgrade to kernel 5.5.

Hey @roadrunner2, thanks for stepping by!
Yes, I'm on 5.6, but I'm not 100% sure if 5.5 has vgaswitcheroo, I have another machine with 5.4 and cannot see it:
```bash
mini-linux# cd /sys/kernel/debug/             
mini-linux# ls
acpi		  dma_buf	      iosf_sb		pm_qos		  swiotlb
aufs		  dri		      kprobes		pwm		  sync
bdi		  dynamic_debug       kvm		ras		  tracing
block		  error_injection     mce		regmap		  usb
bluetooth	  extfrag	      mei0		regulator	  virtio-ports
cec		  fault_around_bytes  memcg_slabinfo	remoteproc	  wakeup_sources
cleancache	  frontswap	      mmc0		sched_debug	  x86
clear_warn_once   gpio		      opp		sched_features	  zswap
clk		  hid		      pinctrl		sleep_time
device_component  ieee80211	      pkg_temp_thermal	split_huge_pages
devices_deferred  intel_powerclamp    pm_genpd		suspend_stats

mini-linux# uname -r
5.4.0-26-generic
```
I'll do some research on how to bring it back though. If anyone knows a way, please let me know.

roadrunner2, 2020-04-29:
Does the other machine have multiple (powered on) GPU's? vgaswitcheroo only appears once a second GPU is registered.

cristianmiranda, 2020-04-29:
Ah! That must be it then, it only has an integrated GPU. Good call.
Ok, then I'll rollback to 5.5 and see what happens. Is this something that we'll never be able to do on 5.6+ ?

roadrunner2, 2020-05-10:
@cristianmiranda Regarding `amdgpu` getting fixed on 5.6, you'll want to follow #142 and related kernel bug reports and discussions.

cristianmiranda, 2020-05-11:
Thank you, @roadrunner2 ! btw, I created a **guide** here: https://dev.to/cmiranda/linux-on-macbook-pro-2016-1onb summarizing all the steps I did in order to get this work. I'd love some feedback.
Thanks again for being so responsive on this subject! :heart:

andyholst, 2020-09-10:
I have checked out v5.5 tag release and I have applied the patch from https://marc.info/?l=grub-devel&m=141586614924917&q=p3 and I had to resolve some merge conflicts, you can see the commit at my linux-stable-fork patch branch https://github.com/andyholst/linux-stable-fork/tree/efi-Identify-as-OS-X-to-EFI-drivers-before-booting

The patch for v5.5 works for me, I can access the /sys/kernel/debug/vgaswitcheroo/switch file and it gives me the following

> 0:DIS:+:Pwr:0000:01:00.0
1:IGD: :Pwr:0000:00:02.0

The command `sudo modprobe amdgpu`works just fine without any freezes.

The command `lspci | grep "VGA"`gives me the list:

> 00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X] (rev c0)

The command `gpu-manager` verifies that the intel card and amd card is loaded with following list:

> last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /run/u-d-c-nvidia-was-loaded file
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/5.5.0+/updates/dkms
Looking for amdgpu modules in /lib/modules/5.5.0+/updates/dkms
Is nvidia loaded? no
Was nvidia unloaded? no
Is nvidia blacklisted? no
Is intel loaded? yes
Is radeon loaded? no
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? no
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? no
Is nvidia kernel module available? no
Is amdgpu kernel module available? no
Vendor/Device Id: 8086:191b
BusID "PCI:0@0:2:0"
Is boot vga? no
Vendor/Device Id: 1002:67ef
BusID "PCI:1@0:0:0"
Is boot vga? yes
Skipping "/dev/dri/card1", driven by "i915"
Found "/dev/dri/card0", driven by "amdgpu"
output 0:
        card0-eDP-1
output 1:
        card0-DP-1
Number of connected outputs for /dev/dri/card0: 2
Skipping "/dev/dri/card1", driven by "i915"
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card1", driven by "i915"
Skipping "/dev/dri/card0", driven by "amdgpu"
Found "/dev/dri/card1", driven by "i915"
Number of connected outputs for /dev/dri/card1: 0
Does it require offloading? no
last cards number = 2
Has amd? yes
Has intel? yes
Has nvidia? no
How many cards? 2
Has the system changed? No
Unsupported discrete card vendor: 8086
Nothing to do

So if you study the new efi boot structure for v5.7+ you should be able to apply the patch for it as well or even make a serious patch for upstream if you are up to it.

andyholst, 2020-09-12:
I have applied the patch to the new refactored efi boot structure with kernel version 5.7 (newer) at https://github.com/andyholst/linux-stable-fork/tree/efi-Identify-as-OS-X-to-EFI-drivers-before-booting-v5.8, if it is going to be included in the upstream it needs to be better then a hack. I don't know the details of the qualifications for a patch to be approved as Linux kernel patch.

marc-git, 2020-12-11:
On 14,3 X won't start when the intel card is enabled. Should I start a new issue or does that fit here?

divad42, 2020-12-11:
X starts for me on my 14,3, but I have to modprobe -r amdgpu first

marc-git, 2020-12-14:
funny when I blacklisted the module it failed to start

divad42, 2020-12-14:
I assume you're using rEFInd to spoof OSX and you've used the gpu-switch tool?

fannullone, 2020-12-23:
I am running FC33 kernel 5.9.15 on macbook pro 13.3 with touchbar.
I have been running for some time on dGPU but I noticed that after heavy use, ie watching video, the retina starts flickering very bad.
Is this caused by over-heating?
If I reboot it works for a while but it may happen again.
Anyone experiencing the same?

fannullone, 2020-12-23:
Following up to my previous post, I moved to iGPU by blacklist the amdgpu module:
[root@mbp ~]# grub2-editenv list | grep kernelopts
kernelopts=root=UUID=**** ro rd.luks.uuid=luks-**** rd.luks.uuid=luks-**** rhgb quiet modprobe.blacklist=amdgpu acpi_backlight=intel_backlight
and then gpu-switch -i
So far so good, no more flickering but need to stress it more before claiming success.
On the other hand, I see an idle consumption of ~16Watts which confirms that dGPU is still powered.
I don't have the option below, even after loading the amdgpu module manually:
[root@mbp ~]# ll  /sys/kernel/debug/vgaswitcheroo/switch
ls: cannot access '/sys/kernel/debug/vgaswitcheroo/switch': No such file or directory
And also this one doesn't work:
[root@mbp ~]# cat /sys/bus/pci/devices/0000:01:00.1/power/control
auto
[root@mbp ~]# echo 'off' > '/sys/bus/pci/devices/0000:01:00.1/power/control'
-bash: echo: write error: Invalid argument
[root@mbp ~]# lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X] (rev ef)

So any update how to switch off the dGPU and save some power?

LanderN, 2021-01-16:
@fannullone, did you find a solution yet? I'm in the same boat: mbp 13,3, on Intel graphics, blacklisted amdgpu, intel gpu loading and working fine, but no vgaswitcheroo "folder", not even after loading amgpu manually

fannullone, 2021-01-17:
@LanderN, unfortunately no progress on this issue

palmerj, 2021-08-23:
Heya. Thanks for this interesting thread. I'm running a Macbook pro 15" mid 2014 with Ubuntu and want to get the intel card working. I followed @arno01 instructions and the OS boots and the Ubuntu process loading screen starts, but just as gdm3 starts it's locks up. 

Via Ubuntu recovery terminal:

`lspci | grep "VGA"`gives:

00:02.0 VGA compatible controller: Intel Corporation Crystal Well Intergrated Graphics Controller (rev 08)
01:00.0 VGA compatible controller: NVIDIA Corporation GK107M [GeForce GT 750M Mac Edition] (rev a1)

gpu-manager gives:

[image]

Note I tried blacklist the nvidia GPU in grub that didn't make any difference. 

I've got the nvidia-470 drivers installed.