Dunedan/mbp-2016-linux #7: How to set the boot parameter

iamwrm, 2017-03-21:
In readme, it says

> Booting
> To boot Linux properly, it's necessary to set intremap=nosid as kernel boot parameter.

How can I set the parameter?
Intall a linux on a USB and run it on another computer, then use terminal to set the parameter?

Dunedan, 2017-03-21:
Setting a kernel parameter is documented in various places around the internet. E.g. https://wiki.ubuntu.com/Kernel/KernelBootParameters or https://wiki.archlinux.org/index.php/kernel_parameters

If you're already having trouble setting such a parameter you might not enjoy running Linux on a MacBook Pro 2016 yet, as there are still some non-trivial manual tweaks required (e.g. to get the internal keyboard and touchpad working) and some hardware components aren't working at all yet (most prominently wifi)! That situation will hopefully be different in future, but for now a lot of fiddling around is required to get a working system.

iamwrm, 2017-03-22:
Thanks for replying
I managed to set the parameter by using another laptop (fedora 25)
The start up menu which let you select linux or rescue system appears (which won't appear if the parameter is not changed), but when I select to enter the fedora, the screen blurred and froze....

Yeah I think for now I'll just play around with linux in the VM and wait for some mature solution or distribution update.

closed 2017-03-22 by iamwrm