Dunedan/mbp-2016-linux #10: Wifi on MacBookPro13,1

christophgysin, 2017-04-17:
There seems to be a different wifi card on the macbook13,1:
```
$ lspci -s 02:00.0 -vnn
02:00.0 Network controller [0280]: Broadcom Limited BCM4350 802.11ac Wireless Network Adapter [14e4:43a3] (rev 05)
        Subsystem: Apple Inc. Device [106b:0159]
        Flags: fast devsel, IRQ 16
        Memory at 92400000 (64-bit, non-prefetchable) [size=32K]
        Memory at 92000000 (64-bit, non-prefetchable) [size=4M]
        Capabilities: <access denied>
        Kernel modules: brcmfmac
```
The device does not seem to be recognized at all:
```
$ dmesg | grep brcmfmac:
[    3.774619] brcmfmac: brcmf_chip_recognition: chip backplane type 15 is not supported
[    3.774685] brcmfmac: brcmf_pcie_probe: failed 14e4:43a3
```
Is there any known workaround for this?

mhkabir, 2017-04-17:
might help : https://askubuntu.com/questions/739048/wireless-adapter-broadcom-14e443a3-does-not-work-in-ubuntu

christophgysin, 2017-04-17:
@mhkabir According to those answers, it should work since kernel 4.4. I'm running 4.10.10 and it is not recognised. Am I missing something?

tudorbarascu, 2017-04-18:
@christophgysin On Debian stretch (kernel 4.9) I had to install the firmware-brcm80211 (from non-free repo) package to make it work.
I tested with kernel 4.11-rc7 and it also works.

christophgysin, 2017-04-18:
@tudorbarascu Thanks! Indeed, I was just missing the firmware (on ArchLinux: linux-firmware).

closed 2017-04-18 by christophgysin

Dunedan, 2017-04-18:
@christophgysin Does it work fine for you even with 5Ghz networks?

christophgysin, 2017-04-18:
@Dunedan Haven't done any extensive testing, but yes, I connected to a 5GHz network and updated my system.

Dunedan, 2017-04-18:
Great to here. So the wifi problems seem to be limited to the MacBookPro13,2 and MacBookPro13,3. I'll update the README later to reflect that.