Dunedan/mbp-2016-linux #11: KMS on MBP 13,1

christophgysin, 2017-04-18:
I'm struggling to get KMS to work on a MBP 13,1.

It boots fine with `nomodeset`, but without it the screen becomes scrambled. Since there is no dedicated graphics card, I assume the workaround in #6 isn't necessary.

Did anyone get this to work? Or is this a known issue?

Dunedan, 2017-04-18:
The scrambled screen is an issue in Linux <4.11. So if you switch to Linux 4.11 it should work fine. Even for versions before the screen should be only scrambled during boot and should be fine again once Xorg took over.

christophgysin, 2017-04-19:
With 4.10.10, Xorg didn't work either. But I can confirm this is fixed in 4.11.0-rc7-g4f7d029b9bf0, for both the console and Xorg.

closed 2017-04-19 by christophgysin