Dunedan/mbp-2016-linux #13: macfanctld

tudorbarascu, 2017-04-20:
As my 13,1 macbook was getting reallyhot when pushed to the limits and I could hear the fan wasn't revving up like it did under MacOs I installed `macfanctld` and it's working great.

Should I make a pull request to document this?

Dunedan, 2017-04-20:
Fans are working for me properly and I had nothing to do to get them working that way.
So your fans were working before but weren't spinning fast enough? Or did they not work at all?

tudorbarascu, 2017-04-21:
They worked but they weren't spinning fast enough. I was doing a kernel compile with all the cores involved and the bottom became really hot (hotter than on mac os x). As I remembered the solution from my old macbook pro, I checked if it worked and as soon as I installed it the fans revved up and cooled it down to a normal (very warm level). If I have the time I'll put out some numbers next time.
Basically, in my mind the macbook shouldn't get hotter than it does on macos.

Dunedan, 2017-04-21:
Let's collect some information about temperatures and fan speeds to see what's going on. I just added a cron job to log them each minute, by using the output of `sensors`. Let's see what we get out of it.

tudorbarascu, 2017-04-27:
From what I could tell, macfanctl is far quicker in revving up the fans.
Without it, you end up at the same fan speeds albeit much slower.

Dunedan, 2017-04-27:
I also haven't seen real issues with thermal management so far. I might have had throttled CPU speeds a few times during CPU-intensive tasks like kernel compilation, but nothing worrying. The MBP also doesn't feel warmer than other Macs I had.

So do we already have a conclusion? Like: The fans cool down the MBP properly while being hardware-controlled, but software-controlled (under macOS or with macfanctld) they run faster and keep the MBP cooler, decreasing thermal stress on the material and decrease the chance of CPU-throttling due to exhausting its thermal budget?

tudorbarascu, 2017-04-27:
I want to see if there are any differences between fan speed/sensors under macos and linux under full load. As I don't have the time now I will get to it next week probably.

Dunedan, 2017-07-11:
I played around with `macfanctld` recently a bit and used it for a while, but decided to stay with the built-in fan control, as I saw no advantage in `macfanctld`. It's certainly an option if one wants to keep his MacBook Pro cooler than the integrated fan control does, but the disadvantages are of course more noise and more power drawn by the fans.

@tudorbarascu Any more insights you can add?

tudorbarascu, 2017-07-12:
@Dunedan sorry for the late reply. Nothing more to add. You summarized it perfectly.
Thanks. Closing down the issue.

closed 2017-07-12 by tudorbarascu