Dunedan/mbp-2016-linux #15: Webcam of MacBookPro13,1

Dunedan, 2017-04-22:
Can one of you MacBookPro13,1 owners please try out, if the webcam works with the [bcwc_pcie](https://github.com/patjak/bcwc_pcie/wiki) driver? Should be a low hanging fruit to get some clarity on.

christophgysin, 2017-04-22:
After a quick try, it doesn't seem to work:
```
[  556.756303] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[  556.756531] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[  556.761888] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[  556.761916] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[  556.771943] facetimehd 0000:03:00.0: PLL reset finished
[  556.771944] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[  556.771957] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[  556.781993] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[  556.802075] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[  556.802089] facetimehd 0000:03:00.0: STRAP valid
[  556.802090] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[  556.802097] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[  556.802106] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[  556.802117] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[  556.802118] facetimehd 0000:03:00.0: Using step size 152
[  556.802122] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  556.802127] facetimehd 0000:03:00.0: Virtual VTT enabled
[  556.822691] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[  556.822705] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[  556.822907] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[  557.261252] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[  567.426281] facetimehd 0000:03:00.0: Init failed! No wake signal
[  567.426411] facetimehd: probe of 0000:03:00.0 failed with error -5
```

christophgysin, 2017-04-22:
Second try, after a reboot:
```
[  183.855911] media: Linux media interface: v0.10
[  183.860886] Linux video capture interface: v2.00
[  183.865998] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[  183.866271] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[  183.871966] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[  183.872082] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[  183.882223] facetimehd 0000:03:00.0: PLL reset finished
[  183.882279] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[  184.137209] facetimehd 0000:03:00.0: Failed to lock S2 PLL: 0xc902c902
[  184.157351] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[  184.157433] facetimehd 0000:03:00.0: STRAP valid
[  184.157495] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[  184.157566] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[  184.157640] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[  184.157725] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[  184.157787] facetimehd 0000:03:00.0: Using step size 150
[  184.157850] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  184.157985] facetimehd 0000:03:00.0: Virtual VTT enabled
[  184.180119] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[  184.181669] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[  184.183419] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[  184.624392] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[  184.781601] facetimehd 0000:03:00.0: ISP woke up after 0ms
[  184.783793] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[  184.785954] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[  184.797927] facetimehd 0000:03:00.0: ISP second int after 0ms
[  184.799404] facetimehd 0000:03:00.0: Channel description table at 00800000
[  184.811027] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[  184.812622] facetimehd 0000:03:00.0: Enabling interrupts
[  184.814320] FWMSG:
[  184.815918] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  184.817402] FWMSG:
[  184.818983] FWMSG:   Restart count : 0
[  184.820526] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  184.822141] FWMSG:   System Clock : 200000000 Hz
[  184.823704] FWMSG:   Processor mode : SUPERVISOR
[  184.825323] FWMSG:   Cache architecture type : SEPARATE
[  184.826889] FWMSG:   Cache type : WRITEBACK
[  184.828506] FWMSG:   Data Cache Line : 32 [0x20]
[  184.830085] FWMSG:   Boot arguments entries : 2
[  184.831706] FWMSG:     0000: 0x00000000 0x00000000
[  184.833276] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  184.834861] FWMSG:   Main memory :   base : 0x00000000
[  184.836474] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  184.838128] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  184.839783] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  184.841410] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  184.843088] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  184.844785] FWMSG:   Shared memory : base : 0x00800000
[  184.846465] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  184.848214] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  184.849973] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  184.851734] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  184.853542] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  184.855325] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  184.857148] FWMSG:   Application : adc [release]
[  184.859005] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  184.860807] FWMSG:   Release : S2ISP-01.43.00
[  184.862590] FWMSG:   H4ISPAPPLE : 11536
[  184.864370] FWMSG:   H4ISPCD : 4081
[  184.866114] FWMSG:   ffw : 4143
[  184.867846] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  184.869637] FWMSG:
[  184.871375] FWMSG: -------------------------------------------------------------------------------
[  184.873209] FWMSG:
[  184.977596] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2667: Sensor is null after hNVStorage Validate
[  185.007683] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[  185.019323] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[  185.030876] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[  185.040410] no set file for sensorid 0005 0248 found
[  185.045688] facetimehd 0000:03:00.0: fthd_hw_deinit
[  185.049812] facetimehd: probe of 0000:03:00.0 failed with error -22

```

christophgysin, 2017-04-22:
Looking at the code, the error is coming from:
https://github.com/patjak/bcwc_pcie/blob/master/fthd_isp.c#L577-L581
```
if (!filename) {
    pr_err("no set file for sensorid %04x %04x found\n",
        dev_priv->sensor_id0, dev_priv->sensor_id1);
		return -EINVAL;
}
```
Simply commenting that makes it work.

christophgysin, 2017-04-22:
Filed an issue upstream: https://github.com/patjak/bcwc_pcie/issues/112

Dunedan, 2017-04-22:
Awesome work. Thanks a lot. I'll keep track of the upstream ticket and will update the README once it got solved. :+1:

christophgysin, 2017-04-23:
For reference, this is the patch that makes it work for me: https://github.com/christophgysin/bcwc_pcie/commit/ce56febc99bbf06107d91f19a48ba93a1fa30938

christophgysin, 2017-04-25:
Upstream issue has been fixed. I guess this can be closed, and the README updated.

Dunedan, 2017-04-25:
Thanks. I added that as a comment in the README now. :smile:

closed 2017-04-25 by Dunedan