Dunedan/mbp-2016-linux #17: Sound on MacBookPro13,1

christophgysin, 2017-04-24:
Sound works over HDMI.

For the builtin speakers and headphone jack some pin configuration quirks seem to be needed, similar to this: https://github.com/torvalds/linux/blob/master/sound/pci/hda/patch_cirrus.c#L462-L467

They can be modified at runtime as described by https://bugzilla.kernel.org/show_bug.cgi?id=110561#c24

Any idea how to find out the needed configuration?

Also, the linked bug report seems to deal with a different model, we should probably open a new bug for the MacBookPro13 line specifically.

tudorbarascu, 2017-04-24:
I can also confirm it works over HDMI. Related, I bought a DELL ws15 dock and everything works under linux. Sound also as I can plug the headphones etc into its jack etc.

Dunedan, 2017-04-24:
Ok, the confirmation that sound is working over HDMI is quite nice, although not really surprising as that should bypass the internal audio hardware as it does for USB-connected audio devices.

I'll adjust the README to make more clear what's working and what's not in regards of sound. Let's keep this issue open for figuring out how to get the internal microphones, speakers and the audio jack working.

tudorbarascu, 2017-04-29:
I just thought of passing the sound to my phone. It seems that it should work for android at least https://www.linux.com/learn/how-stream-audio-your-linux-pc-android .
Maybe there are also solutions for iphone.

Strafos, 2018-01-27:
When it says "What's working is audio via HDMI or any USB-connected audio device, so that can act as a workaround until internal audio is working." on the README, does that mean I could use a usbc to audio jack to connect my headphones?

christophgysin, 2018-01-27:
Yes, absolutely. What doesn't work is playing sound through the internal speakers, or through the builtin headphone jack.

cyrusmg, 2018-04-23:
Is there anything I can do to help resolve this issue ?

tomaytotomato, 2018-11-16:
So my setup I am using HDMI with my headphones which works great.

The only thing is Ubuntu switches back to the internal audio/jack after some time of inactivity.

Is there a way to default hdmi audio output?

christ776, 2018-12-02:
Hi @tomaytotomato , can I ask you how did you set up HDMI audio? I've tried using `amdgpu.audio=1` kernel parameter but that didn't work.

christophgysin, 2018-12-02:
On my mbp13,1 I use:

    $ pactl set-card-profile 0 output:hdmi-stereo-extra1

dvulricht, 2018-12-08:
@christ776 
@Dunedan 
Hi guys,
I've came to the conclusion that debuging the kernel of MacOS's AppleHDA could be quiet hard, I can stop the AppleHDA from initializing with AppleHDA Disabler.kext but I can't load it on my own later on, so it's impossible to follow on. I believe it would require patching of a custom compiled XNU kernel to proceed debuging on OSX.

The official apple Thunderbolt 2 gigabit network adapter (with Thunderbolt3->Thunderbolt2 adapter) is working for network kernel debugging on MacOS, should be working for Windows10 as well.

What could be easier is to debug the HDAudio.sys driver in Windows10. The apple Thunderbolt2 Gigabit Network Adapter is supported by Windows10 Kernel Debugging actually.
The drivers are loaded/unloaded much easier in Windows10 than on OSX(they are quiet integrated into the libkern kext I believe. "kdnet.exe" detects the NeXtreme Gigabit ethernet adapter and sets bcd with a proper key for connecting. However from Host Windows10 I can't connect to the Windows10's kernel booting on the target Macbook. Seems like the network adapter is not existent during boot.
If I connect from a host machine to a Virtual machine (same win10 install thru virtualbox running of USB thru rawvmdk), WinDBG is attaching fine.
I believe the problem is in thunderbolt2 gigabit adapter is not powered up properly during boot. When I was using external GPU thru thunderbolt3, I needed to use "apple_set_os.efi"(google it), so I can get the eGPU powered on before Windows10 loads. I'm working on that currently but the setup is quiet complex because I run the Windows10 on a USB drive, which is something quiet advanced by itself.
Anyone ever debugged Windows10 on Macbook? Can't find anything on the internet about that.

Btw Macbook 8,1, 9,1 and 10,1 are Cirrus Logic CS4208. No I2C setup, headphones are working with little setup, speakers might be soon working, depends when Leif Liddy will do some tests. You got any of those ? I have some settings for testing.
Macbook Pro 13,1 14,1 are CS8409. I2C instructions are required to power up the amps and nothing is working yet, i'm working on the CS8409 model.

Dunedan, 2018-12-08:
> However from Host Windows10 I can't connect to the Windows10's kernel booting on the target Macbook. Seems like the network adapter is not existent during boot.

Have you looked into using an USB-network adapter instead of one with Thunderbolt? Might be easier than to get Thunderbolt working during boot.

> Anyone ever debugged Windows10 on Macbook?

What I did was to install Windows in a VM under Linux. That's a great solution for reverse engineering USB-devices like the TouchBar or the Webcam, but unfortunately doesn't help when debugging things like SPI-, I2C- or Thunderbolt-connected devices. :disappointed: 

> Btw Macbook 8,1, 9,1 and 10,1 are Cirrus Logic CS4208. No I2C setup, headphones are working with little setup, speakers might be soon working, depends when Leif Liddy will do some tests. You got any of those ? I have some settings for testing.

Great news. :+1:
Unfortunately I don't have a 12" MacBook to test, but maybe others have.

MarvinBlstrli, 2019-01-05:
Are there any updates? Can I help for some testing?
I'm running Ubuntu 18.04 with Kernel 4.20 on MBP 14,3

christ776, 2019-01-28:
Hi @dvulricht. Were you able to make some progress with getting CS8409 working?

ClashTheBunny, 2019-01-28:
Folks, this is a place for development primarily.  It is doubtful that asking people for progress or offering to test helps.  People will probably show up with anything they've worked on here and ask for testing when ready, so just subscribing to the issue should get you what you need without emailing a bunch of people.

dvulricht, 2019-04-12:
@MarvinBlstrli no progress from my side, ofcourse you can help
If you can get into Windows 10 Kernel debugging on a real Macbook 2016+ it will be of great help, it can be done through USB3.0 <->USB3.0 (Male-to-Male debug cable) or Ethernet<->Ethernet.
You will need two Windows 10 boxes(can be PC/laptop and Macbook 2016+), same release,same updates.
Windows 10 official documentation provides information about "bcedit" and the specific types of supported Ethernet interfaces or USB3.0 cables.
I suspect USB2.0 is a viable option but the cable is only available online in specific countries and it is very expensive USB2.0 windows debug cable. Thought USB2.0 might be the only solution to get into Windows 10 Kernel debugging on Macbook because of the USB C-> USB3/2 adapter I use, I suspect it might behave better with USB2.0 cable, I cannot tell until tested.
In fact, for this to be solved it might be even needed to create a custom USB-C <-> USB3.0 Windows debug cable, something that doesn't exist on the market even.

If you can get into the Windows 10 kernel on macbook it would be greatly appreciated to provide details how. From there on I will catch up on reverse engineering the audio driver.
I tried Ethernet to Ethernet, did not succeed. 
I tried USB3.0<->USB3.0 but the cable was done by myself destroying two normal USB3.0 cables, and I might have made a loose connection or connected something wrong, I did check it 10 times tho.
Never connected to my Windows 10 on macbook in kernel debugging mode.

I really hope someone can help with this it will be a huge step ahead.

@ClashTheBunny I really don't mind, I just had some other stuff to do that's why I didn't respond earlier. But I was watching the thread.

marbetschar, 2019-04-16:
@dvulricht 

> Btw Macbook 8,1, 9,1 and 10,1 are Cirrus Logic CS4208. No I2C setup, headphones are working with little setup, speakers might be soon working, depends when Leif Liddy will do some tests. You got any of those ? I have some settings for testing.

I own a MacBook 10,1 Retina and would love to help to get audio working. Which settings do you like to be tested?

dvulricht, 2019-04-16:
> @dvulricht
>
>
> I own a MacBook 10,1 Retina and would love to help to get audio working. Which settings do you like to be tested?

We did the GPIO setting tests with Leif that didn't bring success.
Check out my previous reply, help with Windows 10 kernel debugging is needed.

marbetschar, 2019-04-16:
@dvulricht if I got this straight I need a Laptop for debugging and my MacBook. Both running Window 10. Then connect those via USB 2.0 and then I should be able to help with the reverse engineering, right?

Don't own a second Laptop as of now, need to see if I can get one.

dvulricht, 2019-04-16:
@marbetschar 
Cable connections: USB2.0/USB3.0/USB3.1(USB-C) or Ethernet cable.
Recently I found https://wbenny.github.io/2018/10/16/kdnet-over-usb.html which might be worth trying to see if we can get USB-C kernel debugging, which will be great.

You don't need a second Laptop, any old Desktop PC that can run 32/64bit Windows10 is okay for debugging, probably you can have that second PC running of a Linux server in Virtualized environment even, should be even possible  to use a remote server from online hosting but the USB over internet could be a little bit tricky.

marbetschar, 2019-04-16:
@dvulricht right now I might be able to get a second, physical machine for testing and will be able to put it next to my MacBook. Seems like having physical connections will make it a bit easier to get this done (2 Laptops, connected via USB).

Need some time to put Windows on both machines and need to dig out a USB 2.0 / 3.0 cable though. to connect to the USB - C port of my MacBook.

Will report back once everything is setup.

marbetschar, 2019-04-25:
@dvulricht brief Update: Was able to install Windows 10 on two computers and have installed the Windows debugging tools according to the official Microsoft documentation. Now I'm stuck with the following situation - what's the next step?

UsbView on my MacBook 10,1 Retina indicates that the USB 3.0 controller is debug capable - but does not say so for any USB 2.0:

[image]

When I start WinDbg on the host computer (an old ThinkPad T61), it claims that it tries to connect which does not work - which is not the main issue, was expecting that since I don't have a proper cable yet. But what's worrying me is that in the titlebar WinDbg claims that it's going to use USB 2.0:

[image]

Will this work even with a proper cable? Or does the host computer also need USB 3.0 to be able to debug the kernel on the target computer (= MacBook)?

marbetschar, 2019-04-25:
In case things _should_ work, will this cable to the job without cutting anything? The KDNet article claims that _"any USB-C to USB-A should to the trick"_...?

https://www.digitec.ch/de/s1/product/equip-usb-30-c-male-to-a-male-cable-025m-usb-kabel-10122795

dvulricht, 2019-04-26:
@marbetschar 
"USB 3.0: yes", then only USB3.0 is viable option.
usb2:targetname ... it's windows tool, might work with usb3.0 despite that.

USB3.0 should be available on both laptops.

Yes the cable you found from digitec is the same mentioned in KDNet over USB article. Will be sad if this kdnet network adapter is only available on Windows 10 ARM/ARM64.

roadrunner2, 2019-04-27:
Good to see some work being done here! However, the MB's appear to have a substantially different (and simpler) audio setup from the recent MBP's, and hence the solution and driver changes will be completely different. Since this bug was originally about the MBP13,1, I think it would be good idea to put the discussions for audio on MB's into a separate issue and avoid/reduce confusion between the two; discussions for all MBP's though I think should stay here, as those all have substantially the same audio hardware (not identical, but the differences are relatively small and easily dealt with).

marbetschar, 2019-04-27:
@roadrunner2 done. @dvulricht see https://github.com/Dunedan/mbp-2016-linux/issues/102 for subsequent discussion.

Have to check whether I’m able to dig out an USB 3.0 device to start debugging.

leifliddy, 2019-07-10:
Looks like some progress is being made:

https://github.com/davidjo/snd_hda_macbookpro

MrAsimov, 2020-06-21:
@christophgysin did you get internal audio and microphone working for your MacBookPro13,1 model by appling the Cirrus 8409 HDA chip driver from https://github.com/davidjo/snd_hda_macbookpro ? If so, can you provide me the dmesg and hwinfo information over at https://github.com/Dunedan/mbp-2016-linux/pull/149 so I can update the pull request with related dmesg and hwinfo information and update the README.md file properly.

christophgysin, 2020-06-21:
@MrAsimov Good to hear that there's been progress, but I'm afraid I won't be able to test it. as I replaced my laptop a few months ago.

MrAsimov, 2020-06-21:
@christophgysin to bad, and I assume you don't have any other newer MBP model greather then 13,1 with Linux?

christophgysin, 2020-06-21:
Unfortunately not.

mechinn, 2020-09-11:
I don't have that driver installed but I do have a MacbookPro13,1 if it would help @andyholst 
[dmesg](https://termbin.com/skjx)
[hwinfo](https://termbin.com/fcu7d)

andyholst, 2020-09-12:
@mechinn read the README.md how to apply the Cirrus 8409 kernel driver patch then verify that the audio input & output is working on your MBP 13,1 model and if things works; then make a pull request and update the README.md file with proper audio input & output  status for MBP 13,1 and adding the updated dmesg and hwinfo information in the MacBookPro13,1 directory.

mechinn, 2020-09-15:
@andyholst worked like a charm thanks

closed 2020-09-16 by Dunedan

Dunedan, 2020-09-16:
This issue got closed as #161 added the information to the README that audio works now for the MacBookPro13,3. In case there are still problems left, we can either reopen this issue or open a new one.

PR #161 (2020-09-16): MacBookPro13,1 audio working
Fixes #17