Dunedan/mbp-2016-linux #18: 4.11-rc8 : No keyboard before encryption keys entered

mhkabir, 2017-04-24:
@roadrunner2 I had a look at the `keyboard.conf` to see what was loaded, but I couldn't find anything different to load to get the keyboard/touchbar turned on before the encryption key entry stage. It loads fine after I enter the encryption key, so it's a config step somewhere which I'm missing.

Any more tips?

christophgysin, 2017-04-24:
What distribution are you running? Could you attach the output of dmesg?

roadrunner2, 2017-04-25:
Also, verify that all those 4 modules mentioned in the ```keyboard.conf``` are actually present in your initrd (e.g. using something like ```sudo /usr/lib/dracut/skipcpio /boot/initramfs-....img | gunzip -c | cpio -tv```).

Dunedan, 2017-04-26:
I noticed that I had the same problem after the patch which fixed the caps lock LED (https://github.com/cb22/macbook12-spi-driver/pull/14). The error message was quite speaking:
```
[    5.053957] applespi: Unknown symbol crc16 (err 0)
```
Adding `crc16` to the modules to include into the initrd solved it for me.

Not sure if that's the same problem, but I thought it might be worth mentioning.

mhkabir, 2017-04-26:
Will check out everyone's advice later today. Thanks for the pointers!

christophgysin, 2017-04-26:
Interesting. Shouldn't the tool that creates the initrd resolve and include the needed dependencies?

roadrunner2, 2017-04-26:
@Dunedan That sounds like you forgot to to run ```sudo depmod``` after "installing" the module.
@mhkabir The same could be your issue.
@christophgysin You're absolutely right in that dependencies should not need to be manually resolved, but for that to work it requires the ```modules.dep``` to be up to date.

Dunedan, 2017-04-27:
I never ran `depmod` so far. Maybe I should. :wink:

Dunedan, 2017-05-07:
@mhkabir Any news regarding this topic? Did you manage to get it to work?

Dunedan, 2017-07-13:
@mhkabir Any update? If not I'm going to close this issue.

mhkabir, 2017-07-13:
Closing for now; I took a break for Linux on the MBP '16. The wifi firmware issue is the real killer for me.

closed 2017-07-13 by mhkabir