Dunedan/mbp-2016-linux #19: The Caps Lock LED should now be working

luca-viggiani, 2017-04-26:
According to [this commit](https://github.com/cb22/macbook12-spi-driver/commit/2dcaf611904e94f10e1d71d0c6b2a1cd50be0317), the Caps Lock LED should now be working. You may want to upadate README.md

christophgysin, 2017-04-26:
Does it work for you? On which model?

luca-viggiani, 2017-04-26:
Not personally tested yet but according to the [conversation](https://github.com/cb22/macbook12-spi-driver/pull/14) it should work on any MBP 13,* as well as on MB 8.

Dunedan, 2017-04-26:
It works fine with on my MacBookPro13,2 with the latest code of https://github.com/cb22/macbook12-spi-driver since https://github.com/cb22/macbook12-spi-driver/pull/14 got merged.
I updated the README accordingly. Thanks for pointing it out @luca-viggiani and feel free to directly open a PR next time if you want.

closed 2017-04-26 by Dunedan