Dunedan/mbp-2016-linux #25: Can't start my Fedora USB stick at all

romu70, 2017-06-14:
Hello,
I think this is the best place to share my issue. I've a MacbookPro 13,1.

Some weeks ago, I tried to start a Fedora 25 USB stick (kernel 4.10), no way. I've just tried with Fedora 26 Beta (kernel 4.11) and same problem. I tried with the official Apple USB-C adapter and a Dell A200 as well, no change.

When the mac starts and I press the option key, I see:

- macOS
- Fedora Live
- EFI Boot
- EFI Boot

Both "EFI Boot" and "Fedora Live" start the Fedora Live image. I get to the Fedora Live start menu, I choose "try Fedora", the USB stick blinks for few seconds, and finally, nothing happens. No more blink and blank screen.

As I read you almost successfully run Linux on macs, I guess there is something I do the wrong way. Any help would be great, thanks.

tudorbarascu, 2017-06-14:
@romu70 I tested Fedora just for fun around 2 weeks ago and it didn't work for me also (25 and 26 beta) -- black screen after booting although I tried various boot parameters.

I'm using Debian so it wasn't a bother for me.

christophgysin, 2017-06-14:
Can you try removing "rhgb" and "verbose" from the kernel commandline? I assume Fedora uses grub, in which case you can select an entry and press "e" to edit the commandline. That should hopefully produce some output, to get an idea where it gets stuck.

romu70, 2017-06-15:
There are no such parameters, but a "quiet" one. I tried to remove it, but when I click on Ctrl+X to start, the command is not recognised. Any idea how to get the Ctrl key to be triggered?

roadrunner2, 2017-06-15:
@romu70 Use F10 instead of ctrl-x.

Make sure you add the `intremap=nosid` kernel param; also, I recommend removing `quiet`; lastly, if the kernel you're booting is < 4.9 you also need `nomodeset`.

romu70, 2017-06-15:
ok I'll try. The kernel is 4.11.3. More news asap.

romu70, 2017-06-16:
Hello,
I'm not familiar at all with startup kernel tweaking. So here is a photograph of my screen, please tell me if I did this the right way?

The result: nothing. Clicking on F10 does the USB stick to blink for few second, I 've just a small grey rectangle on the screen (the same as the screenshot) and absolutely no trace.

I tried with the 2 days ago daily build of Ubuntu 17.10 and the live starts correctly. The trackpad is not recognised, but this is a different issue.

Thanks.

[image]

romu70, 2017-06-21:
Hello folks,
No answer to my previous message...I jast wanna know of I did things the right way or not. Thanks for the help.

Dunedan, 2017-06-23:
> I'm not familiar at all with startup kernel tweaking. So here is a photograph of my screen, please tell me if I did this the right way?

Looks good. :+1:

> The result: nothing. Clicking on F10 does the USB stick to blink for few second, I 've just a small grey rectangle on the screen (the same as the screenshot) and absolutely no trace.

I've no experience with Fedora (Debian here as well), but it could be related to how they boot there. `linuxefi` is different to Debian/Ubuntu and the mentioning of pxeboot is also interesting.

> I tried with the 2 days ago daily build of Ubuntu 17.10 and the live starts correctly. The trackpad is not recognised, but this is a different issue.

Keyboard shouldn't work as well until you install the [`applespi`](https://github.com/cb22/macbook12-spi-driver) driver.

romu70, 2017-06-24:
Great, thanks you. Small question : how to you install the applespi driver if you can't use the keyboard? Through SSH?

Dunedan, 2017-06-24:
Or just use an external usb-connected keyboard. :wink:

roadrunner2, 2017-06-26:
@romu70 I'm not sure why Fedora isn't displaying/booting properly. I realized after my last comment that with F26 you don't need the `intremap=nosid`, so sorry for that noise. I had originally installed from F25 with a 4.8 kernel, though that was on a MBP13,3 which has the AMD dGPU - but I did recently verify that that Live USB booted fine for me (with the `intremap=nosid` and `nomodeset` kernel params) even if I switched to the iGPU instead. Furthermore I'm currently running F25 with their 4.11.6 kernel, and even a 4.12-rc6 kernel, and those are fine too. So it's a bit odd why the F26 Live USB is having issues, but obviously something's different on that kernel. Sorry I don't have any useful suggestions.

romu70, 2017-06-26:
Pretty strange indeed, I failed booting both F25 and F26. But I'll give it a try again without the intremap parameter. I'll keep this report updated.

puellanivis, 2017-06-26:
I'm running openSUSE Tumbleweed with kernel 4.11.3… and it's still failing to boot if I don't put in `intremap=nosid`… weirdly, also still requires the DSDT change… but weirdly, not after most reboots or even restarts… like, only if it's powered down for a long time does it fail, and need the DSDT change pushed anew… it's weird. :(

Just so much inconsistency that I've been seeing as of late…

romu70, 2017-07-17:
Hello,
Just tried to boot my Fedora 26 stick with "intremap=nosid", same, no boot, no message.

Dunedan, 2017-10-08:
Any update regarding this issue?

romu70, 2017-10-08:
Not on my side, and I gave up with the Macbook Pro to switch back to a XPS13.

Dunedan, 2017-10-09:
I just tried Fedora 27 Beta and can't reproduce the problem with it. After Grub, the screen goes black a while, but comes back, while boot is continuing. I had some problems with timeouts during boot, causing it to land in emergency mode, but they seems to be related to the USB stick I used to boot with.
After several tries it worked fine and I ended up with a fully functional graphical interface (except for internal keyboard and mouse of course). The NVMe SSD gets detected out-of-the-box as well.

As the Fedora 27 live image works fine I'm closing this issue.

gohai, 2017-10-14:
@Dunedan I am having no luck with a new MacBookPro14,1 w/ 256 GB SSD, and Fedora 27 Beta as well as Fedora 26. Same symptoms as @romu70.

FWIW: I am trying it with a USB C (USB 3) flash drive, that I prepared using Fedora Media Writer. I will try another storage medium (USB 2) in a little while, as I remember that I had some issue booting Fedora Live images off fast USB 3 flash drives with my previous MacBook Air.

Dunedan, 2017-10-15:
What are the symptoms? Black screen as originally reported or something else?

gohai, 2017-10-15:
Black screen. If I attempt to modify the kernel command line, and press F10 to start, the grey rectangle that once was the cursor remains visible, everything else turns black.

This is true for all F26 and F27 boot options that I tried, with the exception that when I select the _Fedora Live_ option on F27 Beta, the laptop will boot into macOS rather than show the symptoms above.

@Dunedan How is your setup different than mine?

Dunedan, 2017-10-15:
Well, I don't know. I just had the black screen for a few seconds and then the boot continued as it should.

What I did differently is how I wrote the image to the USB drive. I simply used `dd` instead of the Fedora Media Writer.

roadrunner2, 2017-10-16:
I have to mostly concur with @gohai: if I boot F27 Live with the dGPU (AMD) then the display works fine, but if I switch the EFI vars to boot with the iGPU active then I get only a black screen. I've tried a bunch of kernel parameters (`nomodeset`, `intremap=nosid`, `vga=...`, some `i915.xxx` ones), but so far nothing worked. I use `dd` too, but that should be irrelevant.

Dunedan, 2017-10-16:
Have one of you made the update to MacOS High Sierra before? Because I did and that's right now the only difference I can think of. Doesn't sound too relevant, but as the EFI is updated during the High Sierra update it's worth a shot.

gohai, 2017-10-17:
@Dunedan Were you trying with a **2017** model? I realized that the original post was about the 2016 one, while I am attempting to get it installed on the new one.

Today I updated to High Sierra. Unfortunately it still doesn't work.
I also tried using `dd` instead of the Fedora Media Writer.

Couple of ideas:
Do you have the 256 GB SSD model? (I do)
Did you try with a USB 2.0 storage device? (I tried with 3.0)
Does your machine also boot into macOS when you select the `Fedora Live` option on F27 Beta in the boot menu?

Dunedan, 2017-10-17:
> Were you trying with a 2017 model?

I, as well as @roadrunner2, tried with two different models from 2016.

> Do you have the 256 GB SSD model?

512GB here, but that shouldn't matter. The only difference in terms of SSD is the NVMe controller, but the MacBook Pro 13,2 and MacBook Pro 14,1 use the same one.

> Did you try with a USB 2.0 storage device?

USB 3.0 drive, but connected through an USB 2.0 hub.

> Does your machine also boot into macOS when you select the Fedora Live option on F27 Beta in the boot menu?

I don't know anymore, but at least one of the boot menu entries worked.

gohai, 2017-10-27:
I have now also tried with a USB 2.0 hub - still no joy.

Dunedan, 2017-10-28:
> Does your machine also boot into macOS when you select the Fedora Live option on F27 Beta in the boot menu?

It boots whatever is selected as primary boot option. In my case that's not macOS, but Debian.

I just tried it again and it still works on my MacBookPro13,2. Maybe listing the steps I did helps:

* Download Fedora 27 Beta-image from https://getfedora.org/en/workstation/prerelease/ (direct link: https://download.fedoraproject.org/pub/fedora/linux/releases/test/27_Beta/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-27_Beta-1.5.iso)
* Copy it with `dd` to an USB drive (following the instructions from https://fedoraproject.org/wiki/How_to_create_and_use_Live_USB#Command_line_.22direct_write.22_method_.28most_operating_systems.2C_non-graphical.2C_destructive.29):
```
    dd if=Fedora-Workstation-Live-x86_64-27_Beta-1.5.iso of=/dev/sdx bs=8M status=progress oflag=direct
```
* Connect the USB-drive to the MacBook Pro (this time I directly connected an USB 3.0 USB-drive without additional hub).
* Power the MacBook Pro on an hold the "Option"-key pressed.
* From the boot menu choose the outer-right "EFI boot"
* You should see the Grub menu. Choose one of "Test this media & start Fedora-Workstation-Live 27_Beta" or "Start Fedora-Workstation-Live 27_Beta".
* The screen should go dark for ~10 seconds, afterwards you should see the regular boot messages.
* If you get dropped to an "emergency shell" power the MacBook Pro off and on and try again (happens from time to time, but I don't know why).
* Otherwise after waiting a little, you should be greeted by Gnome with a jellyfish wallpaper.

gohai, 2017-10-28:
@Dunedan Thanks a lot for providing me with these detailed steps - unfortunately I still can't get my 14,1 to work even when I follow them.

At this point I have:
* verified the SHA256 of the image file
* tried two different USB drives
* tried with and without a USB 2.0 hub in between
* tried F27 Beta as well as F26

I want to share a few observation in the hope that you (or someone else with a 14,1 that works) could check this against your configuration, or that it gives someone an _aha_ moment:

Between the EFI menu and the Grub menu I see for a split-second a little bit of text displayed. It's too fast to read, but I managed to take a photo. Here's parts of it:

    APFSStart:1384: Mounting with apfs_efi_osx-748.1.47
    efi_fusion_pairing:620: Container a05...
    efi_fusion_pairing:625: fusion uuid: 000...
    (...)
    sanity_check_allocated_blocks:144: fs_alloc_count mismatch: fs root nodes 93802 extent 4561 omap 1263 snap_meta 1 udate 5361895 fs_alloc_count 5502996
    (...)

So, could it be that perhaps Apple's APFS containers (see below) confuse Grub? Do you also have them on your machine? I think it's conceivable that Apple only sets them up on newly initialized drives, and not when you're updating, and perhaps already have a custom partition scheme?

Here's the output of `diskutil list` on my 14,1. (The `LIN` partition is a placeholder for my Fedora install.)

    /dev/disk0 (internal):
       #:                       TYPE NAME                    SIZE       IDENTIFIER
       0:      GUID_partition_scheme                         251.0 GB   disk0
       1:                        EFI EFI                     314.6 MB   disk0s1
       2:                 Apple_APFS Container disk1         186.7 GB   disk0s2
       3:                  Apple_HFS LIN                     63.9 GB    disk0s3
    
    /dev/disk1 (synthesized):
       #:                       TYPE NAME                    SIZE       IDENTIFIER
       0:      APFS Container Scheme -                      +186.7 GB   disk1
                                     Physical Store disk0s2
       1:                APFS Volume macOS                   11.8 GB    disk1s1
       2:                APFS Volume Preboot                 41.4 MB    disk1s2
       3:                APFS Volume Recovery                1.0 GB     disk1s3
       4:                APFS Volume VM                      1.1 GB     disk1s4

Secondly, when I enter the Grub command line and do a `ls` I got this - curious if this matches what you (or other people see):

    (hd0) (hd0,apple3) (hd0,apple2) (hd0,apple1) (hd0,msdos2) (hd1) (hd1,gpt3) (hd1,gpt2) (hd1,gpt1) (cd0) (cd1) (cd2) (cd3) error: failure reading sector 0x2b7753b from `cd0'.
    error: failure reading sector 0x2b77520 from `cd0'.
    error: failure reading sector 0x2b7753b from `cd1'.
    error: failure reading sector 0x2b77520 from `cd1'.
    error: failure reading sector 0x2b7753b from `cd2'.
    error: failure reading sector 0x2b77520 from `cd2'.
    error: failure reading sector 0x2b7753b from `cd3'.
    error: failure reading sector 0x2b77520 from `cd3'.

Any ideas? `hd0` has parts of Fedora. `hd1` has an `unknown filesystem` so I presume it is APFS.

Dunedan, 2017-10-29:
> Between the EFI menu and the Grub menu I see for a split-second a little bit of text displayed. […]
> Do you also have them on your machine?

I get some text for a split-second as well. I haven't checked if it's really the same text shown, but I'm pretty sure it is. That happens since the update to macOS High Sierra, but I haven't noticed any negative impact.

> Here's the output of `diskutil list` on my 14,1.

My output is slightly different, as I've partitioned the disk with different partition sizes and one more partition, although that shouldn't matter. The only significant difference I see is that my Linux partitions aren't marked as type "Apple_HFS", but as "Linux Filesystem" instead.

Everything relating to the disk shouldn't affect how the Live image boots, as that shouldn't touch the disk (unless you try an installation from within the Live image).

> Secondly, when I enter the Grub command line and do a ls I got this - curious if this matches what you (or other people see):

I get exactly the same output from the Fedora-Live Grub command line.

gohai, 2017-10-30:
@Dunedan Thanks for checking!

A couple more data points: I just tried debian-live-9.2.0-amd64-gnome - this brings me all the way to the graphical Desktop. Perhaps it's really a Fedora thing? With Debian I only see a single EFI entry, the `APFSStart` message pops up also, but when I do `ls` in Grub (version 2.02~beta3-5), I don't get any errors relating to the `cd` devices. Instead the output is:

    (memdisk) (hd0) (hd0,msdos2) (hd1) (hd1,gpt3) (hd1,gpt2) (hd1,gpt1) (cd0) (cd1) (cd2) (cd3)

gohai, 2017-10-30:
I reported this to Fedora's bugtracker: [#1502252](https://bugzilla.redhat.com/show_bug.cgi?id=1502252)

CC @romu70 @tudorbarascu, in case you want to add to that.

gohai, 2017-10-30:
BTW: My EFI ("Boot ROM") version is "MBP141.0167.B00". I read an [article](https://motherboard.vice.com/en_us/article/evpmyw/some-macos-users-arent-getting-the-firmware-security-patches-they-think-they-have) that stated that machines end up running very different versions, as any installation failures are silently ignored during OS updates. Perhaps this could be a reason why Fedora seems to work for some on 14.1 but not for others?

Dunedan, 2017-11-04:
I got "MBP132.0233.B00", but as these versions are apparently model specific I don't know if that helps.

drocsid, 2017-11-10:
Somebody pointed me here from https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7 . I tried booting both fedora27 beta iso and archlinux. Screen goes blank and doesn't come back. Using apple VGA connector with USB port. 2016 mbp 15"

kjano, 2017-11-11:
@drocsid: Sorry to hear that. I have mine running on Ubuntu 17.04 with custom ISO (`isorespin.sh -i [your iso] -k v4.11.4`) to get around some of the installation pain. I also tried Arch and from what I recall it worked.

drocsid, 2017-11-11:
@kjano  yes I tried archlinux. But if you see above some of us are having trouble booting more recent kernels, and there may be a subtle difference between macs. Please see @gohais' comments above.

drocsid, 2017-11-16:
Dunno but I just tried  Fedora-Workstation-netinst-x86_64-27-1.6.iso and it didn't work. Also tried various debian testing images and they didn't work either.

drocsid, 2017-11-16:
All my booting issues were due to filevault2 disk encryption.

Dunedan, 2017-11-18:
> All my booting issues were due to filevault2 disk encryption.

For me booting works fine, even though my macOS partition is encrypted with FileVault as well.

ClashTheBunny, 2017-11-18:
I also have no problems booting both OSs and have filevault on MacOS and
dm-crypt on Debian.  The way I switch between OSs is to hold option for Mac
and use grub for Linux.

gohai, 2017-11-19:
I tried the final Fedora 27 image: no joy on my 14,1. (The Debian LiveCD starts fine for me.)

Last week I also tried the F27 Beta on one of my student's 13,1 (like me with 256GB SSD). It didn't start up on her's either. (She was running 10.12.05 with Boot ROM MBP131.0205.B...)

gohai, 2017-11-19:
@Dunedan At this point I'd be really curious to hear of _anyone_ who is able to run a Fedora Live-CD on either a 13,1 or 14,1.

Dunedan, 2017-11-19:
The final Fedora 27 Live image continues to work fine for me.

A few more wild guesses:
- Maybe the problem isn't that nothing is shown on the display, but that the display backlight gets switched off during boot instead?
- Maybe it only works when there is already a Linux installation on the disk and "known" to EFI (although I still doubt that EFI is the culprit here)?

gohai, 2017-11-19:
As just pointed out [here](https://bugzilla.redhat.com/show_bug.cgi?id=1502252#c2), current Fedora Rawhide images **work** on 14,1 - while F26 and F27 won't.

I confirmed this resolves the issue for me! Best

drocsid, 2017-11-24:
I created a separate issue describing my issues with booting with FV2 enabled to ask how you are managing kernel boot. https://github.com/Dunedan/mbp-2016-linux/issues/45

xlfe, 2017-12-12:
I'm having the same problem on a Macbook Pro 14,1

Have now tried Fedora Workstation Live image 27 and Rawhide from today - both ended up with a blank screen (I was hopeful for RawHide after @gohai's comment [here](https://bugzilla.redhat.com/show_bug.cgi?id=1502252#c2), but it didn't work for me today).

I'm also seeing the same thing as @gohai in [his comment](https://github.com/Dunedan/mbp-2016-linux/issues/25#issuecomment-340202780) with respect to the grub ls and  "error: failure reading sector 0x2b7753b from `cd0'" etc. This makes me think its a clash between grub and the internal mac filesystem which means grub or EFI doesn't boot from the USB correctly. I was trying to figure out how to get grub to ignore my macbook's internal drive but I haven't been able to get it to do it.

I tried with both Mac OS Sierra and then again after upgrading to High Sierra -same result. I also tried a Ubuntu Live USB 17.10 and that booted into the live desktop as expected.

Can anyone figure out what has changed between Rawhide 20171118 and 20171210 with respect to the linux boot? Or where I can download Rawhide 20171118 to try it? Or would someone try a more recent Rawhide on a 14,1 ?

fzdarsky, 2017-12-28:
I've had the same problem on a Macbook Pro 14,1 (13", 2017, 2 Thunderbolt 3 ports) with the F27 live image. However, I just tried the [current RawHide](https://dl.fedoraproject.org/pub/fedora/linux/development/rawhide/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-Rawhide-20171227.n.0.iso) and it works for me: Screen goes blank again, but then after around 10s Fedora starts booting into the live system.

xlfe, 2017-12-29:
Thanks @fzdarsky - I tried that rawhide and I can now boot the live USB. Excellent. 

Now I've encountered a [new bug](https://bugzilla.redhat.com/show_bug.cgi?id=1524700) while trying to install Fedora.

fzdarsky, 2017-12-29:
@xlfe same here... :(

But: You might want to try the .iso linked to by Janek in [comment #5 on BZ1502252](https://bugzilla.redhat.com/show_bug.cgi?id=1502252#c5), which seems to be the F27 live image but with an updated 4.14 kernel.

This got me through the installation process, with the help of the virtual keyboard. Now I'm stuck at boot when I get prompted for the disk encryption password. No virtual keyboard there... Time to buy a cheap physical USB keyboard just to type the password and get the keyboard driver installed...

Dunedan, 2018-02-04:
Are there still issues with booting Fedora or can we close this issue?

Dunedan, 2018-03-18:
Closing this, as this should work now for everybody.

closed 2018-03-18 by Dunedan