Dunedan/mbp-2016-linux #26: Status of MacBookPro14,1

peterychuang, 2017-06-24:
Hi,

I've put Linux on the 2017 base model of MacBook Pro (13 inch without touch bar). The result of ```get-info.sh``` can be found [here](https://github.com/peterychuang/mbp-2016-linux/tree/master/MacBookPro14%2C1).

I haven't tested everything yet, but so far the info on your repository appears mostly accurate for my machine. I will let you know when I see any problems.

Thanks @Dunedan @roadrunner2 and @cb22 for your great work!

Dunedan, 2017-06-24:
Awesome.
Can you please make a pull request with the output so we have it in here as well?

roadrunner2, 2017-06-24:
@peterychuang Thanks for the info. Could I ask you a favour? Could you verify the touchpad has the same ranges as the corresponding 13,* model (or else determine what the new ranges are)?

To do that please get the latest drivers from my clone of the macbook-spi repo, https://github.com/roadrunner2/macbook12-spi-driver, add `CFLAGS_MODULE=-DDEBUG_TOUCHPAD_RANGES` to the make modules command, and rebuild and reload the applespi driver; then move your finger to all edges of the touchpad and you should see the ranges being reported in `dmesg`; finally if you could report the largest (last) range printed here, I can make any necessary adjustments and send a pull request. TIA.

As an added bonus, that repo also has a patch that assumes your trackpad is the same as the 13,* models, so with the updated [hwdb](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#file-61-evdev-local-hwdb) chances are you should have proper behaviour.

peterychuang, 2017-06-25:
I've tested the machine for a day now. So far, other than Audio, Bluetooth, and Suspend and Hibernation, most devices work as indicated in the repo. I haven't tested System Management Controller and Thunderbolt.

However, there's some problems, and I'm not sure if it's specific to my model, my setup, or if it's because I've done something wrong. I have Arch Linux installed inside an LVM, which is encrypted. 

```
nvme0n1                                       259:0    0   113G  0 disk  
├─nvme0n1p1                                   259:1    0   500M  0 part  /boot
└─nvme0n1p2                                   259:2    0 112.5G  0 part  
  └─luks-[UUID]                               254:0    0 112.5G  0 crypt 
    ├─vol-swap                                254:1    0     8G  0 lvm   [SWAP]
    └─vol-root                                254:2    0 104.5G  0 lvm   /
```

When the machine boots, sometimes it appears to look for the root partition without ever prompting my key, and then it times out and fails to boot; sometimes, after unlocking the partition and getting to the desktop environment, the touchpad doesn't work; other times, the wifi doesn't work.

Also, for some reason, it fails to recognise my USB-C/USB-A flash drive (the SanDisk one where there's USB-C on one end and USB-A on another) if I plug it directly to the USB-C port. However, if I plug the USB-A end of the flash drive into a dongle and plug the dongle into the USB-C port of my MacBook, the flash drive is recognised.

I'd appreciate if someone could let me know where I should look at to solve the problems.

@roadrunner2 Sure, though it will take some time, as I haven't done this before.

Dunedan, 2017-06-25:
> When the machine boots, sometimes it appears to look for the root partition without ever prompting my key, and then it times out and fails to boot;

That's a known issue with the NVMe controller. Happens with my MacBookPro13,2 regularly as well. If you disable the splash screen you should see a related error message appear after ~30 seconds. My gut feeling is that it happens more regularly when the notebook wasn't powered down properly before. So far no solution or workaround aside from simply rebooting till it works is known.

peterychuang, 2017-06-25:
@Dunedan Good to know I'm not alone. Perhaps it's a good idea to document this issue on ```readme.md```?

roadrunner2, 2017-06-25:
@peterychuang Perfect, thanks! So that confirms the ranges on the 14,{1,2} are the same as on the 13,{1,2} models.

Dunedan, 2017-06-30:
> Perhaps it's a good idea to document this issue on readme.md?

Did that just now. :smile:

Dunedan, 2017-07-11:
@peterychuang How is Linux running on your machine now? Any other unknown issues or can we already close this issue?

peterychuang, 2017-07-12:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/Dunedan/m
> bp-2016-linux","title":"Dunedan/mbp-2016-linux","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in GitHub","url":"https://gi
> thub.com/Dunedan/mbp-2016-
> linux"}},"updates":{"snippets":[{"icon":"PERSON","message":"@Dunedan
> in #26: @peterychuang How is Linux running on your machine now? Any
> issue?"}],"action":{"name":"View Issue","url":"https://github.com/Dun
> edan/mbp-2016-linux/issues/26#issuecomment-314560307"}}}

I haven't encountered any problems which are not already documented
here. I'll close this issue. Thanks.

closed 2017-07-12 by peterychuang