Dunedan/mbp-2016-linux #29: Bluetooth Problem

peterychuang, 2017-08-05:
So I was poking around the Bluetooth, and I noticed that the DSDT contains this:

```
If (!OSDW ())
{
    Return (UBUF) /* \_SB_.PCI0.URT0.BLTH._CRS.UBUF */
}

Return (ABUF) /* \_SB_.PCI0.URT0.BLTH._CRS.ABUF */
```

This might be a stupid question, but isn't this bit the same as the SPI stuff that used to require patching to get the keyboard and touchpad working?

roadrunner2, 2017-08-05:
@peterychuang The question is far from stupid: you're quite right, this looks very much like SPI stuff and it looks like it would benefit from the generic apple-properties work @l1k has done.

However, on closer examination it turns out the issue is not the missing acpi serial-bus resource, but the missing interrupt information: there's neither an interrupt-resource in the _CRS, nor is there a _GPE name like there is for the SPI device, and hence we get these `hci_bcm` errors in dmesg:
```
hci_bcm BCM2E7C:00: BCM irq: -22
hci_bcm BCM2E7C:00: invalid platform data
hci_bcm: probe of BCM2E7C:00 failed with error -22
```
In fact, I can't find any interrupt info whatsoever for the bluetooth device in the DSDT.

So the first thing to get bluetooth running is to figure out what interrupt it uses. After that, we'll need some small patches to set the oper_speed from the baud apple property, and probably something to call the power-down, power-up, etc acpi methods appropriately (however, I know next to nothing about bluetooth and their drivers in Linux, so take this with a large grain of salt).

l1k, 2017-08-06:
Yes I did mention in the commit message of l1k/linux@9ae5c93fcd54dd078dc072cf73cfe9be2b83c225 that it'll be needed for Bluetooth as well. :-)

Looking at the schematics of the MacBook8,1 I can't find a Bluetooth interrupt pin and apparently none is needed. A quick Google search turns up <a href="http://www.nxp.com/docs/en/application-note/AN10307.pdf">this</a>:

<blockquote>
To a host system, the UART appears as an 8-bit input and output port that it can read from and write to. Whenever the host has data to be sent, it just sends these data to the UART in byte format (8-bit wide), whenever the UART receives data from another serial device it will buffer these data in its FIFO (again 8-bit wide), then it will indicate the availability of these data to the host through an internal register bit, or through a hardware interrupt signal.
</blockquote>

So the interrupt is coming from the UART on behalf of the Bluetooth device once data is received into the FIFO.

What's much more interesting are the power management methods which could be used to extend battery life:
* `BTPU` = BlueTooth Power Up
* `BTPD` = BlueTooth Power Down
* `BTRS` = BlueTooth Reset (power down followed by power up)
* `BTLP` = BlueTooth Low Power (takes argument 0 or 1, apparently toggles sleep mode)

The MBP13,3 has an additional `BTRB` method. No idea. The methods are called from `IOBluetoothHostControllerUARTTransport.kext` and various others.

l1k, 2017-08-06:
Ah looking at `hci_bcm.c` I realize this is not about an interrupt for data reception but for host wakeup. On the MB8,1 that pin is called `SMC_PME_S4_WAKE_L` and goes into the SMC, the SMC then wakes the system. So this happens transparently to the OS on MacBooks and the driver need not care about it.

The `invalid platform data` error is emitted because neither the `device_wakeup` nor `shutdown` GPIO descriptors are set. On MacBooks the driver should instead detect presence of the BTLP and BTPD/BTPU methods. The BTLP method should be called for device wakeup and BTPD/BTPU for shutdown. The driver should cache the ACPI handles of the methods on probe and invoke them from `bcm_gpio_set_power()`, `bcm_suspend_device()` and `bcm_resume_device()`.

Actually this looks fairly straightforward.

roadrunner2, 2017-08-06:
Ah, silly me, didn't actually look at who was using the interrupt for what. Ok, that indeed simplifies things then.

Regarding sleep and wakeup, the DSDT seems to indicate that BTLP should be called for both, at least on Windows:
```
    Method (_PTS, 1, NotSerialized)  // _PTS: Prepare To Sleep
        ...
        If (!OSDW ())
        {
            If (Arg0 == 0x03)
            {
                \_SB.PCI0.URT0.BLTH.BTLP (One)
                Sleep (0x03E8)
                \_SB.PCI0.LPCB.EC.EWPM = One
                \_SB.PCI0.LPCB.EC.EWDK = One
            }
        }
    }

    Method (_WAK, 1, NotSerialized)  // _WAK: Wake
    {
        ...
        If (OSDW ()) {}
        ElseIf (Arg0 == 0x03)
        {
            \_SB.PCI0.URT0.BLTH.BTLP (Zero)
        }
    }
```

This seems a bit odd, though: I (probably naïvely) would expect the runtime PM (`bcm_suspend_device` and `bcm_resume_device`) should be calling BTLP, and the system sleep (`bcm_suspend` and `bcm_resume`) to call BTPD/BTPU.

And yes, I haven't been able to figure out what the BTRB method is supposed to be for either.

l1k, 2017-08-06:
I guess BTLP puts the device in a low power state and enables wakeup. Based on the DSDT snippet you included above it seems Apple also wants this to work in sleep state S3, e.g. to wake the system upon key press on a Bluetooth keyboard. Makes sense to me. For some reason BTLP is not called in the DSDT of the MacBook8,1, this might be a bug.

In any case `hci_bcm.c` calls BTLP both for runtime suspend as well as system sleep, which is in line with what Apple does in the DSDT, so it seems fine to me. The `hci_bcm.c` driver only toggles the power enable pin (BTPU/BTPD) upon `open()` and `close()` of the device, so apparently it's powered down if Bluetooth isn't used at all.

peterychuang, 2017-08-06:
Actually I can't find the ```\_SB.PCI0.URT0.BLTH.BTLP``` bits from the DSDT in 14,1.

In case this helps, this is (maybe) all the [bluetooth related stuff](https://gist.github.com/peterychuang/c91e8e8797aaca5d0df47c7911daddb5) from DSDT in my machine.

roadrunner2, 2017-08-06:
The behaviours seem to be a bit all over the map, then: Looking at MacBook8,1 and MacBook9,1 DSDT's I see that _PTS and _WAK call BTPD and BTPU on the 8, respectively, but on the 9 there are no calls to bluetooth there at all, like on the 14,1.

@l1k I'm slightly confused by your 2nd paragraph about `hci_bcm.c`: are you saying there's already code there that calls these BTxx acpi methods?

l1k, 2017-08-07:
Sorry for not being clear, let me try again. :-)

The combined WiFi/BT chip on these machines has 3 GPIO pins of interest for BT power management: power enable (toggled by BTPU/BTPD), device wake (toggled by BTLP, active low) and host wake (going *out* of the chip and into the SMC for system wake). I meant to say `hci_bcm.c` already contains all the necessary code to toggle the GPIO pins, however it assumes that the pins are directly toggled by the driver via the GPIO subsystem. That assumption is not correct on Macs where Apple invented their own abstraction (consisting of the three ACPI methods) to hide the specific pins used on the PCH (which may vary between models).

The driver seems to do the right thing by enabling device wake both for runtime suspend as well as system sleep, I wouldn't worry too much about presence or nonpresence of BTLP in _PTS and _WAK. It's possible that they're invoking it from their Bootcamp driver on newer models, or they just forgot to enable BT power saving on Windows.

So all that's needed is to search for the BTPU/BTPD/BTLP methods in the ->probe hook if x86_apple_machine is true, cache them in `struct bcm_device` and amend all the places where `gpiod_set_value()` is called to invoke the appropriate ACPI methods (again, if x86_apple_machine is true).

And perhaps it's also necessary to fetch device properties and adjust baud rate etc.

Let me know if my answer is still too cryptic. :-)

roadrunner2, 2017-08-07:
Thanks for the elaboration. Ok, so that basically matches my thinking.

l1k, 2017-08-13:
I don't know if anyone has started working on this, but in case noone has I've cooked up <a href="https://github.com/l1k/linux/commits/hci_bcm">this branch</a>. Top-most commit adds support for Apple's Bluetooth ACPI methods, the three preceding commits are from the spi_properties material queued for 4.14. Perhaps someone can test if this works, I don't have one of the affected machines. Thanks.

Dunedan, 2017-08-13:
@l1k: I tried out your patch, but it still doesn't work. In fact it doesn't look different to the behavior before, printing the same error messages @roadrunner2 mentioned above. Any idea how to debug this?

peterychuang, 2017-08-13:
@l1k I can confirm the patch doesn't work. Below is part of the ```dmesg```:

```
[    7.708006] Bluetooth: Core ver 2.22
[    7.708051] Bluetooth: HCI device and connection manager initialized
[    7.708054] Bluetooth: HCI socket layer initialized
[    7.708056] Bluetooth: L2CAP socket layer initialized
[    7.708063] Bluetooth: SCO socket layer initialized
[    7.712788] Bluetooth: HCI UART driver ver 2.3
[    7.712789] Bluetooth: HCI UART protocol H4 registered
[    7.712790] Bluetooth: HCI UART protocol BCSP registered
[    7.712790] Bluetooth: HCI UART protocol ATH3K registered
[    7.712791] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    7.712817] Bluetooth: HCI UART protocol Intel registered
[    7.717682] dw-apb-uart.2: ttyS0 at MMIO 0x9282b000 (irq = 20, base_baud = 3000000) is a 16550A
[    7.753856] hci_bcm BCM2E7C:00: BCM irq: -22
[    7.753858] hci_bcm BCM2E7C:00: invalid platform data
[    7.753871] hci_bcm: probe of BCM2E7C:00 failed with error -22
[    7.755161] Bluetooth: HCI UART protocol Broadcom registered
[    7.755161] Bluetooth: HCI UART protocol QCA registered
[    7.755162] Bluetooth: HCI UART protocol AG6XX registered
[   10.383828] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.383829] Bluetooth: BNEP filters: protocol multicast
[   10.383831] Bluetooth: BNEP socket layer initialized
```

l1k, 2017-08-13:
@Dunedan @peterychuang: Thanks for testing, looking at the code with a fresh pair of eyeballs I noticed that I had botched the return code of `bcm_apple_probe()`, it would return 1 on success and the caller was expecting 0 on success. Should be fixed now and I've added a commit on top to emit some debug printks. Thanks!

peterychuang, 2017-08-13:
@l1k unfortunately it's still not working. Here are the error messages, including the new debug messages from your last commit:

```
[    9.902071] Bluetooth: Core ver 2.22
[    9.902082] Bluetooth: HCI device and connection manager initialized
[    9.902084] Bluetooth: HCI socket layer initialized
[    9.902086] Bluetooth: L2CAP socket layer initialized
[    9.902090] Bluetooth: SCO socket layer initialized
[    9.908164] Bluetooth: HCI UART driver ver 2.3
[    9.908166] Bluetooth: HCI UART protocol H4 registered
[    9.908166] Bluetooth: HCI UART protocol BCSP registered
[    9.908167] Bluetooth: HCI UART protocol ATH3K registered
[    9.908167] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.908189] Bluetooth: HCI UART protocol Intel registered
[    9.923770] hci_bcm BCM2E7C:00: BCM irq: -22
[    9.923773] hci_bcm BCM2E7C:00: oper_speed=3000000
[    9.923792] hci_bcm BCM2E7C:00: BCM2E7C:00 device registered.
[    9.944515] Bluetooth: HCI UART protocol Broadcom registered
[    9.944517] Bluetooth: HCI UART protocol QCA registered
[    9.944518] Bluetooth: HCI UART protocol AG6XX registered
[   10.146949] dw-apb-uart.2: ttyS0 at MMIO 0x9282b000 (irq = 20, base_baud = 3000000) is a 16550A
[   12.409042] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.409044] Bluetooth: BNEP filters: protocol multicast
[   12.409047] Bluetooth: BNEP socket layer initialized
```

The ```invalid platform data``` seems to have gone.

l1k, 2017-08-13:
@peterychuang: Looks good! The `BCM irq: -22` message is normal, we just don't need the IRQ on Macs. What exactly is not working?

BTW I notice that the baud rate (3000000) was already reported by `dw-apb-uart` without this patch. Perhaps it's not necessary to retrieve the `baud` device property...

l1k, 2017-08-13:
Lest I forget, for the folks with a MacBook8,1: There's a peculiarity on that machine where the UART0 is attached not only to the Bluetooth controller, but also to a debug serial port on the SSD. There's a mux to switch the UART lines between the two, controlled by GPIO36 on the PCH. There is no ACPI method to control this pin directly. I assume it's set by the BIOS but there's a possibility that this GPIO needs to be toggled before Bluetooth will work.

This oddity is not present in the MBP13,3 ACPI dump. I don't have any others for these newer machines.

christophgysin, 2017-08-13:
I just tried the patch on a MBP13,1 and the platform error is gone. /sys/class/bluetooth is still empty though.

I added my DSDT in #30 if that's any help.

peterychuang, 2017-08-13:
@l1k At the moment, I'm not sure. On the surface, the behaviour of the the machine remains more or less unchanged, though I guess we're on the right track.

roadrunner2, 2017-08-13:
I had tried something like this a week ago (well, just disabled the check that caused the error) and also found that `/sys/class/bluetooth/` stayed empty - haven't had the time to dig further yet, though.

peterychuang, 2017-08-13:
```/sys/class/bluetooth``` has stuff inside when I run ```btattach --bredr /dev/ttyS0 -P bcm```. Not that bluetooth works in that circumstance though.

leifliddy, 2017-09-21:
dmesg output from MacBook9,1
**added print statements so I could follow the flow of the driver (start with xxx)

```
[    4.407901] SELinux:  Class bluetooth_socket not defined in policy.
[    5.039534] Bluetooth: Core ver 2.22
[    5.039548] Bluetooth: HCI device and connection manager initialized
[    5.039551] Bluetooth: HCI socket layer initialized
[    5.039554] Bluetooth: L2CAP socket layer initialized
[    5.039561] Bluetooth: SCO socket layer initialized
[    5.259673] Bluetooth: HCI UART driver ver 2.3
[    5.259674] Bluetooth: HCI UART protocol H4 registered
[    5.259674] Bluetooth: HCI UART protocol BCSP registered
[    5.259690] Bluetooth: HCI UART protocol LL registered
[    5.259690] Bluetooth: HCI UART protocol ATH3K registered
[    5.259691] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    5.259715] Bluetooth: HCI UART protocol Intel registered
[    5.259731] Bluetooth: xxx bcm_probe start...
[    5.259732] Bluetooth: xxx bcm_acpi_probe start...
[    5.259733] Bluetooth: xxx bcm_platform_probe start...
[    5.259852] hci_bcm BCM2E7C:00: BCM irq: -22
[    5.259852] Bluetooth: xxx bcm_apple_probe start...
[    5.259854] hci_bcm BCM2E7C:00: oper_speed=3000000
[    5.259878] Bluetooth: xxx bcm_resource start...
[    5.259880] hci_bcm BCM2E7C:00: BCM2E7C:00 device registered.
[    5.259881] Bluetooth: xxx bcm_apple_set_power 1
[    5.272356] dw-apb-uart.2: ttyS4 at MMIO 0xc182b000 (irq = 21, base_baud = 115200) is a 16550A
[    5.276115] Bluetooth: HCI UART protocol Broadcom registered
[    5.276116] Bluetooth: HCI UART protocol QCA registered
[    5.276117] Bluetooth: HCI UART protocol AG6XX registered
[    5.276118] Bluetooth: HCI UART protocol Marvell registered
[    5.288794] dw-apb-uart.3: ttyS5 at MMIO 0xc182c000 (irq = 20, base_baud = 3000000) is a 16550A
[   46.903513] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   46.903517] Bluetooth: BNEP filters: protocol multicast
[   46.903521] Bluetooth: BNEP socket layer initialized
```

[root@mac ~]# btattach --bredr /dev/ttyS5 -P bcm
Attaching Primary controller to /dev/ttyS5
Switched line discipline from 0 to 15
Device index 0 attached

```
[  287.094406] Bluetooth: xxx bcm_open start...
[  287.094420] Bluetooth: (null): hu ffff9a718ec643c0
[  287.094426] Bluetooth: xxx bcm_open mutex_lock
[  287.094429] Bluetooth: xxx bcm_open mutex_unlock
[  287.095790] Bluetooth: xxx bcm_setup start...
[  287.095801] Bluetooth: hci0: hu ffff9a718ec643c0
[  289.149877] Bluetooth: hci0 command 0x0c03 tx timeout
[  297.469902] Bluetooth: hci0: BCM: Reset failed (-110)
[  297.469906] Bluetooth: xxx bcm_setup bcm_initialize return err...
```

roadrunner2, 2017-09-24:
I took another stab at this and got it mostly working, at least on my laptop - see l1k/linux#1.

In addition to the above kernel patches, I am using the following patch (this assumes you have the systemd `btattach-bcm@.service`):
```patch
--- /usr/libexec/bluetooth/btattach-bcm-service.sh.orig	2017-09-12 07:27:30.000000000 -0700
+++ /usr/libexec/bluetooth/btattach-bcm-service.sh	2017-09-22 21:20:17.616533796 -0700
@@ -15,6 +15,12 @@
 BT_DEV="$(readlink -f $BT_DEV)"
 UART_DEV="$(dirname $BT_DEV)"
 
+# Handle intel-lpss UART devices
+DW_DEV=$(ls -d "$UART_DEV"/dw-apb-uart.* 2>/dev/null)
+if [[ -d "$DW_DEV" ]] ; then
+	UART_DEV="$DW_DEV"
+fi
+
 # Stupid GPD-pocket has USB BT with id 0000:0000, but still claims to have
 # an uart attached bt
 if [ "$1" = "BCM2E7E:00" ] && lsusb | grep -q "ID 0000:0000"; then
```
With this the bluetooth device is automatically attached on boot, and can be manually attached/detached via
```
sudo systemctl start btattach-bcm@BCM2E7C:00
sudo systemctl stop btattach-bcm@BCM2E7C:00
```
Of course, already noted by @leifliddy above, you can also just run this command:
```
sudo btattach --bredr /dev/ttyS5 -P bcm
```

@leifliddy The reason for the timeout (-110 is -ETIMEDOUT) is the wrong baudrate being used, which is fixed by the 2nd commit in my pull request.

Lastly, some issues I noticed so far:
- If the gnome bluetooth settings dialog is open, then audio may stutter and file transfers are really slow (I think it's the scanning that it's doing when that dialog is open that is causing this)
- The following error can be found in the kernel log, but unsure of its significance:
```
Sep 23 11:21:59 schlepptouch kernel: DMAR: Allocating domain for idma64.2 failed
Sep 23 11:21:59 schlepptouch kernel: ttyS5 - failed to request DMA
```
- attaching sometimes still fails with -110 (timeout) - re-attaching generally succeeds (the -16 (EBUSY) error can be ignored)

peterychuang, 2017-09-24:
@roadrunner2 To which kernel version am I supposed to apply the patches? I think because I am playing with linux-next, the patches aren't working just yet.

christophgysin, 2017-09-24:
@peterychuang The patches are on top of https://github.com/l1k/linux/tree/hci_bcm, which is based on 4.13.0-rc4.

peterychuang, 2017-09-24:
@christophgysin thanks. I've made some changes to [this commit](https://github.com/l1k/linux/commit/8883d6225a92e8154fe014eaadcbbc4b3d776019) in order to apply it to linux-next, then I applied the patches by @roadrunner2. I guess I've done something wrong somewhere.

roadrunner2, 2017-09-24:
@peterychuang, @christophgysin: I've been testing on 4.13.2 and 4.13.3. I also just pushed a [hci_bcm-4.13](/roadrunner2/linux/tree/hci_bcm-4.13) branch (currently based on v4.13.3) to make it easier to pull.

There have been a few small changes in 4.14 which may affect these patches, besides the fact that the first three in the set from @l1k are not necessary (they are backporting the properties work that is now part of 4.14). I'm looking into redoing the patches for 4.14 and will push another branch when done.

leifliddy, 2017-09-24:
great work roadrunner2!!! I think we're almost there.
I wasn't aware there was a btattach-bcm@.service...it's now working with your patch!

dmesg output from MacBook9,1 (running kernel 4.14-rc1 with l1k's and roadrunner2's bluetooth patches applied)

```
[    7.242471] Bluetooth: xxx bcm_open start...
[    7.242473] Bluetooth: (null): hu ffff9fcb63f98400
[    7.242473] Bluetooth: xxx bcm_open mutex_lock
[    7.242475] Bluetooth: BCM2E7C:00: xxx bcm_open bcm_gpio_set_power ffff9fcb61992200 true
[    7.242476] Bluetooth: xxx bcm_apple_set_power 1
[    7.256019] Bluetooth: xxx bcm_open mutex_unlock
[    7.258535] Bluetooth: xxx bcm_set_baudrate start...
[    7.258873] Bluetooth: hci0: BCM: failed to write update baudrate (-16)
[    7.258874] Bluetooth: xxx bcm_setup start...
[    7.258875] Bluetooth: hci0: hu ffff9fcb63f98400
[    7.258875] Bluetooth: xxx bcm_setup btbcm_set_bdaddr set
[    7.258876] Bluetooth: xxx btbcm.c btbcm_initialize start
[    7.258876] Bluetooth: xxx btbcm.c btbcm_initialize btbcm_reset
[    7.371575] Bluetooth: hci0: BCM: chip id 92
[    7.376021] Bluetooth: hci0: BCM: features 0x2f
[    7.380905] Bluetooth: hci0: BCM4350C0 UART 37.4 MHz Gamay Murata UHE
[    7.383183] Bluetooth: hci0: BCM (003.001.082) build 1288
[    7.387472] Bluetooth: xxx bcm_setup firmware...
[    7.389561] bluetooth hci0: Direct firmware load for brcm/BCM.hcd failed with error -2
[    7.391364] Bluetooth: hci0: BCM: Patch brcm/BCM.hcd not found
[    7.478187] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.480965] Bluetooth: BNEP filters: protocol multicast
[    7.483563] Bluetooth: BNEP socket layer initialized
```

If I unattach and reattach the bluetooth device. this error disappears:
hci0: BCM: failed to write update baudrate (-16)

Ok, looks we just need to work out the firmware situation. 

The windows inf file for this device shows that BCM2E7C corresponds to BCM4350 
%Bus.DeviceDesc%=BluePinole2E7C,    ACPI\BCM2E7C        ; BCM4350 no flash

Not sure what 'no flash' means. It's the only device listed in the inf file as 'no flash'. 
Going to try and rename the BCM4350*.hcd firmware file as BCM.hcd and see if that works.

roadrunner2, 2017-09-25:
I built and tested 4.14-rc2, and pushed as the [hci_bcm-4.14](/roadrunner2/linux/tree/hci_bcm-4.14) branch. Other than resolving some conflicts, nothing really needed changing. @peterychuang, I suggest resetting your tree and cherry-picking the last 6 commits (v4.14-rc..hci_bcm-4.14). If that still doesn't work, then we'll need at least to see the dmesg output from when you run `sudo rmmod hci_uart; sudo modprobe hci_uart dyndbg=+p`.

@leifliddy Thanks for testing and feedback. Interesting that you don't see the DMA errors - which chipset exactly provides UART on your MB? (`lscpi -nn | grep UART`). Also not sure why `hciconfig` isn't working for you - needless to say it works fine for me.

Lastly, previous to your edit you were talking about firmware loading: I've been ignoring that so far, but I don't really know if there's anything critical there.

peterychuang, 2017-09-25:
thanks @roadrunner2, unfortunately it isn't working.

The strange thing in linux-next, both with and without the patches, is that in ```dmesg |  grep Bluetooth```, this is what I find:
```
[ 1282.752054] Bluetooth: HCI UART driver ver 2.3
[ 1282.752056] Bluetooth: HCI UART protocol H4 registered
[ 1282.752057] Bluetooth: HCI UART protocol BCSP registered
[ 1282.752057] Bluetooth: HCI UART protocol ATH3K registered
[ 1282.752058] Bluetooth: HCI UART protocol Three-wire (H5) registered
[ 1282.752088] Bluetooth: HCI UART protocol Intel registered
[ 1282.752088] Bluetooth: HCI UART protocol QCA registered
[ 1282.752089] Bluetooth: HCI UART protocol AG6XX registered
```

I would think something like ```Bluetooth: HCI UART protocol Broadcom registered``` should appear, but nope. In any case, that is perhaps why running ```sudo btattach --bredr /dev/ttyS0 -P bcm``` gives me this:

```
Attaching Primary controller to /dev/ttyS0
Switched line discipline from 0 to 15
Failed to set protocol: Protocol not supported
No controller attached
```

With the patches, ```dmesg``` with ```dyndbg=+p``` enabled on ```hci_uart``` now has something like these two lines:

```
[   56.017372] tty ffff880204c39800
[   56.017420] tty ffff880204c39800
```

If I run ```sudo btattach --bredr /dev/ttyS0``` instead (i.e. without ```-P bcm```, these additional lines appear:

```
[   65.490535] tty ffff88023dccf800

[   65.490546] hu ffff8802616e5600

[   65.491031] hci0 ffff880204d42000
[   65.491040] hci0: type 1 len 3
[   65.491042] hu ffff8802616e5600 skb ffff88020ca3d700

[   65.577815] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   65.577819] Bluetooth: BNEP filters: protocol multicast
[   65.577825] Bluetooth: BNEP socket layer initialized
```

Perhaps I have unknowingly done something wrong when I compile the kernel, or perhaps the 2017 machines are slightly different. I haven't got time to investigate it yet.

christophgysin, 2017-09-25:
@peterychuang Maybe you are missing `CONFIG_BT_HCIUART_BCM=y`?

peterychuang, 2017-09-25:
@christophgysin hm... funny you mention that... I've never turned off that option off, but somehow the option's gone missing in the compiled kernel, and I haven't quite figured out how to turn that option back on as I tried to recompile...

leifliddy, 2017-09-25:
@roadrunner2  I didn't realize the bluetooth controller was working after applying your patches, and I suspected the firmware not loading was the culprit.

The windows inf file for this device shows that BCM2E7C corresponds to BCM4350 and is presumably indicating the no firmware needs to be flashed, so I think we're good. 
%Bus.DeviceDesc%=BluePinole2E7C,    ACPI\BCM2E7C        ; BCM4350 no flash

here's the UART controller info:
[root@mac ~]# lspci -nn | grep UART
00:19.0 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO UART Controller #2 [8086:9d66] (rev 21)
00:1e.0 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 [8086:9d27] (rev 21)

l1k, 2017-09-26:
FWIW, the schematics of the MB8,1 indicate that the meaning of the host wake and device wake pins is swapped in firmware, which seems to imply that Apple is using a custom firmware. So I'm not sure if flashing a new firmware is recommendable or even possible.

Dunedan, 2017-09-26:
> I've never turned off that option off, but somehow the option's gone missing in the compiled kernel

Same for me. I checked the configs and all configs prior to 4.14 contained `CONFIG_BT_HCIUART_BCM`. The one for 4.14 didn't, even though I used the one from 4.13 as a base.

Apparently `CONFIG_BT_HCIUART_BCM` got a [new dependency on `CONFIG_BT_HCIUART_SERDEV` in 4.14](https://github.com/torvalds/linux/commit/33cd149e767be9afbab9fcd3d5165a2de62313c8) which has a dependency on `CONFIG_SERIAL_DEV_BUS` which wasn't enabled in my config.

Dunedan, 2017-10-06:
So, I finally got to try it out properly with Linux `4.14-rc3` on my MacBookPro13,2 and it works like a charm with KDEs Bluetooth userland tools. :+1:

roadrunner2, 2017-10-08:
So I guess we have confirmation now that this works on all MBP13,* models. And most likely therefore on all MBP14,* models too. Good. Thanks for testing.

risen, 2017-10-10:
I'm trying to test on a MacBookPro14,1 but I think I missed some kconfig setting.

No ttySx device seems to be registered: so I don't get the `dw-apb-uart.2: ttyS4 at MMIO 0xc182b000 (irq = 21, base_baud = 115200) is a 16550A` message in my kernel log.

I tried following the `btattach-bcm-service.sh` script's logic (which isn't available on Debian), but there's no tty file appearing in `/sys/devices/pci0000:00/0000:00:1e.0/dw-apb-uart.2/`

Here's my kernel config: [config-4.14.0-rc4-mbp-amd64.txt](https://github.com/Dunedan/mbp-2016-linux/files/1372598/config-4.14.0-rc4-mbp-amd64.txt)

Does anyone have a hint?

roadrunner2, 2017-10-14:
@risen Not having a serial-tty device makes me suspect a missing driver. I looked at your config and see that several `SERIAL_8250_*` are not set but which I have enabled in my config (these are the drivers for the 16550A device), including what I think might be the crucial one, the `SERIAL_8250_LPSS` driver. Try enabling that.

risen, 2017-10-17:
Thanks @roadrunner2 – setting CONFIG_SERIAL_8250_DW=y (and maybe  CONFIG_SERIAL_8250_LPSS=y too) did the trick.

I can confirm that bluetooth more or less seems to work now: scanning finds the devices in the neighbourhood, pairing works etc.

However, I'm still getting these:
```
Oct 17 16:38:57 pino kernel: bt_err_ratelimited: 17 callbacks suppressed
Oct 17 16:38:57 pino kernel: Bluetooth: Unknown advertising packet type: 0x13
Oct 17 16:38:57 pino kernel: Bluetooth: Unknown advertising packet type: 0x10
Oct 17 16:38:57 pino kernel: Bluetooth: Unknown advertising packet type: 0x20
Oct 17 16:38:57 pino kernel: Bluetooth: Unknown advertising packet type: 0x24
```
A2DP sound skips horribly, HSP sound is distorted etc. DUN seems to not be able to connect either. I'm not sure if it's a problem higher up the stack; but I'll report back if I figure it out.

leifliddy, 2017-10-17:
It's definitely a bit flaky. A2DP audio from headphones (MDR-1000X) skips horribly when playing youtube videos --while sound from local video/audio files is fine. A2DP audio from bluetooth speaker (SRS-X11) works well across the board. Bluetooth mouse (Razer Orochi) stops responding when it's been idle for a few minutes --have to disable + re-enable bluetooth controller to get it to reconnect. 

I'm seeing similar errors on my system:
[ 3339.966021] Bluetooth: Unknown advertising packet type: 0x10
[ 4557.202412] Bluetooth: Unknown advertising packet type: 0x20
[ 4557.202723] Bluetooth: Unknown advertising packet type: 0x24
[ 5030.960145] Bluetooth: Unknown advertising packet type: 0x21
[ 5045.276959] Bluetooth: Unknown advertising packet type: 0x11

roadrunner2, 2017-10-18:
@risen Not sure if this is the issue in your case, but make sure the gnome bluetooth-settings window is not open when you play audio etc - I think it's doing excessive scanning or something, and it makes definitely makes audio skip.

Regarding unknown advertising packets, I think those are basically harmless (from a quick perusal of the code) - I'd guess they're from a newer BT spec. It might be interesting to know which devices are sending, though (none of my devices send anything like that, but then again all of them are several years old).

@leifliddy I tried an Apple mouse, and don't see the issue you mention - at worst, after several minutes of inaction, I've seen it take a fraction of a second before the movements register (i.e. the first few movements are lost). So I guess this is device specific.

christophgysin, 2017-11-07:
Try:
https://github.com/roadrunner2/linux/tree/hci_bcm
https://github.com/roadrunner2/linux/tree/hci_bcm-4.13
https://github.com/roadrunner2/linux/tree/hci_bcm-4.14

roadrunner2, 2017-11-08:
@omnibrain Apologies for the broken links - fixed now.

risen, 2017-11-17:
Purely FYI, it works with just:
```
CONFIG_SERIAL_8250=y
CONFIG_SERIAL_8250_DW=y
```
So this one doesn't seem to make a difference:
```
# CONFIG_SERIAL_8250_LPSS is not set
```
The skipping audio was indeed most likely due to gnome-bluetooth doing excessive scanning.

Thanks ^_^

Maximus-, 2017-12-21:
Has anyone been able to verify this working on macbook8,1? I've tried most everything, but still get ```[  497.056067] Bluetooth: hci0 command 0x0c03 tx timeout
[  505.312000] Bluetooth: hci0: BCM: Reset failed (-110)
```

I'm using the newest patches from @roadrunner2 

As well, with btmgmt, I'm getting ```[mgmt]# power on
Set Powered for hci0 failed with status 0x11 (Invalid Index)
```

The device appears to be correct in /sys/class/bluetooth/hci0 as well.

roadrunner2, 2017-12-22:
@Maximus- for completeness sake: do you have the all of the following kernel config enabled: `CONFIG_SERIAL_8250_LPSS`, `CONFIG_BT_HCIUART_BCM`, `CONFIG_BT_HCIUART_SERDEV`, `CONFIG_SERIAL_DEV_BUS`, `CONFIG_SERIAL_DEV_CTRL_TTYPORT`,  and `CONFIG_SERIAL_8250_DW`? 

There may well be something missing for 8,1 - see e.g. [l1k's comment](#issuecomment-322065634) above about a mux.

Maximus-, 2017-12-22:
Yep! they're all enable in my config. Hmm

l1k, 2017-12-23:
@Maximus- could you apply commit l1k/linux@5e30ec7ead67 on top of v4.15-rc4 (or Linus' current tree or linux-next) and post the resulting dmesg in a gist or something? Thanks!

Maximus-, 2017-12-23:
Applied the patch on tags/v4.15-rc4, ensured the same config flags were enabled that @roadrunner2 specified, and built/installed. Here's the dmesg. (Also now the ttyS0 device doesn't show)

[dm.txt](https://github.com/Dunedan/mbp-2016-linux/files/1584051/dm.txt)

l1k, 2017-12-23:
@Maximus-:

Here's the problem:

```
[    0.361650] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
[    0.362633] 0000:00:15.5: ttyS0 at MMIO 0xc1819000 (irq = 21, base_baud = 2764800) is a 16550A
[    0.362769] serial serial0-0: controller busy
[    0.362815] serial serial0-0: failure adding ACPI serdev device. status -16
[    0.362860] serial serial0: tty port ttyS0 registered
```

And the "controller busy" message is coming from <a href="http://elixir.free-electrons.com/linux/latest/source/drivers/tty/serdev/core.c#L66">serdev_device_add()</a> in case there are multiple devices below the UART. Which is precisely the case on the MB8,1 because of the debug serial port on the SSD I mentioned earlier. serdev currently doesn't support multiple devices below the same UART. It doesn't know how to handle that, noone's implemented a GPIO-controlled mux to switch between multiple serdev slaves yet. What you could try is patching the DSDT to remove:

```
             Scope (\_SB.PCI0.URT0)
            {
                Device (SSDC)
                {
                    Name (_CID, "apple-uart-ssdc")  // _CID: Compatible ID
...
                }
            }
```

Try if that helps and post the dmesg output again.

That you're no longer seeing a ttyS0 char device is normal, you also no longer need to run btattach, it all happens automatically.

Some other observations, with 4.11+ you apparently no longer need `intremap=nosid` on the command line and with 4.14+ you no longer need to patch the DSDT for the SPI topcase. I'm also wondering if `pci=
nocrs` is really necessary.

Maximus-, 2017-12-23:
Good job! it works! 
[dm.txt](https://github.com/Dunedan/mbp-2016-linux/files/1584193/dm.txt)
Removed intremap=nosid, moved to a stock DSDT table, then applied your patch, and everything works. Bluetooth audio is /pretty/ laggy, but I'm hoping that's not your fault. Also there was a stack trace in my dmesg, so maybe an Oops somewhere. Attached new dmesg.

l1k, 2017-12-23:
@Maximus-:

Okay, if you apply commit l1k/linux@9d8147938ff5 on top of what you currently have, does the WARN splat go away? This doesn't seem to be a Mac-specific issue, it's just that you happen to not have `CONFIG_GPIOLIB` enabled and this patch should enable it.

I've amended the commit message of the other patch with a paragraph explainig that the DSDT needs to be patched on the MB8,1. Pushed the whole thing to the <a href="https://github.com/l1k/linux/commits/hci_bcm_v1">hci_bcm_v1</a> branch again.

roadrunner2, 2017-12-24:
@l1k I rebased your `hci_bcm_v1` branch on v4.15.0-rc5 and verified it works fine on MBP13,3. LGTM. Thanks!

@Maximus- regarding lag: make sure the gnome bluetooth settings window is not open - that causes a lot of stutter and slowdown.

roadrunner2, 2017-12-24:
Oh, I did just notice an interesting stack trace in dmesg (but haven't investigated if this is serious or not):
```
=============================
WARNING: suspicious RCU usage
4.15.0-rc5+ #13 Tainted: G           OE  
-----------------------------
kernel/rcu/sync.c:69 suspicious rcu_sync_is_idle() usage!

other info that might help us debug this: 

rcu_scheduler_active = 2, debug_locks = 1
2 locks held by kworker/u17:5/535:
 #0:  ((wq_completion)"%s"hdev->name#2){+.+.}, at: [<00000000d7ef3d12>] process_one_work+0x1c3/0x6b0
 #1:  ((work_completion)(&hdev->cmd_work)){+.+.}, at: [<00000000d7ef3d12>] process_one_work+0x1c3/0x6b0

stack backtrace:
CPU: 4 PID: 535 Comm: kworker/u17:5 Tainted: G           OE    4.15.0-rc5+ #13
Hardware name: Apple Inc. MacBookPro13,3/Mac-A5C67F76ED83108C, BIOS MBP133.88Z.0226.B23.1704201604 04/20/2017
Workqueue: hci0 hci_cmd_work [bluetooth]
Call Trace:
 dump_stack+0x85/0xbf
 hci_uart_tx_wakeup+0x2c/0x130 [hci_uart]
 hci_uart_send_frame+0x2e/0x60 [hci_uart]
 hci_send_frame+0x6b/0xe0 [bluetooth]
 hci_cmd_work+0x79/0x110 [bluetooth]
 process_one_work+0x243/0x6b0
 worker_thread+0x3a/0x390
 ? process_one_work+0x6b0/0x6b0
 kthread+0x11f/0x140
 ? kthread_create_worker_on_cpu+0x70/0x70
 ret_from_fork+0x24/0x30
INFO: trying to register non-static key.
the code is fine but needs lockdep annotation.
turning off the locking correctness validator.
CPU: 4 PID: 535 Comm: kworker/u17:5 Tainted: G           OE    4.15.0-rc5+ #13
Hardware name: Apple Inc. MacBookPro13,3/Mac-A5C67F76ED83108C, BIOS MBP133.88Z.0226.B23.1704201604 04/20/2017
Workqueue: hci0 hci_cmd_work [bluetooth]
Call Trace:
 dump_stack+0x85/0xbf
 register_lock_class+0x17b/0x580
 __lock_acquire+0xd1/0x1340
 ? hci_uart_send_frame+0x2e/0x60 [hci_uart]
 ? trace_hardirqs_on_thunk+0x1a/0x1c
 ? retint_kernel+0x10/0x10
 ? lock_acquire+0x9f/0x1f0
 lock_acquire+0x9f/0x1f0
 ? hci_uart_send_frame+0x2e/0x60 [hci_uart]
 ? dump_stack+0xb4/0xbf
 hci_uart_tx_wakeup+0x69/0x130 [hci_uart]
 ? hci_uart_send_frame+0x2e/0x60 [hci_uart]
 hci_uart_send_frame+0x2e/0x60 [hci_uart]
 hci_send_frame+0x6b/0xe0 [bluetooth]
 hci_cmd_work+0x79/0x110 [bluetooth]
 process_one_work+0x243/0x6b0
 worker_thread+0x3a/0x390
 ? process_one_work+0x6b0/0x6b0
 kthread+0x11f/0x140
 ? kthread_create_worker_on_cpu+0x70/0x70
 ret_from_fork+0x24/0x30
```

cyrusmg, 2017-12-24:
I have tested Evolveo SportLife and Sony MDR-ZX770BN headphones on MBP13,1 running Arch Linux with patched 14.1 kernel. I can hear occasional break with a crack in the headphones and sound is more and more delayed from the video in mplayer and Youtube Videos.

I am not running any applet or anything special. I have paired the headphones using `bluetoothctl` and then turned scanning off.

Maximus-, 2017-12-26:
The stuttering seems to even itself out after enough time playing... well, ok. 

Also, will enable GPIOLIB and try. : )

l1k, 2017-12-26:
@Maximus- The GPIOLIB should be enabled automatically with https://github.com/l1k/linux/commit/9d8147938ff5 and you shouldn't need to enable it manually.

In a separate step you could also try enabling CONFIG_LPC_ICH and CONFIG_GPIO_ICH. This should in theory give you access to GPIO 36. But be careful and backup your SSD before fiddling with the pin, it's unclear how it will react if you talk to it over the debug port.

BTW, for the folks with an MBP13,3 and similar models, there's a device called `apple-uart-soc-als` attached to URT2, could be the adaptive light sensor, someone will probably have to reverse-engineer the protocol and write a hwmon or iio driver. There's also an `apple-uart-soc` device below URT1, no idea what this could be.

Maximus-, 2017-12-26:
I have my own config I'm using, which is why I didn't just pull yours. But yeah, now the stack trace is gone. I'll considering playing with LPC_ICH and GPIO_ICH when I get some free time and don't need my machine in a working state. 👍

l1k, 2017-12-26:
<blockquote>
BTW, for the folks with an MBP13,3 and similar models, there's a device called apple-uart-soc-als attached to URT2, could be the adaptive light sensor, someone will probably have to reverse-engineer the protocol and write a hwmon or iio driver. There's also an apple-uart-soc device below URT1, no idea what this could be.
</blockquote>

Update: On the MBP14,2 they've changed the names to `apple-uart-ssdc` (same name as the SSD debug port on MacBook8,1) and `apple-uart-als`. The baudrate of the latter changed from 115200 to 1250000. The baudrate of the former stayed at 115200. Both devices are missing on the MBP14,1.

l1k, 2017-12-31:
I've pushed a new iteration of the Bluetooth patches to the <a href="https://github.com/l1k/linux/commits/hci_bcm_v2">hci_bcm_v2</a> branch. Could someone test this please?

I've realized that I had a bug in v1 wherein `bcm_gpio_set_power()` would toggle the device wake pin on non-Macs but not on Macs. This is fixed now. I've also incorporated all the feedback I've received for v1.

[249 more replies]