Dunedan/mbp-2016-linux #35: Display randomly going down (ubuntu on mbp13,3)

chaosbunker, 2017-09-11:
Several months ago I gave up, because this issue made it impossible to work on something without having to worry that suddenly the screen goes down, forcing a hard reset (color of the screen is not always the same)...

I can't seem to find anyone else having this problem.

I have tried with ubuntu 17.10, 17.04, 16.04 and it happens with all of them

the 17.10 i tried yesterday is using kernel 4.12.0
today i used isorespin to put kernel v4.12.12 on ubuntu 16.04
i am about to try debian 9 out of curiosity and have fedora 26 ready too but did not manage to boot the usb stick yet.

right now I got everything working fine (except the stuff that doesn't work yet) with ubuntu 16.04 and kernel v4.12.12 and am trying to figure out why the screen goes kaput sometimes. unfortunately i did not recognize a pattern and can't reproduce.

it seems like it happens faster when i tamper with the display/screen in any way (like turn brightness down, do something in settings>display), but i am not sure. right now it has been running for 30minutes without issues, but i did not turn the brightness down (and am wearing sunglasses instead :-P man, it's so damn bright!)

Does anyone know how i could troubleshoot this?

Dunedan, 2017-09-12:
So your display just randomly shuts off and won't get on again unless you do a reboot? I haven't had such an issue yet.
Can it be related to the software you're using?

chaosbunker, 2017-09-12:
Yes, exactly. It goes white, purple or some color and freezes. I doubt it is software related. It sometimes happens right after booting up, sometimes it happens after 30 min of usage or more. The only software I had been using then is Firefox and terminal..

Dunedan, 2017-09-13:
Do the logs (syslog, Xorg.log, etc.) contain something useful?
Is they machine still running normally (except for the screen of course)?

chaosbunker, 2017-09-13:
Last time it happened i checked Xorg.log and syslog (actually all the logs that were modified around the time it happened) but didn't find anything useful. Possible I overlooked somehting, so i will check again next time

Apart from this everything is running normally

marcwimmer, 2017-09-13:
hi dont know if it helps or so: i own a macbook 12 without cooling fan. when it gets hot i get memory errors in xorg.conf log and x-server restarts. this is repeatable. is your computer hot or so when it happens? i have "CPU2: Package temperature above threshold, cpu clock throttled (total events = 6891)
" then.

and then any app like chromium or so:
[11676.817262] chromium-browse: page allocation failure: order:0, mode:0x14210d2(GFP_HIGHUSER|__GFP_NORETRY|__GFP_RECLAIMABLE), nodemask=(null)
[11676.817269] chromium-browse cpuset=/ mems_allowed=0
[11676.817274] CPU: 3 PID: 2815 Comm: chromium-browse Tainted: G        W  OE   4.12.0-041200rc5-generic #201706112031
[11676.817275] Hardware name: Apple Inc. MacBook9,1/Mac-9AE82516C7C6B903, BIOS MB91.88Z.0154.B17.1702161815 02/16/2017
[11676.817275] Call Trace:
[11676.817288]  dump_stack+0x63/0x8d
[11676.817291]  warn_alloc+0x114/0x1c0
[11676.817294]  ? find_next_bit+0xb/0x10
[11676.817296]  __alloc_pages_slowpath+0xe07/0xe10
[11676.817298]  __alloc_pages_nodemask+0x233/0x250
[11676.817301]  alloc_pages_vma+0xab/0x250
[11676.817303]  shmem_alloc_page+0x70/0xc0
[11676.817306]  ? kfree+0x14a/0x160
[11676.817308]  ? kfree+0x14a/0x160

roadrunner2, 2017-09-14:
@chaosbunker This does sound like a h/w issue - have you tried running the diagnostics? (hold down the 'd' key right after power-on)

chaosbunker, 2017-09-14:
@roadrunner2 it does seem like that, not that you point it out, since it doesn't happen to anyone else with the same setup and the logs are 'quiet'. Unfortunately the diagnostics test didn't show anything. personally i don't think the test seems too basic and is not comprehensive enough :(

If anyone knows a better tool that would be good to test the hardware, let me know

chaosbunker, 2017-09-14:
I reset NVRAM and SMC in hope that it had something to do with that .. will be testing and report if it still happens. too bad no issues show in macOS, then I could easily go to the apple store for a replacement

Dunedan, 2017-09-15:
As you're using a MacBookPro13,3, which features two different graphics chips: Which one are you using? Maybe the one you're using is the AMD GPU, which is somehow broken and you don't observe these problems with MacOS as it defaults to the Intel GPU. Just thinking out loud.

chaosbunker, 2017-09-21:
Well, I do have the `AMD Radeon Pro 460`, what do you mean with "somewhat broken" - in general (including on macOS) or on Linux? How could I test if it is broken on macOS?

Dunedan, 2017-09-21:
In addition to your "AMD Radeon Pro 460" you also have an integrated "Intel HD Graphics 530". Maybe you're experiencing the problems with one of them under Linux while the other one is used by macOS when you don't experience problems.

Let's assume you use the AMD GPU under Linux and macOS defaults to the Intel GPU, then you'd have to force macOS to use the AMD GPU (e.g. by running a graphics-heavy application like a game which makes macOS switch from the Intel card to the AMD card) to see if you're encountering your Linux problems there as well.

roadrunner2, 2017-09-22:
FYI, you can use the "Activity Monitor" to see if any app is causing the [discrete GPU to be used](https://support.apple.com/en-us/HT202053)

chaosbunker, 2017-09-22:
Hm, I do use graphics-heavy applications frequently on macOS without ever experiencing any issues (for example right now I have a video editing app running and an image-editing software, which according to Activity Monitor are using the discrete card). I think in the very beginning I stress tested the GPU with heaven benchmark and it ran just fine...

Are you guys actively using linux right now on your machines and if so, which distros? @roadrunner2, in your gist it sounds like you are using the macbookpro13,3 and fedora .. i guess you did not upgrade the discrete card to the 460?

christophgysin, 2017-09-23:
I'm running Arch on a mbp13,1. So only Intel GPU for me, but no issues at all.

Dunedan, 2017-09-23:
I'm using Debian/unstable on a MacBookPro13,2 also just with integrated Intel GPU. Aside from the known limitations it's working pretty well.

roadrunner2, 2017-09-23:
@chaosbunker I indeed have the basic card (the 450, though it shows up as "Radeon RX 460" under Linux lspci - go figure). Furthermore, I've been running exclusively on the integrated Intel GPU for several months now, as that uses substantially less power (at least 4W) and I don't need the performance of the discrete GPU. (I need to write up the steps to do this, but they're still a bit hokey). So I can't really comment on the stability of the discrete GPU under Linux, except to say that I never experienced any issues while I was using it, even when trying out a game (0 A.D.) to verify the graphics acceleration was working (all that was with the `amdgpu` driver).

chaosbunker, 2017-09-25:
I can't believe I again opted for a MacBook Pro with discrete GPU, after all the issues I had with my 2011 MBP ...

On linux I also don't need the discrete card, I am really looking forward to your writeup on how to deactivate it under linux (if I understood correctly).. @roadrunner2

Dunedan, 2018-05-27:
The writeup you're looking for is available now: https://github.com/Dunedan/mbp-2016-linux/issues/6#issuecomment-387526390

Aside from that, are there any news regarding this issue?

rwuwon, 2018-07-09:
I believe it's not only the 2011 MBP line that's had discrete graphics card issues - my understanding is there's not a lot of cooling with MBPs...

@chaosbunker if you haven't already, might it be worth seeing how it handles a graphics card benchmark/stress test under macOS? For what it's worth, similar to @christophgysin, my 13,1 has no apparent graphics issues - Fedora 28 with 4.16.16 in my case.

Dunedan, 2018-09-20:
Closing this, because of missing feedback and because it's likely to be a hardware-related problem.

closed 2018-09-20 by Dunedan

vjpr, 2022-07-18:
@capitaineplanete Late 2016 MBP. Exact same issue. It's a problem with the dGPU driver.

Solution is here: https://github.com/Dunedan/mbp-2016-linux/issues/6#issuecomment-387526390

It's not as complicated as it looks. It just disables dGPU and enables Intel.

capitaineplanete, 2022-07-19:
@vjpr yeah that's what I'm doing but it's far from perfect. i.e. I can't watch any videos due to video/audio synchronisation issues, my tty has the wrong resolution which can be frustrating at times, and I'm can't completely switch off the dGPU to save on battery.

vjpr, 2022-07-19:
@capitaineplanete Were you able to get an external monitor connected? I'm tryig usb-c to display port and its not working.

capitaineplanete, 2022-07-22:
@vjpr I don't have an external monitor to test I'm afraid