Dunedan/mbp-2016-linux #36: USB hotplugging

peterychuang, 2017-09-15:
Hi, I'm wondering if anyone finds hotplugging USB devices (e.g. flash drive) kind of hit-or-miss.

Dunedan, 2017-09-15:
I had no problems at all with USB so far, but I remember somebody having problems with certain flash drives/USB-C adapters.

roadrunner2, 2017-09-17:
I haven't run into any issues either, but then again I've only tried a handful of devices (a CD/DVD player, a thumbdrive, and external ssd, and two wifi sticks) - all of them via Apple USB-C adapter.

risen, 2017-09-21:
I had issues, but it turned out to be my el-cheapo USB-C to USB type-A adapter that was broken. Eventually it stopped working completely.

peterychuang, 2017-09-22:
Thanks everyone.
I have most problems with this [Sandisk flash drive (32GB)](https://www.amazon.com/SanDisk-Ultra-Drive-Type-C-SDDDC2-032G-G46/dp/B01EZ0X23W). It could be indeed the problem with the flash drives. Will investigate further.

Dunedan, 2017-11-20:
Is this still an issue or can we close it?

peterychuang, 2017-11-21:
Nothing has changed for me, I still have problems with hotplugging USB flash drives from time to time.

Dunedan, 2018-02-04:
@peterychuang Any news on this? Can you provide any kind of logs for such problems?

peterychuang, 2018-02-05:
The last time I checked, it seemed all the same to me. Unfortunately I couldn't see anything on logs when it failed.

I've switched back to macOS for now, since I'm too busy to be fiddling with the machine, so I wouldn't be able to test anything for the time being...

Dunedan, 2018-02-11:
All right. As you're the only user experiencing this problem so far and won't be able to investigate further, I'm closing this issue for now. Let's reopen it, when that changes.

closed 2018-02-11 by Dunedan