Dunedan/mbp-2016-linux #39: Thanks for your efforts!!!

n1ckh3ll, 2017-10-14:
Hi, I have been a mac user for a long time and an linux enthusiast maybe long before that. I have tried quite a lot of distros since redhat 6... Unfortunately I never have achieved a pleasant usage of GNU-linux. Always there is something that does not work well (I mean well) not just works...(andf that after countless hours of forum investigating)  But anyway I still try... I suppose its a love-hate relation I have with Linux. 

Now I have been following this github for some months and finally decided to give a try on my 13.1 with antergos-KDE (kernel 13.x.x). I love how smooth it runs, even streamed videos. 

I think I will change to a full Linux experience (single boot) when audio and suspend work.

I, of course will not ask for a Time Frame but I would like to ask.... Do you think that in the future will be easy to make everything work for us, users? Like follow a tutorial... You know Im not afraid of tinkering but things like compile kernel.... well are a little too much for me and some instructions sound like klingon..
Thanks for your time and efforts.
Keep going!!!!!

christophgysin, 2017-10-14:
Instructions to patch the kernel and packages will vary depending on what distribution you are using. Hopefully all patches will eventually make it into the mainline kernel and all packages will be available in all distributions. Then, everything will just work out of the box. Until then, you will need to stick with macOS, or learn some of that Klingon ;)

n1ckh3ll, 2017-10-14:
Hahhaha. I´m old and busy but I try to do my best.

Dunedan, 2017-10-14:
> Hopefully all patches will eventually make it into the mainline kernel and all packages will be available in all distributions.

Well, we need a lot of hope for that. E.g. look at older MacBook Pros which still need an out-of-tree driver for the webcam.

Aside from that, it'll be still quite a long journey till we reach a satisfying point. The MacBook Pro 2016 is now a year old and look where we currently are with support. It's getting better, for sure, but it takes its time.

And let's not start talking about features like graphical Touch Bar functionality, working Touch ID, advanced Touchpad functionality (force press, customizable vibration feedback, ...) or even the advanced power-saving stuff Apple is doing (e.g. I just learned that they do dynamic switching of the RAM voltage to save some power).

But it's quite fun, so let's see how far we get. 😄

monouser7dig, 2017-10-14:
>  or even the advanced power-saving stuff Apple is doing (e.g. I just learned that they do dynamic switching of the RAM voltage to save some power)

I'd think linux saves some power on its own just because it has less cloud and fuzz?

once wifi would start working I'd try on the 2016 but laptop without wifi is kind of pointless 😆

peterychuang, 2017-10-14:
@n1ckh3ll Since you are using Antergos, compiling kernel should be relatively straightforward using Arch Build System, as you may have already discovered. It may help if you have a more powerful desktop machine to build the kernel though, as that saves a lot of time.

My biggest pain point remains to be [NVMe timing out at boot](https://github.com/Dunedan/mbp-2016-linux/issues/9#issuecomment-295609909). I wonder if there's any progress on figuring out the problem. I'm also looking forward to [the patches that power down thunderbolt for better battery life](https://github.com/Dunedan/mbp-2016-linux/issues/24#issuecomment-310952415).

Otherwise, I am okay with how Linux runs on my machine so far.

n1ckh3ll, 2017-10-14:
@peterychuang thanks,  I´m on it.

peterychuang, 2017-10-14:
@n1ckh3ll I've just created [a repository for the pkgbuild](https://github.com/peterychuang/linux-macbook-pro-pkgbuild) I use. If you are interested, you can clone the repository, run ```makepkg```, and see how that goes. Be warned though, it's based on linux-next, so it may not be 100% safe.

christophgysin, 2017-10-14:
@peterychuang Maybe you could upload it to the [AUR](https://aur.archlinux.org)? There is already [linux-macbook](https://aur.archlinux.org/packages/linux-macbook), so maybe call it linux-macbook13?

peterychuang, 2017-10-15:
@christophgysin I'll look into it later.

n1ckh3ll, 2017-10-16:
Hi, OMG thanks !!!!!

n1ckh3ll, 2017-10-17:
@peterychuang I have tried to compile but got errors. I am sure they are my fault as while compiling I got 3 questions:
something about cherry trail support (choosed Y)
something about btrfs ref verify tool compiled in (choosed Y)
something about kernel unwinder (choosed opt 1)

And got :
`make[4]: *** [scripts/Makefile.build:314: drivers/gpu/drm/i915/i915_gem.o] Error 1
make[3]: *** [scripts/Makefile.build:572: drivers/gpu/drm/i915] Error 2
make[2]: *** [scripts/Makefile.build:572: drivers/gpu/drm] Error 2
make[1]: *** [scripts/Makefile.build:572: drivers/gpu] Error 2
make: *** [Makefile:1020: drivers] Error 2
==> ERROR: Se produjo un fallo en build().
`

peterychuang, 2017-10-17:
@n1ckh3ll The linux-next tree for Oct 16 seems to have a few problems. I haven't been able to build it either. What I am going to do is to wait for the next tree. Alternatively, you can try to built the Oct 13 tree.

Regarding the kernel options, cherry trail should be unnecessary, I'd think, though it probably won't make much difference if you enable it. The one regarding btrfs should be irrelevant unless you are using btrfs. For unwinder, I chose [ORC](https://www.phoronix.com/scan.php?page=news_item&px=ORC-Unwinder-Linux-4.14), but the machine should work with either one of them.

peterychuang, 2017-10-18:
@n1ckh3ll the 20171017 tree is buildable again. I've updated my pkgbuild repository for the new tree.

n1ckh3ll, 2017-10-18:
I have it working!!! Thanks!!!!

Dunedan, 2018-02-04:
Closing, because the issue seems to be resolved.

closed 2018-02-04 by Dunedan