Dunedan/mbp-2016-linux #45: How to use EFI to boot the linux kernel with FV2.

drocsid, 2017-11-24:
I'm new to installing Linux on the macbook and was trying to install the rEFInd boot manager.

I went through the procedure "rEFInd with SIP Enabled" on a 2016 MBP 13,3 running Sierra, documented here: http://www.rodsbooks.com/refind/sip.html#sip_enabled. 

I have to use FV2 encryption on the device, but can disable it temporarily.

Note: This was necessary to see the macOS Volume from the terminal in the recovery in order to install rEFInd.

After restarting I got the rEFInd splash screen. But then after rebooting macOS , I'm prompted to reenable FV2 to login to macOS; due to constraints. After FV2 was re enabled and I rebooted, the rEFInd splash screen went away. I assume that possibly FV2 re-wrote the boot loader.

Then I think I must choose one of the Disabling SIP options. 

Then I'd like to know what EFI booting procedure is working well for others who are using FV2 from a similar setup.

Do you have any recommendations on which approach(es) to try next RE: Disabling SIP options or otherwise?

drocsid, 2018-01-02:
@Dunedan are you running with FV2 enabled? Have you tested this or can you give some input?

Dunedan, 2018-01-03:
I enabled FileVault even before I installed Linux on the MacBook Pro. From my experience it doesn't matter if you have FileVault enabled or not, but I'm not using rEFInd, just plain old grub and EFI configured to load Linux as default and I boot macOS by pressing the Option-key during startup.

drocsid, 2018-01-09:
Did you need to bless the grub EFI to have it show up as a boot option? Do you have a reference on configuring EFI? I tried blessing my grub EFI file with set boot but lost the macOS option, and get the grub command line, but not configured to boot a kernel..

drocsid, 2018-01-11:
systemdboot seemed to work like this for me. Here is a reference in case someone else is facing this issue.

https://wiki.archlinux.org/index.php/mac#Using_the_native_Apple_bootloader_with_systemd-boot_.28Recommended.29

Dunedan, 2018-01-11:
> Did you need to bless the grub EFI to have it show up as a boot option?

No, blessing isn't necessary since ages anymore. Using the standard EFI boot manager is sufficient.

> Do you have a reference on configuring EFI?

Not really, as Debian does that all out-of-the-box, but in the end it's pretty similar to the Arch Linux instructions you posted.

Dunedan, 2018-02-04:
Closing as this issue seems to be resolved.

closed 2018-02-04 by Dunedan