Dunedan/mbp-2016-linux #47: mbp 13.2 wifi

monouser7dig, 2017-12-23:
https://bugzilla.kernel.org/show_bug.cgi?id=193121#c9

> The country code X0 is apparently stored in the device itself. Now this country needs to be supported by the firmware we released to linux-firmware and my guess is that it does not.

As far as I see this man has fixed several of these issues already, does someone know if resolving these 13.2 wifi issues is on the horizon as well? Like is there something that could motivate him to do this or is it a gamble?

Dunedan, 2017-12-23:
All infos we got are in the Bugzilla thread. Sadly we didn't get any further response from Broadcom. 😞 
My guess is that it would be pretty easy for them to fix, but isn't on their list of priorities right now.

monouser7dig, 2017-12-26:
> My guess is that it would be pretty easy for them to fix, but isn't on their list of priorities right now.

I'd interpret his statement the same 

> didn't get any further response from Broadcom. 😞

okay 😞😞

monouser7dig, 2018-04-23:
From the bugzilla link:

```
Hello guys. I was in correspondence with Arend some time ago and got this answer (and permission to post his reply to this forum):

======= snip ==========

The firmware update is not an easy task as it updates regulatory data in firmware it means a full verification cycle is required to assure the power levels stay within regulatory limits. This is always a hard sell for us as there is no return of investment for the engineering costs involved. Hence we release fullmac firmware that has been verified in running projects. Sadly the 43602 in the MacBook does not use a fullmac driver architecture so there is no fullmac firmware available. I have forwarded your request, but no high hopes we can resolve this.

Regards,
Arend

======= snip ===========

I decided to sell my Macbook Pro and started using a Dell laptop instead with Linux working on it like a charm.
```

tumra86, 2018-04-25:
I'm running ubuntu 16.04 LTS on a MacBook Pro 2017 13,2 (without Touch Bar), kernel 4.16 . The wifi is working perfectly, I only remastered the kernel with the keyboard drivers.
I'm a noob when it comes to tinkering the kernel, but if I can get any information from my system that might help just let me know.

monouser7dig, 2018-04-25:
so my WiFi Card reports as:
```
AirPort Extreme  (0x14E4, 0x157)
Broadcom BCM43xx 1.0 (7.77.37.29.1a7)
```
I've got a 13.2 with touchbar actually 

Maybe I'll have to try to boot from external media and see how it works with 4.16.

Dunedan, 2018-04-25:
@tumra86 Wifi in the models without Touch Bar isn't an issue at all and is known to work fine. It's only a problem for the Touch Bar models.

kljsandjb, 2018-04-26:
Great !

monouser7dig, 2018-05-08:
If there was a USB C Wifi stick I'd be ok with that as well.
The sort of stick that barely protrudes.
I have not found such a thing for USB C yet though.

craigerrington, 2018-05-10:
https://bugzilla.kernel.org/show_bug.cgi?id=193121#c25

(there's a diff in the thread)

> Andrey Arapov 2018-05-10 21:56:39 UTC

> Ok, I am really tired of using a usb wifi dongle, so I was digging a bit deeper and here is what I've got.
> 
> Changing ccode=X0 to 0 and regrev=15 to 0 in the firmware seems like improving things a little and makes the connection more usable:
> 
> - instantly connecting to the WiFi;
> - link quality=70/70 (Max!);
> - connections is stable, no disconnects;
> - I've got stable 50 Mbps download and 20 Mbps upload (my ISP allows 200/20 Max as per my contract);
> - The latency gets higher as I am downloading things, but otherwise it is as low as it should be;
> - still no 5GHz ... not sure why, maybe some hints what can be changed in the FW ? I've tried many things already...

jboyens, 2018-05-11:
I can confirm that this "patch" enables the 14,3 to connect and hold a WiFi signal.

jboyens, 2018-05-11:
Make no mistake. It's slow. REALLY slow. With the Canakit dongle (not a great one by any means) I can get 50mbit up/50mbit down. With brcmfmac I get 3mbit up/20mbit down. A step in the right direction...

ClashTheBunny, 2018-05-14:
It's a step in the right direction, but still generally not great:
```
--- www.google.com ping statistics ---
564 packets transmitted, 408 received, +1 duplicates, 27% packet loss, time 567305ms
rtt min/avg/max/mdev = 12.525/1973.478/11679.836/3011.685 ms, pipe 12
```

m0oz, 2018-05-15:
> I can confirm that this "patch" enables the 14,3 to connect and hold a signal.
Could you elaborate on how to change regrev and ccode in the brcmfmac firmware?
I am not really experienced on kernel and firmware topics. It would be great to get some advice ;)

roadrunner2, 2018-05-25:
@m0oz Personally I use `bvi` (binary vi) for this sort of thing,

jboyens, 2018-06-21:
Messing with this again tonight. I get a strong (full signal) 2.4Ghz connection with ccode=XV regrev=1. I haven't yet tried in a less forgiving environment, but it seems more solid.

I'm continuing to iterate to figure out if I can get 5Ghz to kick over.

I checked out the Windows drivers, but no joy (they have no values for this bits encoded)

Dunedan, 2018-07-01:
So I finally tried out the binary patching workarounds and I'm pretty disappointed. While it's definitely a step into the right direction, it's still not usable for me. Packet loss and general connection speed are awful.

To make it easier to try it out yourself, here are a step by step instructions to patch the firmware:
* Make a backup of your original firmware to a new location (as it'll be modified directly):
```
cp -a /lib/firmware/brcm/brcmfmac43602-pcie.bin /lib/firmware/brcm/brcmfmac43602-pcie.bin.orig
```
* install `radare2`
* create a `rapatch`-patch file with the following content:
```
0x0007c82f "wx 3000"
0x0007c839 "wx 3000"
```
* Patch the firmware using the following command:
```
radare2 -w -q -P your_patch_file brcmfmac43602-pcie.bin
```
* Verify the file got correctly patched, by showing the diff between the original file and your patched file:
```
radiff2 -x brcmfmac43602-pcie.bin.orig brcmfmac43602-pcie.bin
```
* Reboot your MacBook Pro

tim-digitised, 2018-07-19:
I just wanted to say I'm really happy to see that there's some groundwork going, but still quite unhappy that it's so unstable. Thanks everyone for the work on this and I'll be watching this thread hoping to see some updates in the future.

miki2o, 2018-10-07:
I was thinking. Why don't we make a bounty? The dev that actually patches at least wifi or wifi and audio will get this.

heywoodlh, 2018-10-07:
If that bounty was set up as a pool that community members could contribute to (i.e. contribute $0-20 or as much as each person could contribute) I would definitely contribute to that pool.

Dunedan, 2018-10-12:
I'm afraid a bounty wouldn't change the situation, as this problem needs to get fixed by Broadcom and as already stated by a Broadcom employee, there is little hope they consider that important enough. I'm pretty certain a few hundred bucks wouldn't change their mind, especially as the cost of engineering and re-certification would be probably still way higher.

I see only one possibility to get out of this situation right now: Get enough people to tell Broadcom that this is an important problem for them. It's quite a difference if a few dozen people have this problem or a few thousand. That of course would require that many affected people and I'm afraid there are fewer than that wanting to use these MacBook Pro's with Linux. :disappointed:

libsystem-ethan, 2018-10-12:
Forgive me if these is questions are "silly" (I don't have the best understanding of the certification and regulation that applies to WiFi drivers/firmware), but if Broadcom needed to update the firmware for some other reasons (and thus require re-certification) is it likely that they would try to get a fix for this bug included with the (hypothetical) new firmware? Or does this bug itself require a costly fix, on top of recertification costs? Also, would Broadcom be able to provide a fix but with an added disclaimer that it hasn't undergone as rigorous testing an certification than their other firmware - or could this lead to legal issues due to not having ensured that the power levels are within the regulatory limits?

Dunedan, 2018-10-13:
There are no silly questions, but some questions can probably be only answered by Broadcom.

>  but if Broadcom needed to update the firmware for some other reasons (and thus require re-certification) is it likely that they would try to get a fix for this bug included with the (hypothetical) new firmware?

I would hope so, but no indea if they really would.

> Or does this bug itself require a costly fix, on top of recertification costs?

Nobody outside of Broadcom knows for sure, but my gut feeling is, that the bug fix itself would be straight forward.

> Also, would Broadcom be able to provide a fix but with an added disclaimer that it hasn't undergone as rigorous testing an certification than their other firmware - or could this lead to legal issues due to not having ensured that the power levels are within the regulatory limits?

Legal issues is the key here: Such certifications are necessary to be allowed to ship such products (or in this case firmware version) to market.

I guess if Broadcom wouldn't have to re-certificate firmware version after changes, they would've already fixed this bug and shipped a fixed firmware version.

tomaytotomato, 2018-10-23:
So I installed Ubuntu 18.04 last week and Wifi was working on my 2017 13inch (2 thunderbolt) fine.

I now upgraded to 18.10 with linux kernel `4.18.0-10-generic` and the wifi adapter is not recognized.

Any ideas?

jboyens, 2018-11-15:
Kernel 4.19 (supposedly) added support for a bunch of 802.11ac chipsets that I'm digging into to find a small-ish wifi adapter solution. Some of the ones I've seen and currently use are a little big and cumbersome. Either way, expect to live the "dongle-life" to the extreme if you want to run Linux on this laptop. There exist 0 USB-C WiFi adapters, so it's even bigger with a USB-A -> USB-C dongle.

tomaytotomato, 2018-11-15:
> Hey guys, thanks for all your work and research into this. I'm new to
> Linux and I just dual booted my 2016 MacBook Pro (I believe 13,2) with
> Touchbar with a Ubuntu Kernel 4.15.0. Initially, my WiFi was only showing
> 4-5 networks. Apparently my MBP was only detecting 2,4 GHz channels 1-11.
> After change ccode=0 and regrev=0 as suggested above, I am now seeing
> channels 12 and 13 also enabled. Channel 14 still says disabled. My home
> WiFi is now working as I was able to change my WiFi AP settings to only use
> a channel within the range my MBP currently supports. But this is obviously
> not a final solution as I can't ask everyone to change their WiFi setting
> just for me. I'll definitely be watching this thread with the hope of
> future updates. In the meantime, does anyone have any other suggestions?

Hi Fabian, try upgrading your kernel to a newer version? That worked for me
on a 13,1 MacBook

Dunedan, 2018-11-17:
@tomaytotomato The MacBook Pros without Touch Bar (like your 13,1) contain a different Wifi chip than the models with Touch Bar. The one in the models without Touch Bar works fine, the one in the models without Touch Bar doesn't. That was already stated in this issue, as well as in the README, so please read past comments to avoid such unnecessary comments.

Dunedan, 2018-12-31:
I just stumbled over [`nexmon`](https://github.com/seemoo-lab/nexmon):

> The C-based Firmware Patching Framework for Broadcom/Cypress WiFi Chips that enables Monitor Mode, Frame Injection and much more 

Could be an approach to produce functioning firmware patches, which fix all issue we're facing. In case somebody wants to get his hands dirty.

marc-git, 2019-07-30:
I personally find the `iwconfig wlp3s0 txpower 10`  (or even txpower 1) works very well and results in a stable connection (14,3) 
Anyone know how to do that automatically at startup ? I'm on debian 10, and so impressed with the work you guys are doing I'm considering of switching to debian as my primary system. Typing the command at startup each time is frustrating though.  I passed a command in crontab as root but this is either not being run (crontab service is running) with 
`@reboot iwconfig wlp3s0 txpower 10`
but it isn't working. Maybe wifi settings are overridden by users? Anyone have ideas or already solved this? I guess a Debian solution will apply to Ubuntu too.

marc-git, 2019-07-30:
P.S.: It is my experience that the bluetooth and wifi interfere. I disabled mine at startup using :
`gsettings set org.blueman.plugins.powermanager auto-power-on false`

which allows me to re-enable using blueman.

jgmbenoit, 2019-07-30:
For Debian (and Ubuntu) I will look for a ifupdown script (in /etc/network).

liberostelios, 2019-11-09:
According to the latest posts from the original discussion, just a few days a go someone managed to set the correct parameters for the driver in order to make WiFi 5GHz work with mbp 13.2 and 13.3:

1. Download [this](https://bugzilla.kernel.org/attachment.cgi?id=285753) file.
2. Open it with a text editor and set the mac address line with your own card's address.
3. Place it in /lib/firmware/brcm.

I tried it and it works with my 13.3 in a Manjaro live USB. Haven't installed it yet, so I can't provide a review for long-term use, though.

miki2o, 2019-11-09:
> According to the latest posts from the original discussion, just a few days a go someone managed to set the correct parameters for the driver in order to make WiFi 5GHz work with mbp 13.2 and 13.3: [...]

Trying right now on Ubuntu 19.10 installed on internal SSD.
Will update with news soon.

marc-git, 2019-11-09:
> According to the latest posts from the original discussion, 
Ref to comment: https://bugzilla.kernel.org/show_bug.cgi?id=193121#c52

marc-git, 2019-11-09:
OK I am posting this over my 5GHz connection. That was quick. I am on MBP14,3
Note the mac address in the firmware is 00:90:4c:0d:f4:3e for all devices. And this works fine for me.

Dunedan, 2019-11-10:
@marc-git As I already mentioned in bugzilla that's not yet a complete solution. I did some tests and connecting to nearby 2.4Ghz access points (no matter if they use encryption or not) simply doesn't work for me with the latest changes applied. 5Ghz works fine so far though.

marc-git, 2019-11-10:
> @marc-git As I already mentioned in bugzilla that's not yet a complete solution. I did some tests and connecting to nearby 2.4Ghz access points (no matter if they use encryption or not) simply doesn't work for me with the latest changes applied. 5Ghz works fine so far though.

This is working fine for me. I had been doing `sudo iwconfig wlp3s0 txpower 10` for a while and could only get 2.4GHz networks, but that was quite reliable for me. 
Now with this fix I can do both? Just switched from the 5GHz to 2.4GHz network while typing.  Very stable on my 14,3.

miki2o, 2019-11-10:
> This is working fine for me. I had been doing `sudo iwconfig wlp3s0 txpower 10` for a while and could only get 2.4GHz networks, but that was quite reliable for me.
> Now with this fix I can do both? [...]

Same. Can switch flawlessy between 2.4 and 5 GHz.

devalanche, 2019-12-07:
dunno if this would help or make it easier for anyone but putting/overwriting those files in /lib/firmware/brcm/ works for my 14,3 mbp on ubuntu 19.10 kernel 5.3 (even on a customiso)
[brcmfmac43602-pcie.zip](https://github.com/Dunedan/mbp-2016-linux/files/3935013/brcmfmac43602-pcie.zip)

MrAsimov, 2019-12-28:
> > According to the latest posts from the original discussion, just a few days a go someone managed to set the correct parameters for the driver in order to make WiFi 5GHz work with mbp 13.2 and 13.3:
> > 
> > 1. Download [this](https://bugzilla.kernel.org/attachment.cgi?id=285753) file.
> > 2. Open it with a text editor and set the mac address line with your own card's address.
> > 3. Place it in /lib/firmware/brcm.
> > 
> > I tried it and it works with my 13.3 in a Manjaro live USB. Haven't installed it yet, so I can't provide a review for long-term use, though.
> 
> Trying right now on Ubuntu 19.10 installed on internal SSD.
> Will update with news soon.
> 
> EDIT: Can confirm that it works!!!
> 5GHz network connected and it's also fast(and stable)!(Not as good as on macOS tho, on macOS i got ≃ 900+Mbps, on Ubuntu i got ≃ 300+Mbps).
> Tested on 13.3, Ubuntu 19.10 stock kernel.
> 
> @Dunedan, you can further test this if needed and eventually add to the README.

I can confirm this worked as well for me with MBP 13.2 with Linux Mint 19.3 and kernel version 5.4.6.

```
System:    Host: asimov-MacBookPro Kernel: 5.4.6-050406-generic x86_64 bits: 64 compiler: gcc 
           v: 9.2.1 Desktop: Cinnamon 4.4.5 wm: muffin dm: LightDM Distro: Linux Mint 19.3 Tricia 
           base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: Apple product: MacBookPro13,3 v: 1.0 serial: <filter> Chassis: 
           type: 9 v: Mac-A5C67F76ED83108C serial: <filter> 
           Mobo: Apple model: Mac-A5C67F76ED83108C v: MacBookPro13,3 serial: <filter> UEFI: Apple 
           v: 263.0.0.0.0 date: 10/30/2019 
Battery:   ID-1: BAT0 charge: 52.8 Wh condition: 55.1/76.7 Wh (72%) volts: 12.6/11.5 
           model: SMP bq20z451 serial: N/A status: Full 
           Device-1: hidpp_battery_0 model: Logitech Wireless Keyboard serial: <filter> 
           charge: 55% status: Discharging 
CPU:       Topology: Quad Core model: Intel Core i7-6920HQ bits: 64 type: MT MCP arch: Skylake-S 
           rev: 3 L2 cache: 8192 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 46398 
           Speed: 900 MHz min/max: 800/3800 MHz Core speeds (MHz): 1: 897 2: 900 3: 900 4: 895 
           5: 900 6: 900 7: 900 8: 899 
Graphics:  Device-1: AMD Baffin [Radeon RX 460/560D / Pro 450/455/460/555/560] vendor: Apple 
           driver: amdgpu v: kernel bus ID: 01:00.0 chip ID: 1002:67ef 
           Display: x11 server: X.Org 1.20.4 driver: amdgpu,ati unloaded: fbdev,modesetting,vesa 
           resolution: 1920x1080~60Hz, 1920x1080~60Hz 
           OpenGL: renderer: AMD Radeon RX Graphics v: 4.6.13564 Core Profile Context 
           compat-v: 4.6.13564 direct render: Yes 
Audio:     Device-1: Intel 100 Series/C230 Series Family HD Audio driver: snd_hda_intel v: kernel 
           bus ID: 00:1f.3 chip ID: 8086:a170 
           Device-2: AMD driver: snd_hda_intel v: kernel bus ID: 01:00.1 chip ID: 1002:aae0 
           Sound Server: ALSA v: k5.4.6-050406-generic 
Network:   Device-1: Broadcom and subsidiaries BCM43602 802.11ac Wireless LAN SoC vendor: Apple 
           driver: brcmfmac v: kernel port: 3000 bus ID: 03:00.0 chip ID: 14e4:43ba 
           IF: wlp3s0 state: up mac: <filter> 
Drives:    Local Storage: total: 465.92 GiB used: 12.40 GiB (2.7%) 
           ID-1: /dev/nvme0n1 vendor: Apple model: SSD SM0512L size: 465.92 GiB speed: 31.6 Gb/s 
           lanes: 4 serial: <filter> 
Partition: ID-1: / size: 457.11 GiB used: 12.40 GiB (2.7%) fs: ext4 dev: /dev/nvme0n1p2 
Sensors:   System Temperatures: cpu: 59.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/amdgpu-pro-local.list 
           1: deb [ trusted=yes ] file: /var/opt/amdgpu-pro-local/ ./
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //ftp.acc.umu.se/mirror/linuxmint.com/packages tricia main upstream import backport
           2: deb http: //archive.ubuntu.com/ubuntu bionic main restricted universe multiverse
           3: deb http: //archive.ubuntu.com/ubuntu bionic-updates main restricted universe multiverse
           4: deb http: //archive.ubuntu.com/ubuntu bionic-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ bionic partner
Info:      Processes: 275 Uptime: 3m Memory: 15.54 GiB used: 1.57 GiB (10.1%) Init: systemd v: 237 
           runlevel: 5 Compilers: gcc: 7.4.0 alt: 7 Client: Unknown python3.6 client inxi: 3.0.32 
```

tawheedmanzoor, 2020-02-09:
> According to the latest posts from the original discussion, just a few days a go someone managed to set the correct parameters for the driver in order to make WiFi 5GHz work with mbp 13.2 and [...]

I can confirm working on 13,2 Ubuntu 19.10.

For me 2.4 GHz worked even without this file if I used it with open networks (even Xfinity in the US).

svenper, 2020-04-09:
> 1. Download [this](https://bugzilla.kernel.org/attachment.cgi?id=285753) file.
> 2. Open it with a text editor and set the mac address line with your own card's address.
> 3. Place it in /lib/firmware/brcm.

I have a `MacBookPro12,1` with <code>BCM43602 [14e4:43ba] <b>(rev 01)</b></code>. Should I expect this to enable 5 GHz for me?

cristianmiranda, 2020-04-29:
Hi, I have a MacBook Pro mid 2016 with touchbar, internet works fine after applying this fix: https://bugzilla.kernel.org/show_bug.cgi?id=193121#c62
The only problem I currently have is that (at least with Ubuntu 20.04 and Manjaro 19), wifi won't reconnect to previous connected networks after sleep (suspend). Have anyone experienced the same issue?

MrAsimov, 2020-06-08:
@Dunedan you should update the README.md section for WIFI both 2.4/5 GHz wifi is working for me on MacBookPro13,3.

Dunedan, 2020-06-09:
Feel free to open a PR with the necessary updates to the README. I never got Wifi to work reliably on both 2.4Ghz and 5Ghz bands, but as I don't own a recent MacBook Pro anymore, I can't try out things possible solutions anymore.

marc-git, 2020-07-23:
> Feel free to open a PR with the necessary updates to the README. I never got Wifi to work reliably on both 2.4Ghz and 5Ghz bands, but as I don't own a recent MacBook Pro anymore, I can't try out things possible solutions anymore.

What is your machine? 14,3?
Anybody have a 14,3 for which the bugzilla workaround isn't a fix? Or can people mention if they have a BCM43602 that isn't working with this workaround and what their setup is.

andyholst, 2020-07-23:
@christophgysin I believe I lost WIFI connection after a suspension session. Never tried it thoroughly. I am ready to open up a PR if the WIFI driver patch for MBP models 13,2; 13,3; 14,2 and 14,3 is working for both 2.4/5 Ghz for the BCM43602 802.11ac Wireless LAN SoC (rev 02) device.

Just have to test it properly locally first.

andyholst, 2020-07-23:
For 2 Ghz band to work I have to set the tx power to 10 dBm `iwconfig wlp3s0 txpower 10dBm` with default brcmfmac43602-pcie.bin firmware. For it to work with 5 Ghz band, I have to use the brcmfmac43602-pcie.txt file and override the firmware binary configuration https://bugzilla.kernel.org/show_bug.cgi?id=193121#c52 with proper mac address set for it to work.

marc-git, 2020-07-24:
Yes would be great to hear from more people first. At the moment we could say that some people with these machines have working wifi using those workarounds. Would be great to hear from somebody for whom it isn't working and what their config is.

andyholst, 2020-07-24:
What is most interesting for me is that if you set the region code (ccode) globaly to 00 (with bvi editor) in the binary /lib/firmware/brcm/brcmfmac43602-pcie.bin file without the txt file included, then 2 Ghz band work (might have to reduce the tx power to increase stability).

So if you copy the 2 Ghz band property values from the binary to the brcmfmac43602-pcie.txt you should get working 2.4/5 Ghz band connections.

andyholst, 2020-07-24:
Ok, so I have been fiddling with 2 configuration brcmfmac43602-pcie files; one from the https://bugzilla.kernel.org/show_bug.cgi?id=193121#c52 and one from the https://bugzilla.kernel.org/show_bug.cgi?id=193121#c64 NVRAM captured via VFIO trace of Windows broadcom 43602 driver.

I only kept the swctrlmap_5g and swctrlmapext_5g key values and added those to the generated NVRAM VFIO trace broadcom 43602 Windows txt file and removed some keys values that caused 5 Ghz band to stop working.

With this modification I got an increase in download speed from 20/40 Mbit/s to 70/80 Mbit/s.

[brcmfmac43602-pcie.txt](https://github.com/Dunedan/mbp-2016-linux/files/4974264/brcmfmac43602-pcie.txt)

andyholst, 2020-07-25:
I did a few more modifications to the brcmfmac43602-pcie.txt configuration file to be more similar to the original NVRAM VFIO trace file for the brcmfmac 43602 driver generated by Sonal Santan https://github.com/sonals/macbookpro14-3/blob/master/doc/MMIOTrace.rst

andyholst, 2020-07-25:
It might help to interpreter the board flags for the broadcom device at https://android.googlesource.com/kernel/common.git/+/bcmdhd-3.10/drivers/net/wireless/bcmdhd/include/bcmdevs.h while testing the brcmfmac43602-pcie.txt configuration.

andyholst, 2020-07-26:
Ok, I am going at the right direction, I did compare the brcmfmac43602-pcie.txt configuration generated by the NVRAM VFIO trace file and compared the boardflags2 and boardflags3 hex values with the original brcmfmac43602-pcie.bin file when none configuration is set. I can confirm that the boardflags3 is the one 'deciding' if 2.4Ghz band and 5Ghz band is going to be used.

The original boardflags 2/3 hex values from the original brcmfmac43602-pcie.bin without configuration is:

boardflags2=0x00000002
boardflags3=0x00000003

with this configuration along with the NVRAM VFIO trace file data, the 2 Ghz band connection works just fine during boot or through `sudo rmmod brcmfmac && sudo modprobe brcmfmac` command.

By setting the boardflags3 to 0x00000303 the 5 Ghz band works very well with 60/80 Mbits internet download connection.

The last piece if possible is to figure out what combination needs to be set for boardflags3 to enable 2.4/5 Ghz band at the same time.

[brcmfmac43602-pcie.txt](https://github.com/Dunedan/mbp-2016-linux/files/4977023/brcmfmac43602-pcie.txt)

andyholst, 2020-07-26:
I did some further tweaking to get better 2Ghz band download speed up to 9/20 Mbit/s.

[brcmfmac43602-pcie.txt](https://github.com/Dunedan/mbp-2016-linux/files/4977048/brcmfmac43602-pcie.txt)

marc-git, 2020-08-06:
OK I also realised that my 2.4GHz was not working simultaneously with 5GHz now.  
@andyholst did you get any further?

andyholst, 2020-08-07:
@marc-git  can you confirm that brcmfmac43602-pcie.txt I added did it for you? Or did you test another configuration? I have not tested the combinations for a while, but the last 2 weeks I can confirm that the 2 Ghz band works just fine if you are within 20 meters, I usually use the 5 Ghz band.

andyholst, 2020-08-09:
I have created bash script to brute force the boardflags3 configuration to see if there is a combination to allow connecting to 2.4 Ghz or 5 Ghz band without changing the boardflags3 configurations https://github.com/andyholst/BCM43602_macbookpro

Feel free to fork it and add optimizations, I will try to brute force the configurations next week.

marc-git, 2020-08-11:
> @marc-git can you confirm that brcmfmac43602-pcie.txt I added did it for you? Or did you test another configuration? I have not tested the combinations for a while, but the last 2 weeks I can confirm that the 2 Ghz band works just fine if you are within 20 meters with no so much interference such as many routers or thick walls (if you have strong router signal), I usually use the 5 Ghz band though.

Your file works fine but I can't tell what frequency I am on. Will have to wait until I get a 2.4GHz network only.

andyholst, 2020-08-18:
Interesting, boardflags3=0x00000003 seems to work both for 2.4Ghz and 5Ghz band. I believe I have had to much interference before with to many 2.4 Ghz band router in range... I can easily switch between 2.4 Ghz and 5 Ghz band with this boardflags3 configuration.

marc-git, 2021-08-11:
Hello again. @andyholst 
what boardflags3 setting are you using nowadays?

[7 more replies]