Dunedan/mbp-2016-linux #48: Driver Manager Linux Mint 18.3 Cinnamon 64-bit on Macbook Pro 14,3

coenboomkamp, 2017-12-27:
Hi everyone,

Anyone else having problems with the driver manager?

It shows an update for my Intel(R) Core(TM) i7-7700HQ CPU @ 2.80Ghz
I can do a update, to version 3.20170707.1~ubuntu16.04.0

I update it, and reboot, when i start the driver manager again, it shows the same update.
Because of this i am unable to get into the actual driver manager itself..

Cheers

Dunedan, 2017-12-29:
That sounds like a Linux distribution specific problem, not related to the hardware itself. Have you already tried updating the `intel-microcode` package using `apt-get`?

coenboomkamp, 2018-01-01:
Yes I did, does not work ether. It says in the terminal that it will finish the installation on the next boot. But that never happens. Something goes wrong there during the boot I guess. I am running on dual boot with REFInd.

Dunedan, 2018-01-01:
You probably want to fill a bug report with your Linux distribution. That's a sounds like a more promising place for such kind of problems.

Dunedan, 2018-02-04:
Closing as it seems to be a Linux distribution-specific problem.

closed 2018-02-04 by Dunedan