Dunedan/mbp-2016-linux #52: Touchbar not working on Macbook14,3

aapo, 2018-02-10:
I can modprobe appletb and then it is shown by lsmod. dmesg is not showing anything. 

sudo mbp-2016-linux/touchbar 1
causes: sending usb-control: Broken pipe

roadrunner2, 2018-02-11:
This would probably best be reported over at [my repo](https://github.com/roadrunner2/macbook12-spi-driver) where I am working on this driver.

On to your issue: According to the `lsusb` output you posted in your pull request #51 the iBridge device appears to be missing; there's an Apple device there, but with a different USB id: 0x1281 instead of 0x8600. And it's labelling itself as a "Apple Mobile Device [Recovery Mode]", doesn't appear to have the same subdevices as the iBridge, and none of the subdevices are a HID. In fact the only HID that shows up is your external Dell keyboard. So I'm thinking thinking that somehow the iBridge chip isn't being activated.

Could you boot into MacOS and take a look at the USB device tree there and see if an iBridge device (with id 05ac:8600) is present?

Dunedan, 2018-02-11:
> Could you boot into MacOS and take a look at the USB device tree there and see if an iBridge device (with id 05ac:8600) is present?

It'd also be good to know if the TouchBar works properly with macOS.

roadrunner2, 2018-02-11:
Looking at the info in pull #53, that looks like what is expected. Coupled with the suspicious naming of the new, unknown device on your system, "Apple Mobile Device [Recovery Mode]", I'm wondering how you booted Linux. Are you using Apple's EFI bootmanager, or have you installed a different one (such as rEFInd)? And then is it starting grub, or something else?

aapo, 2018-02-12:
I never booted MacOS (only saw language selection when first time opened the lid). Installed xubuntu from the USB-stick, using whole disk. So I'm not sure what happens during boot.

roadrunner2, 2018-02-12:
Ah, interesting: that could imply you blew away the EFI system partition (this is a separate partition, usually the first, formatted as FAT, and 200MB in size) and that could well be the issue here, because that contains firmware that I'm guessing is loaded at startup. Take a look at your partitions and see if the EFI System Partition (ESP) is present, and if so whether it contains a `efi/apple/` directory or not.

aapo, 2018-02-12:
I have grub-efi-amd64 installed and I can see GRUB-menu, so yes, grub is in use.
Grub boots: vmlinuz-4.13.0.32-generig.efi.signed

I have
efivars on /sys/firmware/efi/efivars
and **/dev/nvme0n1p1 on /boot/efi** . And there is no apple, only ubuntu and BOOT.

roadrunner2, 2018-02-12:
So then it looks like I may have been right: the original ESP appears to have been erased and replaced with one created by the installer, and which is therefore missing the firmware etc that the EFI loader expects. My best guess is that you'll need to figure out how to restore that (maybe somebody else with the same model and an intact original ESP, (maybe @ClashTheBunny?) can tar up their `efi/apple` directory and send it to you).

Just to be clear: I don't know for sure that the missing `efi/apple` contents are your problem, as I don't know how either Apple's EFI loader works in particular nor do I know how the iBridge device is initialized. But this is my best guess given the above info.

Dunedan, 2018-02-12:
I'm not sure if Apple is doing that through EFI or from macOS. My guess would be that macOS flashes the iBridge-firmware on first boot. But definitely an interesting topic.

aapo, 2018-02-12:
-Backup your own files (they will be erased!).
-Shut down Macbook
-Press and hold: Option-Command-R [ https://support.apple.com/en-us/HT204904 ]
-Press Power key
  -> Boots to the "Upgrade to the latest macOS that is compatible with your Mac"
-Release keys now.
-Select wifi (give password). Wait ~5 minutes.
-Select language
-Select "Disk Utility"
-Internal, Erase, Format="Mac OS Extended (Journaled)".
-Quit Disk Utility
-Select "Reinstall macOS", Continue, Continue, Agree, Agree
-Select the (only) disk. (If you don't see your disk, use Disk utility.) Install. 
-Wait ~15 minutes.

...To be continued

aapo, 2018-02-12:
Boot from xubuntu-18.04 (beta) -usb-stick:
-Press and hold 'option'
-Press power key
-Wait bootmenu
-Release 'option'.
-Select usb-stick.
-"Test xubuntu" 

Backup EFI:
-Connect second usb, and mount it to (e.g) /tmp/mp
-mkdir /tmp/efi ; sudo mount /dev/nvme0n1p1 /tmp/efi -oro
-sudo cp -a /tmp/efi /tmp/mp
-sudo umount /tmp/mp
-sudo umount /tmp/efi

-Install Xubuntu using "Erase disk and install Xubuntu"

After starting fresh installed system:
find /boot/efi

    /boot/efi/
    /boot/efi/EFI
    /boot/efi/EFI/ubuntu
    /boot/efi/EFI/ubuntu/fw
    /boot/efi/EFI/ubuntu/fwupx64.efi
    /boot/efi/EFI/ubuntu/grubx64.efi
    /boot/efi/EFI/ubuntu/grub.cfg
    /boot/efi/EFI/ubuntu/shimx64.efi
    /boot/efi/EFI/ubuntu/mmx64.efi
    /boot/efi/EFI/ubuntu/BOOTX64.CSV
    /boot/efi/EFI/BOOT
    /boot/efi/EFI/BOOT/BOOTX64.EFI
    /boot/efi/EFI/BOOT/fbx64.efi
 
 -> No apple stuff (Xubuntu installer indeed overwrites it).
sudo lsusb -vvv | grep -i ibridge
-> no hits.

----

sudo cp -a /media/user/backup/efi/EFI/APPLE /boot/efi/EFI/

So it looks:

    /boot/efi/
    /boot/efi/EFI
    /boot/efi/EFI/ubuntu
    /boot/efi/EFI/ubuntu/fw
    /boot/efi/EFI/ubuntu/fwupx64.efi
    /boot/efi/EFI/ubuntu/grubx64.efi
    /boot/efi/EFI/ubuntu/grub.cfg
    /boot/efi/EFI/ubuntu/shimx64.efi
    /boot/efi/EFI/ubuntu/mmx64.efi
    /boot/efi/EFI/ubuntu/BOOTX64.CSV
    /boot/efi/EFI/BOOT
    /boot/efi/EFI/BOOT/BOOTX64.EFI
    /boot/efi/EFI/BOOT/fbx64.efi
    /boot/efi/EFI/APPLE
    /boot/efi/EFI/APPLE/FIRMWARE
    /boot/efi/EFI/APPLE/FIRMWARE/MBP143_0169_B00.fd
    /boot/efi/EFI/APPLE/EXTENSIONS
    /boot/efi/EFI/APPLE/EXTENSIONS/Firmware.scap
    /boot/efi/EFI/APPLE/EMBEDDEDOS
    /boot/efi/EFI/APPLE/EMBEDDEDOS/combined.memboot
    /boot/efi/EFI/APPLE/EMBEDDEDOS/FDRData
    /boot/efi/EFI/APPLE/EMBEDDEDOS/version.plist

Reboot.
sudo lsusb -vvv | grep -i ibridge
-> iBridge found!

----
And after that 
**sudo insmod appletb.ko**
-> Touchbar works!

closed 2018-02-12 by aapo

roadrunner2, 2018-02-12:
Thanks for the details! Good to see it's working now.

Dunedan, 2018-02-12:
So, I did try out a bunch of things and it turns out @roadrunner2 was right: Without the `EFI/APPLE` directory on the ESP, the iBridge doesn't get initialized, the TouchBar doesn't work and the `Apple Mobile Device [Recovery Mode]` device is visible instead.

I'll add a note to the README stating that the ESP-portion of macOS is necessary for the TouchBar to work. A next step for us could be to figure out how to load the firmware to the iBridge, so we can do it independently from the files on the ESP.

Thanks @aapo for stumbling over that issue.

jas8522, 2022-01-27:
> I'll add a note to the README stating that the ESP-portion of macOS is necessary for the TouchBar to work. A next step for us could be to figure out how to load the firmware to the iBridge, so we can do it independently from the files on the ESP.

Any chance there's been any progress made on this? I'm in a super odd position where I'm booting from a USB-C attached SSD because the internal nvme controller has 'melted down' in that the entirety of the internal disk is read-only. Even in recovery mode (Command-R or Command-Option-R) I can only see the disk and read files. Similarly within Ubuntu, if I run badblocks on the disk it works great until I provide a write flag (volatile or not), at which point it hangs for a while and then the nvme controller drops from the system entirely. I believe it's also preventing the MacBook from reading from the EFI partition, as booting to Recovery Mode *also* renders the touchbar totally dead.

Since reading technically still works in Ubuntu, I was able to copy the drivers from the EFI partition on the internal drive in the manner described above, but I'm assuming because it's not on the internal disk, the Mac won't read them on boot.

And so, back to my question: any idea if I can force the kernel to read those drivers from my external drive rather than the internal, or am I SOL without replacing the logic board?