Dunedan/mbp-2016-linux #54: mbp14,3: ctrl+x not working in grub

aapo, 2018-03-04:
MacBookPro14,3
Ubuntu: 18.04
Grub: 2.02

In grub menu, press 'e' to edit selected entry. 
Help text says that "Press Ctrl-x or F10 to boot". But **pressing Ctrl-x produces letter "x".** This happens with built-in keyboard and with external usb-keyboard. Ctrl-c is working (and starts command line).
There are no F10 in in-built keyboard, because of touch bar, so this is harmful issue.

I tested ubuntu 18.04 on virtualbox and there ctrl-x is working as expected.

roadrunner2, 2018-03-04:
Ctrl-x not working (i.e. just producing an "x") appears to be common: e.g. it doesn't work when using my external usb keyboard either, and if you google for this behaviour/issue you'll find many complaints about it.

Because of the missing ESC and F10 keys when the touchbar is not enabled, a while ago I briefly looked into enabling the touchbar in grub. It would be fairly easy to do, except for a small detail: the iBridge requires an xhci driver, but grub currently only has ohci, uhci, and ehci drivers. Now, there was some work on getting an xhci driver working for grub (see https://lists.gnu.org/archive/html/grub-devel/2017-03/msg00012.html), but that appears to have stalled.

So for right now if you need to edit the command line in grub, you'll need to use an external keyboard :frowning_face:

Dunedan, 2018-10-02:
I dug into this issue a bit and to me it seems that CTRL+x is working fine on non-Apple hardware starting with GRUB 2.02~beta3, thanks to commit [9e5f701](https://salsa.debian.org/grub-team/grub/commit/9e5f70174ec960a0077f20bb74cb9f4da9b57e7b). Apparently before it was broken on some (or even all) UEFI systems.

That it doesn't work on Apple hardware for certain key combinations (even with external keyboards) could be caused by a bug in Apples EFI implementation.

I see two possible ways to proceed here:
1. Figure out why GRUB isn't able to handle some modifier key combinations and add a workaround to GRUB (if possible).
2. Add xhci- and Touch Bar-support to GRUB.

I believe 1. would be the way to go, as it'd not just fix this issue for the MacBook Pro 2016 and later, but for all Intel-based Macs and the GRUB developers might be reluctant including device specific drivers upstream as 2. would require.

Some references:
https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/722950
http://lists.ipxe.org/pipermail/ipxe-devel/2015-May/004230.html (and the other emails in this thread)

rustycl0ck, 2019-01-05:
A simple workaround for this problem (for those who are yet to buy an external keyboard) - 
Instead of trying to edit the command line and running it with `ctrl-x`, directly jump to the command line prompt with `ctrl-c` and execute the command directly. Following works for me for live booting - 
```
$ linux /casper/vmlinuz boot=casper quiet splash
$ initrd /casper/initrd.lz
$ boot
```
You can add the `nomodeset` or `toram` or any other parameters that you need to the `linux` kernel line.

inpefess, 2019-03-01:
@rustycl0ck that helped me to start Ubuntu live but a keyboard seems not working in Ubuntu anyway:)

YongxinJackGuo, 2020-09-06:
> @rustycl0ck that helped me to start Ubuntu live but a keyboard seems not working in Ubuntu anyway:)

My keyboard and mouse both not working after logging into Ubuntu live. Did you manage to fix it now?

Dunedan, 2020-09-06:
@YongxinJackGuo This repository isn't for distribution specific help. Aside from that this is a completely different problem than this issue is about and the reason is very likely that Ubuntu Live doesn't ship with `applespi` included. If that's not the case and you're certain that it's not just an Ubuntu-only problem, please open a separate issue for this problem.