Dunedan/mbp-2016-linux #55: mbp14,3: grub, visual glitch, missing arrows

aapo, 2018-03-04:
MacBookPro14,3
Ubuntu: 18.04
Grub: 2.02

In Grub menu there are missing characters: up-arrow and down-arrow.

This is only cosmetic issue, but might relate #54 
[image]

aapo, 2018-03-04:
Ubuntu 18.04 tested in virtualbox works as expected.
[image]

aapo, 2018-03-04:
Context in code: grub2-2.02/grub-core/normal/menu_text.c:170
 
```
     msg_translated = grub_xasprintf (_("Use the %C and %C keys to select which "
					 "entry is highlighted."),
				       GRUB_UNICODE_UPARROW,
				       GRUB_UNICODE_DOWNARROW);
```

Dunedan, 2018-03-04:
Works for me with Debian/unstable, but I also have a graphical grub menu enabled. I doubt that this is a hardware related problem.

aapo, 2018-03-13:
Fixed in Grub: 2.02-2ubuntu8

closed 2018-03-13 by aapo