Dunedan/mbp-2016-linux #56: wifi not working on mbp 14,1

pranavg189, 2018-04-26:
Hello,

I installed Manjaro on my mbp 14,1 but the 'brcmfmac' module fails to load for me with the following error:

`
brcmfmac: brcmf_chip_recognition: chip backplane type 15 is not supported
brcmfmac: brcmf_pcie_probe: failed 14e4:43a3
`

I tried installing all possible driver for the Broadcom BCM4350 wifi chipset given here:

https://wiki.archlinux.org/index.php/broadcom_wireless

Surprisingly, when I boot into the manjaro live usb, wifi work fine. But If I reload the 'brcmfmac' module, it again gives the above error.

Has anyone got the wifi working on mbp 14,1 ?

craigerrington, 2018-04-27:
Apologies, misread your model.

pranavg189, 2018-04-28:
> I'm on a MacBookPro14,1 running Ubuntu 17.10 and wifi worked out of the box, and I haven't experienced any issues at all (there are no dropouts, I can connect to 5GHz networks, etc).

Maybe it is an issue with 'brcmfmac' firmware shipped with Manjaro. Can anyone confirm the working of mbp 14,1 wifi on other distributions like Archlinux, Fedora etc ?

pranavg189, 2018-05-05:
Update: I just installed Ubuntu 18.04 on my mbp 14,1 and the wifi is working properly now along with the applespi (touchpad and keyboard) driver.

Another thing I noticed is that, If I login into "Ubuntu on Wayland" session, the wifi stops working as it used to in Manjaro and can not be restarted again. I had to reboot and select the normal "Ubuntu" session. Hence, I think logging into the Wayland session currently has some issue with the "brcmfmac" driver which disables the wifi.

Hope this helps someone installing linux on mbp 14,1 .

It would be great if some note is added in the README.md file about this. I will confirm again if this really is the problem. I will need confirmation from other users too if Wayland has issues with brcmfmac driver.

Dunedan, 2018-06-16:
As all MacBook Pro models with Touch Bar use the same wifi chip, let's keep the discussion in a single issue: #47 
Therefore closing this.

closed 2018-06-16 by Dunedan