Dunedan/mbp-2016-linux #57: How install keyboard and mouse fix for MacBook pro 2017 (Kali Linux)?

MOLOKOmilk, 2018-05-05:
Hi everyone! 
What commands should be used in the terminal in order to correct the operation of the keyboard and mouse? I do not understand myself.

MacBookPro14,3 , Kali Linux on flash usb (16 gb)

Dunedan, 2018-05-05:
I suggest you refer to the documentation of the `applespi` driver how to get it working: https://github.com/cb22/macbook12-spi-driver
There is also a separate gist explaining how to install the driver: https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#keyboardtouchpadtouchbar

MOLOKOmilk, 2018-05-05:
'First some extra packages: sudo dnf install git kernel-devel dkms' 
and i get this
root@kali:~# sudo dnf install git kernel-devel dkms
sudo: dnf: command not found

MOLOKOmilk, 2018-05-05:
can u add me in discord please LunarEvening #9864

MOLOKOmilk, 2018-05-06:
Can you write code here? Because of ignorance of the language and Linux, I can not figure this out.

Dunedan, 2018-05-18:
As you've already noticed, Linux doesn't work out of the box on the Macbook Pro 2016 & 2017 yet (at least not everything). While we're working on getting it supported properly, we're not there yet. So for now you'll need at least some basic understanding of Linux to get it working, as it's simply too much effort to provide step by step instructions for everybody who wants to get Linux running on these devices and our time is better spent on getting Linux to work better on them, as that benefits all users in the long run. Therefore I'm closing this issue for now.

closed 2018-05-18 by Dunedan