Dunedan/mbp-2016-linux #58: 2016 Macbook Pro not booting after power outage

vik748, 2018-05-07:
Hi,
I have a 2016 Macbook Pro which has been quite stable for a while now.  However, I let the power drain completely and now the Ubuntu installation is broken.  When I try to recovery mode I see that the message:
```bash
suspending console(s) (use no_console_suspend to debug)
```
I have tried using that flag in grub, but I have been unable to boot the system.  Additionally, I have tried nomodeset option as well.
Any ideas/suggestion that you can point me to will be very helpful.

Thanks,
Vik

ClashTheBunny, 2018-05-07:
Did you update your kernel since your last reboot?  Maybe choose an older
kernel version in GRUB?

Randall Mason
randall@mason.ch

roadrunner2, 2018-05-08:
Try booting with the `noresume` kernel parameter (edit the grub entry).

Dunedan, 2018-05-24:
@vik748 Did you manage to solve this issue?

shahvi, 2018-05-24:
Yes, basically I had to go back to a Kernel which didn't have this issue.  More details at 
https://askubuntu.com/questions/1033307/ubuntu-16-04-on-2016-macbook-fails-to-boot-after-power-failure-freezes-at-suspe/1033376#1033376

Dunedan, 2018-05-24:
Awesome. :+1:
Closing this now, as it got solved.

closed 2018-05-24 by Dunedan