Dunedan/mbp-2016-linux #59: Temperature issue

cyrusmg, 2018-05-12:
When I switch to Linux, the machine gets quite warm compared to MacOS (mainly left of the touchpad). The temperature reported via `sensors` is about 58 degrees Celsius. Does anyone else experience this ? Any ideas what to do to fix that ?

- Arch Linux
- MacbookPro13,1
- i3wm

roadrunner2, 2018-05-12:
I presume you've checked that nothing's keeping the cpu busy. Apart from that, you might want to take a look at `powertop`'s output - while I've found the power estimations for individual processes/devices to be fairly unreliable, check out the frequency and idle stats and verify things are really idling; also, look at the tunables and see if enabling power-management for everything helps.

cyrusmg, 2018-05-14:
Great, `powertop` seems to help. I get ~ 47 degrees with powertop tunables enabled. Thank you!

Dunedan, 2018-05-18:
Just a few days ago I used my MBP in direct sunlight for about an hour. Fans were running near maximum rpm, but temperatures were ok and it did work as expected. So I believe there is no general issue with cooling under Linux.

Dunedan, 2018-05-24:
As this seems to be resolved, I'm closing it now.
Feel free to reopen if such temperature issues appear again.

closed 2018-05-24 by Dunedan