Dunedan/mbp-2016-linux #60: tb3 Nvidia eGPU

matt-deboer, 2018-05-17:
macbookpro 14,3
ubuntu 18.04, linux 4.15.0-20-generic

Attempting to use an external 1080ti for CUDA, and `nvidia-smi` results in following kernel messages:

```
[Thu May 17 14:22:53 2018] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
                           NVRM: BAR1 is 0M @ 0x0 (PCI:0000:0a:00.0)
[Thu May 17 14:22:53 2018] NVRM: The system BIOS may have misconfigured your GPU.
[Thu May 17 14:22:53 2018] nvidia: probe of 0000:0a:00.0 failed with error -1
[Thu May 17 14:22:53 2018] NVRM: The NVIDIA probe routine failed for 1 device(s).
[Thu May 17 14:22:53 2018] NVRM: None of the NVIDIA graphics adapters were initialized!
```
any ideas where to start?

Note: Already tried `pci=nocrs` kernel param to no avail.

Dunedan, 2018-05-24:
Does it work using macOS in the same machine?

matt-deboer, 2018-05-24:
Yes (up to 10.13.3; Apple "enhanced" (read: broke) eGPU support in 10.13.4)

Looks like a similar issue:
https://askubuntu.com/questions/967223/running-into-pci-resource-allocation-issues-17-10-kernel-4-13-0-16-generic-64bit
--looks like his path forward was to enable PCI addressing above 4GB...but I'm unsure of how to approach that.

Dunedan, 2018-05-24:
I neither have an idea how to fix that, but I'd start trying out several pci-related boot flags, like `pci=nobios`, `pci=realloc=on`, `pci=realloc=off`, `pci=assign-busses`, ...

Also dynamically rescanning PCI devices could help: https://www.linuxquestions.org/questions/linux-hardware-18/issue-installing-a-external-gpu-device-888879/page3.html or https://bbs.archlinux.org/viewtopic.php?pid=1602551#p1602551

As there are people who seem to be able to use an eGPU with Windows through Bootcamp on a Mac (https://egpu.io/forums/pc-setup/egpu-plugplay-in-bootcamp/), which is unsupported by Apple, I'm optimistic that it's possible to get to work with Linux as well.

matt-deboer, 2018-06-11:
Thanks for the tips; I finally got a little time to play with this over the weekend.
I can confirm that manually removing and then rescanning the bridge device (to which the eGPU is attached) works (requires `pci=realloc` kernel param).

The runfile nvidia drivers must be used (ubuntu drivers bomb the kernel log with PKCS#7 signature errors).

The next hurdle (not necessarily a MBP issue) is that vga_switcheroo steps in and disables the discrete gpu whenever the eGPU is connected--I need a way to make it ignore specific devices' hotplug events...

So I'd consider this partially solved, but I'll report back with any more progress/findings.

Dunedan, 2018-06-11:
Awesome. Thanks for reporting back. :+1:
I'll leave this issue open for you to report your findings. Feel free to close it at any time, if you feel that there is nothing more to say. :smile:

jacooba, 2018-06-15:
Hi there, I could really use some help! 
I have Ubuntu 16.04 LTS installed on my MBP (15 inch, touchbar, late 2016 i believe). I am trying to get my gamebox aurous with a gtx 1080 working. I have it working on windows, but I need it to work with python on linux for my work.
Right now, when I try to insert the driver I get "insmod: ERROR: could not insert module /lib/modules/4.13.0-45-generic/updates/dkms/nvidia_384.ko: No such device". (also "modprobe nvidia" gives "modprobe: FATAL: Module nvidia not found in directory /lib/modules/4.13.0-45-generic")
In addition, dmesg says "NVRM: BAR1 is 0M @ 0x0 (PCI:0000:0a:00.0)
[   85.546913] NVRM: The system BIOS may have misconfigured your GPU."
I have tried all of the following:
1. "echo 1 > /sys/bus/pci/devices/0000\:00\:1c.0/remove
2. and echo 1 > /sys/bus/pci/rescan
3. adding pci=realloc to my kernel parameters
4. adding pci=nocrs
5. installing nvidia 384 driver
6. installing  nvidia-modprobe
7. installing nvidia 384 from a runfile (which i couldnt get to work for various reasons)
8. installing the driver from the ppa repository
9. installing through the "additional drivers" page in settings.
If it is any help, the output of dmesg and lspci are bellow:
00:00.0 Host bridge: Intel Corporation Sky Lake Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x16) (rev 07)
00:01.1 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x8) (rev 07)
00:01.2 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x4) (rev 07)
00:14.0 USB controller: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS I2C Controller #0 (rev 31)
00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1 (rev 31)
00:19.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS UART Controller #2 (rev 31)
00:1b.0 PCI bridge: Intel Corporation Sunrise Point-H PCI Root Port #17 (rev f1)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #1 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS UART #0 (rev 31)
00:1e.1 Signal processing controller: Intel Corporation Sunrise Point-H LPSS UART #1 (rev 31)
00:1e.2 Signal processing controller: Intel Corporation Sunrise Point-H LPSS SPI #0 (rev 31)
00:1e.3 Signal processing controller: Intel Corporation Sunrise Point-H LPSS SPI #1 (rev 31)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-H LPC Controller (rev 31)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-H PMC (rev 31)
00:1f.3 Audio device: Intel Corporation Sunrise Point-H HD Audio (rev 31)
00:1f.4 SMBus: Intel Corporation Sunrise Point-H SMBus (rev 31)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Device 67ef (rev c0)
01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device aae0
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a804
03:00.0 Network controller: Broadcom Corporation BCM43602 802.11ac Wireless LAN SoC (rev 02)
04:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt [Alpine Ridge] (rev 02)
05:00.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
05:01.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
05:02.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
05:04.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
06:00.0 System peripheral: Intel Corporation Device 15d2 (rev 02)
07:00.0 USB controller: Intel Corporation Device 15d4 (rev 02)
08:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt [Alpine Ridge]
09:01.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt [Alpine Ridge]
09:04.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt [Alpine Ridge]
0a:00.0 VGA compatible controller: NVIDIA Corporation Device 1b80 (rev a1)
0a:00.1 Audio device: NVIDIA Corporation Device 10f0 (rev a1)
0b:00.0 USB controller: Intel Corporation DSL6540 USB 3.1 Controller [Alpine Ridge]
7a:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt [Alpine Ridge] (rev 02)
7b:00.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
7b:01.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
7b:02.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
7b:04.0 PCI bridge: Intel Corporation Device 15d3 (rev 02)
7c:00.0 System peripheral: Intel Corporation Device 15d2 (rev 02)
7d:00.0 USB controller: Intel Corporation Device 15d4 (rev 02)
[   63.948786] nvidia-nvlink: Nvlink Core is being initialized, major device number 240
[   63.949134] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:0a:00.0)
[   63.949135] NVRM: The system BIOS may have misconfigured your GPU.
[   63.949139] nvidia: probe of 0000:0a:00.0 failed with error -1
[   63.949167] NVRM: The NVIDIA probe routine failed for 1 device(s).
[   63.949167] NVRM: None of the NVIDIA graphics adapters were initialized!
[   63.949326] nvidia-nvlink: Unregistered the Nvlink Core, major device number 240
[   76.475919] nvidia-nvlink: Nvlink Core is being initialized, major device number 240
[   76.476146] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:0a:00.0)
[   76.476147] NVRM: The system BIOS may have misconfigured your GPU.
[... 15 lines cut ...]
[   83.730265] pcieport 0000:7b:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
[... cut ...]
[ 1026.005735] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:0a:00.0)
[ 1026.005735] NVRM: The system BIOS may have misconfigured your GPU.
[ 1026.005738] nvidia: probe of 0000:0a:00.0 failed with error -1
[ 1026.005753] NVRM: The NVIDIA probe routine failed for 1 device(s).
[ 1026.005754] NVRM: None of the NVIDIA graphics adapters were initialized!
[ 1026.005845] nvidia-nvlink: Unregistered the Nvlink Core, major device number 240
​

matt-deboer, 2018-06-15:
I believe the device you need to manually remove is either ‘00:01.1’ or ‘
00:01.2’   (left and right PCIe bridge, whichever side your gpu is plugged
into); you might also try adding the `pci=scan-busses` kernel parameter (I
use it, but I needed it for my tb dock to be recognized)
(You can use `lspci -vt` to show the tree)

jacooba, 2018-06-16:
@matt-deboer 
This is the output of lspci -vt, but I don't know what it means:

00:00.0 Host bridge: Intel Corporation Sky Lake Host Bridge/DRAM Registers (rev 07)
	DeviceName: SATA
	Subsystem: Apple Inc. Skylake Host Bridge/DRAM Registers
	Flags: bus master, fast devsel, latency 0
	Capabilities: <access denied>
	Kernel driver in use: skl_uncore

00:01.0 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x16) (rev 07) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00004000-00004fff
	Memory behind bridge: 82600000-826fffff
	Prefetchable memory behind bridge: 00000000b0000000-00000000c01fffff
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:01.1 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x8) (rev 07) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Bus: primary=00, secondary=04, subordinate=79, sec-latency=0
	I/O behind bridge: 00006000-00009fff
	Memory behind bridge: 82800000-909fffff
	Prefetchable memory behind bridge: 00000000c0200000-00000000ce1fffff
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:01.2 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x4) (rev 07) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Bus: primary=00, secondary=7a, subordinate=ef, sec-latency=0
	I/O behind bridge: 0000a000-0000dfff
	Memory behind bridge: 90a00000-9ebfffff
	Prefetchable memory behind bridge: 000000009ec00000-00000000ad3fffff
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:14.0 USB controller: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31) (prog-if 30 [XHCI])
	Subsystem: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller
	Flags: bus master, medium devsel, latency 0, IRQ 37
	Memory at 82700000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd

00:15.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS I2C Controller #0 (rev 31)
	Subsystem: Intel Corporation Sunrise Point-H Serial IO I2C Controller
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at 82728000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci

00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1 (rev 31)
	Subsystem: Intel Corporation Sunrise Point-H CSME HECI
	Flags: bus master, fast devsel, latency 0, IRQ 60
	Memory at 82729000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: mei_me
	Kernel modules: mei_me

00:19.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS UART Controller #2 (rev 31)
	Subsystem: Intel Corporation Sunrise Point-H Serial IO UART Controller
	Flags: bus master, fast devsel, latency 0, IRQ 21
	Memory at 8272a000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: intel-lpss
	Kernel modules: intel_lpss_pci
[... 265 lines cut ...]

7b:04.0 PCI bridge: Intel Corporation Device 15d3 (rev 02) (prog-if 00 [Normal decode])
	Physical Slot: 2-1
	Flags: bus master, fast devsel, latency 0, IRQ 35
	Bus: primary=7b, secondary=b7, subordinate=ef, sec-latency=0
	I/O behind bridge: 0000c000-0000cfff
	Prefetchable memory behind bridge: 00000000a6200000-00000000ad3fffff
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

7c:00.0 System peripheral: Intel Corporation Device 15d2 (rev 02)
	Subsystem: Intel Corporation Device 0000
	Flags: fast devsel, IRQ 18
	Memory at <ignored> (32-bit, non-prefetchable)
	Memory at <ignored> (32-bit, non-prefetchable)
	Capabilities: <access denied>
	Kernel modules: thunderbolt

7d:00.0 USB controller: Intel Corporation Device 15d4 (rev 02) (prog-if 30 [XHCI])
	Subsystem: Intel Corporation Device 0000
	Physical Slot: 0-1
	Flags: fast devsel, IRQ 16
	Memory at <ignored> (32-bit, non-prefetchable) [disabled]
	Capabilities: <access denied>

jacooba, 2018-06-16:
@matt-deboer 

I got the gpu working! However, I still have some dmesg errors that seem bad. Any ideas if I need to fix these / how to fix these ?
If anyone is curious, as to how I fixed it:
I had add the ppa repo, install nvidia-396 through the "additional drivers", then:
echo 1 > /sys/bus/pci/devices/0000\:00\:01.1/remove
echo 1 > /sys/bus/pci/devices/0000\:00\:01.2/remove
echo 1 > /sys/bus/pci/rescan
dmesg errors:
[    0.069522] ACPI Error: AE_NO_MEMORY, SSDT 0xFFFF8F5EEBC5E800 Table is duplicated (20180313/tbdata-528)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU0.GCAP, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU0._OSC, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU1.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU1._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU2.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU2._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU3.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU3._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU4.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU4._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU5.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU5._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU6.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069522] ACPI Error: Method parse/execution failed \_PR.CPU6._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069730] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.069742] ACPI Error: Method parse/execution failed \_PR.CPU7.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.069746] ACPI Error: Method parse/execution failed \_PR.CPU7._OSC, AE_NOT_FOUND (20180313/psparse-516)
[    0.069746] ACPI Error: AE_NO_MEMORY, SSDT 0xFFFF8F5EEB828800 Table is duplicated (20180313/tbdata-528)
[    0.069746] ACPI Error: Method parse/execution failed \_PR.CPU0.GCAP, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.072004] ACPI Error: Method parse/execution failed \_PR.CPU0._PDC, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.072405] ACPI Error: AE_NO_MEMORY, SSDT 0xFFFF8F5EEBC07000 Table is duplicated (20180313/tbdata-528)
[    0.072413] ACPI Error: Method parse/execution failed \_PR.CPU1.APPT, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.072421] ACPI Error: Method parse/execution failed \_PR.CPU1.GCAP, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.072421] ACPI Error: Method parse/execution failed \_PR.CPU1._PDC, AE_ALREADY_EXISTS (20180313/psparse-516)
[    0.072421] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.072421] ACPI Error: Method parse/execution failed \_PR.CPU2.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.072421] ACPI Error: Method parse/execution failed \_PR.CPU2._PDC, AE_NOT_FOUND (20180313/psparse-516)
[    0.072597] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.072609] ACPI Error: Method parse/execution failed \_PR.CPU3.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.072614] ACPI Error: Method parse/execution failed \_PR.CPU3._PDC, AE_NOT_FOUND (20180313/psparse-516)
[    0.072892] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.072904] ACPI Error: Method parse/execution failed \_PR.CPU4.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.072909] ACPI Error: Method parse/execution failed \_PR.CPU4._PDC, AE_NOT_FOUND (20180313/psparse-516)
[    0.073186] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.073198] ACPI Error: Method parse/execution failed \_PR.CPU5.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.073202] ACPI Error: Method parse/execution failed \_PR.CPU5._PDC, AE_NOT_FOUND (20180313/psparse-516)
[    0.073480] ACPI BIOS Error (bug): Failure looking up [\_SB.OSCP], AE_NOT_FOUND (20180313/psargs-330)
[    0.073492] ACPI Error: Method parse/execution failed \_PR.CPU6.GCAP, AE_NOT_FOUND (20180313/psparse-516)
[    0.073497] ACPI Error: Method parse/execution failed \_PR.CPU6._PDC, AE_NOT_FOUND (20180313/psparse-516)
[... cut ...]
[   49.006316] hdaudio hdaudioC2D5: no AFG or MFG node found
[   49.006321] hdaudio hdaudioC2D6: no AFG or MFG node found
[   49.006326] hdaudio hdaudioC2D7: no AFG or MFG node found
[   49.006327] snd_hda_intel 0000:b9:00.1: no codecs initialized
[   54.972256] xhci_hcd 0000:ba:00.0: can't setup: -19
[   54.972319] xhci_hcd 0000:ba:00.0: init 0000:ba:00.0 fail, -19
[   56.450755] usbhid 7-2.2:1.1: couldn't find an input interrupt endpoint
[  104.853827] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52
​

Dunedan, 2018-09-20:
Closing this for now, as there hasn't been any activity for a while now anymore. Feel free to reopen if necessary.

closed 2018-09-20 by Dunedan

Bear-TheWolf312, 2019-09-01:
>macbookpro 14,3 ...

@matt-deboer I realize I’m late to the party but, eGPU usage on a nonMacOS operating system is unsupported by Apple, the EFI is set up to block this functionality. Secondly, NVidia GPUs aren’t officially supported by Apple either under macOS.

JorgeChiles, 2020-06-18:
217/5000
can you please help me
i have a mac book pro (2017) + Razer CoreX with Geforce RTX2080 super and i want to run it on Ubuntu 18.04 to speed up processes with tensorflow:

$ lspci -k | grep NVIDIA
b9:00.0 VGA compatible controller: NVIDIA Corporation Device 1e81 (rev a1)
b9:00.1 Audio device: NVIDIA Corporation Device 10f8 (rev a1)
b9:00.2 USB controller: NVIDIA Corporation Device 1ad8 (rev a1)
b9:00.3 Serial bus controller [0c80]: NVIDIA Corporation Device 1ad9 (rev a1)

$ nvidia-smi
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

I followed the procedures that show up to ...

adding pci=realloc to my kernel parameters OK.
but ..

$ echo 1 > /sys/bus/pci/devices/0000:00:01.1/remove
bash: /sys/bus/pci/devices/0000:00:01.1/remove: Permission denied

i cant change the permission..

can you help me, please
I'm not a Linux expert
excuse me
Thank you

KenaiTheWolf, 2020-06-18:
> 217/5000
> can you please help me
> i have a mac book pro (2017) + Razer CoreX with Geforce RTX2080 super and i want to run it on Ubuntu 18.04 to speed up processes with tensorflow:
>
> $ lspci -k | grep [...]

You might want to take stock of the repository you’re in. This is for getting T series chip equipped Macs to run new versions of Linux. This particular issue is about running an eGPU.

JorgeChiles, 2020-06-18:
...take stock of the repository you’re in..... ???
what does this mean ... excuse me but I'm not an expert

marc-git, 2020-06-18:
Jorge it means you are not asking this in the right place at all. Permission denied might be because you need to run as root. Try asking for help somewhere else.

JorgeChiles, 2020-06-18:
Ahhhh /.... Ok ... I did it
 i found this comand:

 echo 1 | sudo tee /sys/bus/pci/devices/0000\:00\:01.2/remove
[sudo] password for jorgeandrespenaaraya: 
tee: '/sys/bus/pci/devices/0000:00:01.2/remove': Permission denied
1