Dunedan/mbp-2016-linux #61: Call for keyboard data collection

Dunedan, 2018-05-19:
Over at https://github.com/cb22/macbook12-spi-driver/issues/64 we're currently collecting the results of info queries from the keyboard, to figure out what two unknown bytes in there mean. So if you have some time to spare, please head over there, gather the data and post it there together with your MBP model and your keyboard layout.

The most likely meaning of these two bytes is the keyboard layout, which would allow automated detection of the keyboard layout by the driver. Another theory is that it could contain a vendor id and product id which would allow automatic enabling of some Apple quirks for the Touchpad. In any case, figuring out what these two bytes mean, will likely result a nice benefit.

christophgysin, 2018-05-19:
Quick instructions:
```
$ git clone https://github.com/roadrunner2/macbook12-spi-driver.git
$ cd macbook12-spi-driver
$ make
$ sudo sh -c 'rmmod applespi; insmod applespi.ko'
$ dmesg | grep "applespi: info"
```

Dunedan, 2018-09-20:
Data collection was pretty successful. Thanks to everybody who participated. :+1:

closed 2018-09-20 by Dunedan