Dunedan/mbp-2016-linux #63: MBP 14.3 Fixed wifi

ivanrigovsky, 2018-06-07:
I just added 
`sudo iwconfig wlp3s0 txpower 10dBm `
and it's working ok now. 
Previous value 31dBm

mzanetti, 2018-06-11:
Just to confirm that I've been using this now for an entire day without an issue. I did video calls and lots of other web related activities. Not a single connection drop. I seem to get 30Mb/s downstream which is not great, but oh well, it works at least.

Thanks so much for providing this information. I really got fed up with using a USB WiFi dongle.

ivanrigovsky, 2018-06-14:
@mzanetti  I'm glad to hear it :) Also only 2.4GHz?

jboyens, 2018-06-14:
If you use the patched firmware referenced elsewhere, you can get less
stable 5Ghz, but it's good enough to be usable long-term.

mhkabir, 2018-06-14:
@jboyens how do we acquire this patched firmware?

jboyens, 2018-06-15:
https://github.com/Dunedan/mbp-2016-linux/issues/47

Dunedan, 2018-06-16:
As all MacBook Pro models with Touch Bar use the same wifi chip, let's keep the discussion in a single issue: #47 
Therefore closing this.

closed 2018-06-16 by Dunedan