Dunedan/mbp-2016-linux #65: Can't boot live USB's on MacBookPro 14,1

bemeurer, 2018-06-27:
I'm trying to dualboot Arch Linux on my MacBookPro 14,1; but I am completely unable to boot off of any liveUSB devices. When booting the computer with Alt held down I can see the EFI Boot option, but when I select it, it just boots into macOS. Here's what I've tried:

- Arch, Ubuntu, and Gentoo installation images, all of which had the exact same behavior
- USB2.0 and USB3.0 drives, both of which had the same behavior
- Attaching the drive through a dongle or through a dock, which also made no difference
- Creating the drive with Etcher, or manually; both of which yielding the same result
- Disabling secure boot (Utilities menu on the screen that comes up with cmd+R at boot)
- Shaking the computer

Am I missing something? Is there some special step that is required for me to be able to boot the liveUSB?

rwuwon, 2018-06-27:
Since you said Arch, I'm assuming you're trying to boot off the 4.17 kernel - is it the same as issue #62?

I've just installed Fedora 28 with the 4.16 kernel it comes with and that boots but after updating to the latest available there (4.17.2-200.fc28), it gives me the blank screen treatment as per #62 so I've removed that for now.

bemeurer, 2018-06-27:
@rwuwon The Arch liveusb iso comes with 4.16.12 as per the [downloads page](https://www.archlinux.org/download/). Moreover, I don't even get to see the boot menu, it just shows me the disk selection screen, and then when I select the EFI boot it boots into macOS.

rwuwon, 2018-06-27:
Oh I see. What method did you write to the USB with? For me, I used multibootusb onto a FAT32 volume (just the one 32G partition) and that seems to show up perfectly fine after holding down alt/option. MBP 13,1 here.

rwuwon, 2018-06-27:
Sorry, I'm kind of skipping some of your earlier lines in your first post, please disregard the previous.

bemeurer, 2018-06-27:
@rwuwon If I plug the drive on my secondary laptop, a Dell running Arch, it boots into it just fine, for what it's worth.

rwuwon, 2018-06-27:
That is quite baffling. So you can see the two UEFI boot options, but when you go to select the Linux one, it instead just proceeds to boot macOS - this is selecting through both right arrow + enter as well as clicking with the Trackpad?

Just in case this helps, `mount` reports for me:
/dev/sdb1 on /run/media/username/0762-98AA type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)

And GParted and Nautilus say there's 1.78 GiB used and 28.10 GiB unused on the fat32 file system with a boot flag. Same set-up for you?

bemeurer, 2018-06-27:
I just tried doing the following in order to replicate your layout:

- Plugged drive on Linux machine
- Partitioned to GPT and FAT32 with `parted`
- Set boot/esp flag on for the partition
- Mounted it and copied files over from a [Gentoo livecd iso](https://wiki.gentoo.org/wiki/User:Fearedbliss)
- Plugged into MacBook and rebooted

The behavior was _exactly_ the same as described previously. Weird and frustrating.

rwuwon, 2018-06-28:
Indeed! The only difference that appears to now remain is that I used multibootusb (which itself at times can throw in complications) and F28. I have previously had a Mac volume on the stick so that's why I more recently deleted that using GParted and redid it to FAT32. It doesn't sound likely that multibootusb + F28 will help you in this case, but it also doesn't seem likely that the 14,1 is that different to the 13,1 either.

Hopefully there's no weird hardware issue with yours and that it's something simple being overlooked. I'm otherwise on macOS 10.13.5 (whether that applied any firmware changes that are different to yours, I have no idea).

bemeurer, 2018-06-28:
@rwuwon, @Dunedan: I, very briefly, ran macOS Mojave Beta 2 on this machine. I now believe there has been a breaking firmware update on that release that rendered this computer unable to boot any non-macOS external drives. [Other people seem to be having the same issue](https://forums.developer.apple.com/thread/105081). I guess this means that until Mojave is properly released, booting Linux on this machine will be impossible.

mrtns, 2018-06-28:
> I guess this means that until Mojave is properly released, booting Linux on this machine will be impossible.

Ouch; thanks for the info here.

I'll chime in that I have a 14,1 and have been able to live boot Ubuntu 18.04 (which ships with kernel 4.15).

bemeurer, 2018-07-05:
I just installed the macOS 10.13.6 beta, which came with a firmware update that has fixed the issue. I can now boot into external drives again!

closed 2018-07-05 by bemeurer