Dunedan/mbp-2016-linux #67: Lock up / crash during libinput reinitialize

jboyens, 2018-07-10:
Bit of a weird one here that I haven't been able to reproduce consistently.

On a Macbook 14,3 when using Wayland with libinput handling the Touchpad. I've seen a lock up if I reload my Sway config which re-inits libinput to set device options. I've narrowed it down there because without it being defined in the config file, it seems to not die on reload.

What I see is a lock up of the graphical subsystem. No mouse, no change to the screen, nothing. Hardware power button works and will shut the machine down. Machine is back to normal on reboot. Network is still active during the "lockup" as I can ssh in. I haven't had success killing Wayland yet, when the lock up happens, but that's my next trick.

Happy to help debug in anyway that I can

Dunedan, 2018-09-20:
Is this still an ongoing problem? Any new insights into this?

Dunedan, 2018-10-27:
Closing this, because of missing feedback.

closed 2018-10-27 by Dunedan