Dunedan/mbp-2016-linux #69: mbp 13,2, Bluetootch not shown

Ninlives, 2018-07-25:
Does any one have the same problem? It's said that bluetooth works out of box on Linux 4.16 or above. I'm on 4.17 and the system seems not detecting the bluetooth hardware, neither `rfkill list` nor `dmesg` can find  the bluetooth.

Dunedan, 2018-07-25:
It does work out of the box, if the necessary kernel modules are available and loaded. Depending on the kernel you use that might not be the case.

Check the following comment for a list of necessary kernel modules: https://github.com/Dunedan/mbp-2016-linux/issues/29#issuecomment-368332604

Ninlives, 2018-07-26:
@Dunedan Thanks for your tips, recompile the kernel with those config turn on the bluetooth (although the sound over bluetooth headphone was...)

Now another quesion is, can I recompile maybe a module ? instead of the whole kernel to make it work? Apologize if you have mentioned in the thread, I just did a quick scan...

Dunedan, 2018-08-25:
Compiling individual modules is possible, but the process for it might depend on the Linux distribution you're using. E.g. check out the following instructions for Arch Linux: https://wiki.archlinux.org/index.php/Compile_kernel_module

As the Bluetooth issue itself is solved I'm closing this issue now.

closed 2018-08-25 by Dunedan