Dunedan/mbp-2016-linux #73: Linux kernel 4.18 breaks display

jboyens, 2018-08-19:
Just upgraded to 4.18 and it breaks the display in some way. I haven't figured it out yet, but it locks up to a back light black screen.

I'm not using a Login Manager or anything like that. Disabling kernel mode setting got me back to a usable prompt and let me downgrade the kernel.

jboyens, 2018-08-19:
I've tried with 4.18 -> 4.18.3 to no avail. There were a ton of AMDGPU "fixes" that went into 4.18 so maybe something there broke it.

jboyens, 2018-08-19:
Definitely the amdgpu driver. If loaded early in boot, it black screens faster.

roadrunner2, 2018-08-26:
I can confirm that booting using the dGPU gives a black screen under 4.18 on a MBP13,3 too; but booting with the iGPU works fine (see #6 for how to switch to using the iGPU).

turenar, 2018-08-29:
I tried to boot Linux 4.18.5 with MBP 14,3 (dGPU), and found my external display (connected by HDMI) working and treated as the only display connected with my MBP.
The built-in display is recognized first, but seems to be missed after a few seconds (when amdgpu module initialized?)

andersensam, 2018-09-05:
Downgrading to 4.15.18 on Ubuntu fixed the issue.

I noticed when building 4.18.x that I was getting errors about missing firmware for amdgpu, perhaps forcing a revert to radeon...

jboyens, 2018-09-08:
Looking into this a bit more, I see this in sway: 

```Output eDP-1 'Apple Computer Inc Color LCD 0x00000000'
  Current mode: 0x0 @ 0.000000 Hz
  Position: 5120,0
  Scale factor: 1x
  Transform: normal
```

That indicates to me that the display is being seen, but it seems like the backlight is off.

arno01, 2018-09-10:
Same with 4.19-rc3 :/

jboyens, 2018-09-11:
Booting with the iGPU does not work for me on 4.18. I get some serious graphic glitches. On <=4.17 the external displays won't work with the iGPU only on dGPU. External displays only are working on 4.18 and the eDP-1 is detected, just not enabled (no backlight). Using drm debug flags I can see that the 2880x1440 modeline is being called invalid. Wonder if I can provide a KMS modeline as a kernel arg.

I dug through a bunch of amdgpu bugs looking for hints or fixes, but I didn't find anything terribly useful.

roadrunner2, 2018-09-11:
@wsy2220 and @jboyens: have you tried booting with `i915.fastboot=1` (on the kernel command line)?

christophgysin, 2018-09-11:
I believe I have the same issue on 13,1 starting from 4.17. Currently running 4.16.8.

I have no AMD GPU. It seems that the display goes black immediately after the bootloader, even without i915 in the initrd (i915.fastboot=1 does not make a difference).

melentye, 2018-09-11:
Not sure if that's the same issue because for 14,3 it started with kernel 4.18, not with 4.17.

jboyens, 2018-09-11:
> @wsy2220 and @jboyens: have you tried booting with `i915.fastboot=1` (on the kernel command line)?

I worked on this a bit this morning. On 4.17, I needed to remove amdgpu from the initrd and blacklist the module. I also added i915 fastbook and the iGPU seems to be working fine, but only for the embedded display and not any externals.

jboyens, 2018-09-11:
@roadrunner2 Using your various tutorials and bug reports I was able to get both GPUs enabled at the same time and switch off the dGPU.

It required adding: amdgpu.dc=1 i915.fastboot=1 to the kernel commandline and NOT adding any modules to the initrd.

arno01, 2018-09-17:
I have just tried booting ``4.18.8`` with ``amdgpu.dc=1`` (without playing with switching iGPU/dGPU but rather sticking to the default dGPU) and it is working. I'll probably just stick to this option then so I can run the new kernels on my ``MBP 14,3``. :-)

melentye, 2018-09-17:
Kernel parameter `amdgpu.dc=0` suggested by @arno01 works for me on macbook 14,3 with kernel 4.18.7 too.

arno01, 2018-09-17:
@melentye yes, it does look like it is relevant!
I am inclined to believe now that the issue is happening due to a combination of two following things rather than an assumption that this only has to do with the kernels of versions like v4.18 and upwards:
1. lack of ``CONFIG_DRM_AMD_DC_PRE_VEGA=y`` in the kernel config;
2. ``amdgpu.dc=1`` which is active by default as @roadrunner2 observed above;

So when these two options are valid, users see the black screen.

The ``CONFIG_DRM_AMD_DC_PRE_VEGA`` option appeared in 4.15 (according to a quick search https://cateee.net/lkddb/web-lkddb/DRM_AMD_DC_PRE_VEGA.html )

arno01, 2018-09-18:
I think I've found the culprit. I was bisecting the AMD GPU related commits which came to ``v4.18-rc1``.

And if you revert the following commit https://github.com/torvalds/linux/commit/e03fd3f300f6184c1264186a4c815e93bf658abb, you will not have the "black screen" issue. At least in my case. Please try reverting it, rebuilding your kernel without this commit and let us know.

```
commit e03fd3f300f6184c1264186a4c815e93bf658abb
Author: Mikita Lipski <mikita.lipski@amd.com>
Date:   Wed May 16 16:46:18 2018 -0400

    drm/amd/display: Do not limit color depth to 8bpc
    
    Delete if statement that would force any display's color depth higher
    than 8 bpc to 8
    
    Signed-off-by: Mikita Lipski <mikita.lipski@amd.com>
    Reviewed-by: Harry Wentland <Harry.Wentland@amd.com>
    Signed-off-by: Alex Deucher <alexander.deucher@amd.com>

diff --git a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
index 1ce10bc2d37b..52e57b52cdbb 100644
--- a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
+++ b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
@@ -2095,12 +2095,6 @@ convert_color_depth_from_display_info(const struct drm_connector *connector)
 {
        uint32_t bpc = connector->display_info.bpc;
 
-       /* Limited color depth to 8bit
-        * TODO: Still need to handle deep color
-        */
-       if (bpc > 8)
-               bpc = 8;
-
        switch (bpc) {
        case 0:
                /* Temporary Work around, DRM don't parse color depth for
```

roadrunner2, 2018-09-19:
I just tried booting again with the dGPU. On my MBP13,3 I see the following behaviour:
- amdgpu.dc=0 everything works fine (usable display on boot and on login)
- dc=1 and the above patch: display during boot is fixed, but get black screen (backlight is on, though) after login (both Wayland and X)

Then I had a hunch the problem in the second case might be the resolution after login (I was using 1920x1200) - changing the resolution to 2880x1800 or 1440x900 fixed this. However, when I changed back to 1920x1200 and tried one more then it suddenly was fine. Go figure. In short, there appears to still be some resolution related flakiness.

P.S. Gnome appears to remember display resolutions based on GPU, so setting the resolution while on iGPU does not affect the resolution when later booting with dGPU, and visa versa. So if anybody plays with resolutions, make sure you change them while running the same GPU that you're trying to test under.

roadrunner2, 2018-09-20:
Btw., forgot to mention, but excellent work @arno01 on bisecting the issue - that's a tedious process!

Dunedan, 2018-09-20:
Great work @arno01. Is my understanding correct that this is just a bug which got introduced with 4.18 and will be fixed with 4.19 again? If yes, anything else to discuss here or can we just close this issue?

arno01, 2018-09-21:
@Dunedan I'd say yes. I'd love OP to confirm it though.
@jboyens are you able to try the kernel without that particular commit? If you have difficulty with compiling the kernel, I can send you mine 4.19-rc4 (for Ubuntu)

ClashTheBunny, 2018-09-25:
I also had an interesting experience on 4.18 related to display.  Did the backlight change with the new kernel?  It seems like backlightctl now controls `amdgpu_bl1` and not `gmux_backlight`.  It only turns off or on the display.  The brightness is still controlled with `gmux_backlight`

hutualive, 2018-10-19:
@arno01 Hi I tried with amdgpu.dc=0, still not working, both v4.19-rc8 and v4.18.
hw: MacBookPro14,3 15.4" 2017.

the way I set kernel parameter is:
$sudo vi /etc/default/grub
--> add amdgpu.dc=0 in the line GRUB_CMDLINE_LINUX_DEFAULT="quiet splash amdgpu.dc=0"
$sudo grub-mkconfig -o /boot/grub/grub.cfg
$sudo reboot

still not working as the display is blank(black), but display through HDMI is working as usual.

what's wrong with the process ?

btw: I checked inside mac os, MacBookPro14,3 15.4" 2017 with 2.9GHz Core i7, the GPU is Intel graphics 630, through glxinfo still show the info for AMD Radeon.

roadrunner2, 2018-10-20:
@arno01 or @jboyens: can I suggest one of you open a ticket upstream (on https://bugs.freedesktop.org/, `DRM/Intel` component) so this gets looked at there?

arno01, 2018-11-08:
@roadrunner2 I have submitted the ticket there https://bugs.freedesktop.org/show_bug.cgi?id=108693

arno01, 2018-11-20:
**update**: a proper patch would be https://gist.github.com/arno01/8115ef124fe7f42bbf1c317892013e86

roadrunner2, 2018-11-27:
@arno01 Thanks for shepherding this. I can confirm the patches have made it into 4.20-rc4 and fix the issue :+1: .

frazer-jamieson, 2018-12-11:
Is there an iso file available with this kernel please?

frazer-jamieson, 2018-12-13:
Please could anyone advise on a suitable linux .iso for a Macbook Pro 2018 - this'll be my first attempt at Linux on Mac - Linux I am familiar with (favourite distro being Manjaro) - Mac's I am not....

Many thanks indeed everyone

arno01, 2018-12-13:
@frazer-jamieson please try this one:

```
ISO with Ubuntu 18.04 for MacBook (Linux 4.15.0-22), touchbar and keyboard are working:

Download 2.4G: https://files.nixaid.com/ubuntu4mac.iso
GPG signed checksum: https://files.nixaid.com/ubuntu4mac.iso.gpg
```

frazer-jamieson, 2018-12-14:
@arno01 thank you sincerely for your help :)

Dunedan, 2018-12-29:
Isn't this issue fixed with 4.20?

arno01, 2018-12-29:
I can confirm this in January when I undust my MBP :-) Close this if @jboyens confirms it sooner than I.

jboyens, 2018-12-30:
I've confirmed that this seems to work perfectly on 4.20. It even worked on Arch's 4.19.12 kernel (which is the current latest; 4.20 is in testing).

Either way, it seems the problem has been resolved from my end. I was without my MBP for a significant amount of time, nearly since I opened the issue, so I'm gutted that I couldn't help debug a bit more, but thrilled this was resolved. Thank you.

Closing the issue.

closed 2018-12-30 by jboyens