Dunedan/mbp-2016-linux #78: 14,1 wifi

naught101, 2018-09-19:
I have a MBP14,1, and wifi is not working out of the box on Kubuntu 18.04. I have tried a few kernels, including mainline kernels up to 4.19rc4, and none of them work.

I *have* got the wifi working by force-installing the Debian version of the drivers from https://packages.debian.org/sid/all/firmware-brcm80211/download, however the wifi driver interrupt starts using a tonne of CPU time after a while, and producing a lot of heat.

naught101, 2018-09-19:
OK, and after removing that driver and going back to the 4.15 bionic kernel, it decided to start working after a reboot. Not sure why, it hasn't before. Still runs warm, and uses more CPU than it probably should.

roadrunner2, 2018-09-20:
Time is still being spent in the wifi interrupt handler? I ask, because there's also #70 which should apply to all 13,* and 14,* models and which could be a source of CPU activity and hence heat and power consumption.

Dunedan, 2018-09-27:
@naught101: Apparently you didn't have the necessary firmware installed. Instead of installing the mentioned Debian package, you should be fine installing the `linux-firmware`-package provided by Ubuntu: https://packages.ubuntu.com/bionic/all/linux-firmware

nedclimaterisk, 2018-10-08:
Yeah, this is a bad report. After a re-install it seems to be working fine. Not sure what I did wrong, unfortunately. Sorry for the noise.

closed 2018-10-08 by naught101