Dunedan/mbp-2016-linux #82: As far as I can see, the discussion currently revolves around three issues:

gdaddar, 2018-10-24:
1. get Bluetooth to work at all for the MacBook8,1
2. get Bluetooth to work reliably for the MacBookPro13,1 and MacBookPro14,1
3. audio stuttering when streaming audio via Bluetooth in certain situations (when starting playback, while a Bluetooth scan is running etc.)

From my point of view 3. is out of scope for this issue, as that's no Apple-hardware-specific problem and happens with other hardware as well. E.g. I was immediately able to reproduce the audio stuttering issue when opening the Bluetooth device management on a non-Apple computer with an USB-connected Bluetooth dongle. Additional, pulseaudio (I guess all of you are using for audio streaming) is known for such issues in general.

To test 1. and 2., using audio streaming as a test case is probably the worst one you can use, as it's the one with the highest software complexity. Aside from a working Bluetooth driver you'll need working pulseaudio, proper bluetooth-pulseaudio module, correct audio settings, etc. What I usually do to check if Bluetooth works at all is to transfer files between the computer and a smartphone. Also using a Bluetooth peripheral like a Bluetooth-enabled mouse is a good test case.

I suggest going forward we don't use audio streaming as test case if Bluetooth works, but instead focus on file transfer and peripheral devices. To get a better overview what the status for the 13,1 and 14,1 is I assembled the following table with all known owners and their status. Is that up-to-date for everybody? If not, please provide more information.

| Owner | Device | Kernel | [15ms patch?](https://github.com/lategoodbye/rpi-zero/commit/ed5900296dfb) | working? |
| --- | --- | --- | --- | --- |
| @chadberg | MacBookPro13,1 | 4.16-rc6 | ? | yes |
| @christophgysin   | MacBookPro13,1 | 4.16-rc4 | yes | yes |
| @cyrusmg | MacBookPro13,1 | 4.16-rc4 | no | no |
| @cyrusmg  | MacBookPro13,1 | 4.16-rc4 | yes | yes |
| @bhvr-fmarceau | MacBookPro14,1 | 4.16-rc5 | yes | no |
| @Lostish | MacBookPro14,1 | 4.16-rc6 | yes | no |
| @ClashTheBunny | ? | ? | ? | yes |
| @gdaddar | MacBookPro14,1 | 4.18.0-10-generic | ? | ? |

_Originally posted by @Dunedan in https://github.com/Dunedan/mbp-2016-linux/issues/29#issuecomment-376429986_

chadberg, 2018-10-24:
Under released 4.16 kernels (I'm on 4.16.13 at the moment), bluetooth is not working for me. I've not yet tried the 15ms patch.

tomaytotomato, 2018-10-24:
Bluetooth is not working on 4.18 using Ubuntu 18.10 LTS  for a MBP 13,1