Dunedan/mbp-2016-linux #84: Could you share MacBookPro14,3 ACPI & IOREG & SPX dumps

RockJesus, 2018-11-04:
Hi Could you share MacBookPro14,3 ACPI & IOREG & SPX dumps ,Thx

ClashTheBunny, 2018-11-04:
What's missing from my dump?

ClashTheBunny, 2018-11-04:
I installed `acpidump` and re-ran.  Is that better?

RockJesus, 2018-11-05:
Thx a lot

[IORegistryExplorer.app.zip](https://github.com/Dunedan/mbp-2016-linux/files/2546676/IORegistryExplorer.app.zip)
Using this app on macos could get a dump of Ioreg file ,I’m truly grateful for your help!

ClashTheBunny, 2018-11-05:
I'm not big into running random binaries off the internet.  Maybe somebody else here has the guts to run those?

RockJesus, 2018-11-05:
ok,anyway,Thx for help

closed 2018-11-05 by Dunedan

Dunedan, 2018-11-05:
@RockJesus: What do you need that data for? Is it Hackintosh related?

RockJesus, 2018-11-06:
> @RockJesus: What do you need that data for? Is it Hackintosh related?

Yes,any help will be appreciated!