Dunedan/mbp-2016-linux #88: MBP 13,1 multiscreen high dpi for ubuntu 18.10?

tomaytotomato, 2018-12-05:
So far so good, I am really enjoying Ubuntu 18.10 on my MBP 13,1!

One thing that I have not figured out is how to scale the display settings for the primary display (2560x1600).

I have 3 displays in the following configuration (1920x1200, 1920x1200, 2560x1600) with screen scaling set to 100%.

[image]

I am unable to individually specify screen scaling for each display. 

As you can see in the photo below my Spotify and other apps are near impossible to read. 

If I set the scaling to 200% then I can read pages clearly on the MBP screen, but then the external monitors are rendering applications or content way too big.

[image]

After googling I tried the ubuntu tweaker and font scaling but that didn't help. 

Any ideas?

WhyNotHugo, 2018-12-05:
Xorg doesn't really support per-display scaling, so there's no way around this.

It's an inherit Xorg limitation, so it's way beyond the scope of this repo really. Wayland, I believe, seeks to address this. Dunno about the progress made though.

Dunedan, 2018-12-08:
As the others already mentioned this isn't related to the hardware, but rather a software-side problem all users with displays with different DPI are facing, no matter if they use Apple hardware or not. Therefore I'm closing this now, as there a better suited places to discuss such problems.

closed 2018-12-08 by Dunedan