Dunedan/mbp-2016-linux #89: 2018 MacBookAir8,1

uwqwxing886, 2018-12-31:
The 2018 MacBook Air boots from a USB drive with Ubuntu however the touchpad and keyboard don't work. Has anyone been able to get these input devices to work?

I'm using the ISO from https://nixaid.com/linux-on-macbookpro/ that includes keyboard and touchpad drivers that work on the 2018 MacBook pros.

uwqwxing886, 2019-01-05:
Do you have an ISO that will run the script and display the output on the screen? I could take a photo of it.

ppaulweber, 2019-01-07:
@uwqwxing886 I'll take care of this issue (PR of `get-info.sh`) today. I'm using an MacBookAir8,1 as well (archlinux) and I'm currently in the progress in debugging and providing proper drivers.

ppaulweber, 2019-08-17:
@uwqwxing886 FYI: https://github.com/ppaulweber/linux-mba