Dunedan/mbp-2016-linux #91: Battery capacity

Ninlives, 2019-01-08:
I'm running Linux 4.20 on MBP 2016 with Touch Bar, and I recently notice the battery capacity decrease quickly, today it has 83% remained and the loop count is 270 now. Is this an expected behavior?

roadrunner2, 2019-01-11:
Hmm, I'm at 83 charge cycles with about 92% design capacity; and another system that has been running only MacOS shows about 94% and 127 cycles. So I think your capacity is in the ballpark, given your number of charge cycles - however, those do seem high. A lot depends on how often and deeply you discharge your batteries, however.

Dunedan, 2019-01-20:
My battery has 95% left after 66 cycles. As you can see, I'm rarely using it running on battery.
@Ninlives What's your usage pattern? Do you use it regularly running on battery?

Ninlives, 2019-01-21:
@Dunedan No, actually I almost always run on AC power

Dunedan, 2019-01-21:
Do you use the original Apple 61W power adapter or another one?

Ninlives, 2019-01-22:
Original adapter, but change the wire. I mean, the wire connect to the AC power but not the laptop(sorry I don't know how to say it in English :persevere: )

Dunedan, 2019-01-26:
So you're not using the original cable (https://www.apple.com/shop/product/MK122LL/A/power-adapter-extension-cable) to connect the power adapter to AC power, but a third party one? If that's the case that shouldn't be an issue.

Dunedan, 2019-03-28:
Closing this, because there is no response from the author.

closed 2019-03-28 by Dunedan