Dunedan/mbp-2016-linux #94: Sound via Apple's USB-C to 3.5mm Headphone Jack Adapter

risen, 2019-02-19:
Using Apple's USB-C to 3.5mm Headphone Jack Adapter seemed like a good idea to get around the MacBookPro's lack of audio support.

Apple introduced it for the iPad Pro, it's very compact, and AFAICT it contains a DAC capable of 44/48khz and 16/24bit audio. I hoped it would straight up support the USB Audio class, but alas, no such luck, it seems to only bind to the generic HID driver.

Now, I wanted to open this issue to track the linux support for this dongle, see if anybody had heard of a sound driver for HID devices (using kernel driver or quirk, libusb or hidraw), and leave some more detailed info for others.

Plugging it in, this shows up in the kernel log:

```
[  334.000849] usb 1-2: new full-speed USB device number 2 using xhci_hcd
[  334.149509] usb 1-2: New USB device found, idVendor=05ac, idProduct=110a
[  334.149520] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  334.149527] usb 1-2: Product: USB-C to 3.5mm Headphone Jack Adapter
[  334.149533] usb 1-2: Manufacturer: Apple, Inc.
[  334.149539] usb 1-2: SerialNumber: DWH845603KLL1MQAF
[  334.169655] usbcore: registered new interface driver usbhid
[  334.169656] usbhid: USB HID core driver
[  334.175686] hid-generic 0003:05AC:110A.0001: hiddev0,hidraw0: USB HID v1.11 Device [Apple, Inc. USB-C to 3.5mm Headphone Jack Adapter] on usb-0000:00:14.0-2/input0
```

```
# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

```
# lsusb -s 1:2 -v

Bus 001 Device 002: ID 05ac:110a Apple, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x05ac Apple, Inc.
  idProduct          0x110a 
  bcdDevice           26.11
  iManufacturer           1 Apple, Inc.
  iProduct                2 USB-C to 3.5mm Headphone Jack Adapter
  iSerial                 3 DWH845603KLL1MQAF
  bNumConfigurations      3
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      42
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     2
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      42
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     3
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      42
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          4
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000000
      (Missing must-be-set LPM bit!)
  Configuration Summary Device Capability:
    bLength                    10
    bDescriptorType            16
    bDevCapabilityType         16
    bcdVersion               1.00
    bClass                      1
    bSubClass                   1
    bProtocol                  48
    bConfigurationCount         1
    bConfigurationIndex(0)      3
  Configuration Summary Device Capability:
    bLength                    10
    bDescriptorType            16
    bDevCapabilityType         16
    bcdVersion               1.00
    bClass                      1
    bSubClass                  33
    bProtocol                  48
    bConfigurationCount         1
    bConfigurationIndex(0)      2
  Configuration Summary Device Capability:
    bLength                    10
    bDescriptorType            16
    bDevCapabilityType         16
    bcdVersion               1.00
    bClass                      1
    bSubClass                   0
    bProtocol                  32
    bConfigurationCount         1
    bConfigurationIndex(0)      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

risen, 2019-02-22:
Apparently jack detection wasn't working with the specific (broken?) audio cable I was using. Switching it out made more audio-specific configuration appear in lsusb, and sound worked out of the box. Amazingly, even the hardware volume buttons on the iPhone compatible earbuds are correctly supported in Linux.

So this 9$ compact dongle is a very useful DAC for MacBook Linux users that don't have sound, and want a simple alternative for the internal speakers or bluetooth audio.

Closing this bug, since all is working as expected.

```
# lsusb -s 1:2 -v

Bus 001 Device 002: ID 05ac:110a Apple, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x05ac Apple, Inc.
  idProduct          0x110a 
  bcdDevice           26.11
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      3
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00f8
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        bCategory               4
        wTotalLength       0x0040
        bmControls           0x00
      AudioControl Interface Descriptor:
        bLength                17
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
[... 612 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      42
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

closed 2019-02-22 by risen

ivanrigovsky, 2019-02-25:
My MPB 14.3 doesn't see any USB devices with mini jack.

risen, 2019-03-01:
@ivanrigovsky do you mean the internal one, or the dongle?

The internal sound card is indeed still not functional, but I got [Apple's USB-C to 3.5 mm Headphone Jack Adapter](https://www.apple.com/shop/product/MU7E2AM/A/usb-c-to-35-mm-headphone-jack-adapter) to be able to use headphones in a hassle-free way.

ivanrigovsky, 2019-03-03:
@risen I mean a Apple's USB-C to 3.5 mm Adapter. There is no DAC, isn't it? 
### 'lsusb -s 1:2 -v' doesn't display any new devices.

risen, 2019-03-03:
@ivanrigovsky there's a DAC in there, surprisingly, capable of 44/48khz and 16/24bit. I can also control my sound volume with the hardware volume controls of Apple iPhone earbuds, for instance.

Just use `lsusb -v`, not the `-s 1:2` which is for the specific USB port that I had mine plugged into, you could be using another port.

ivanrigovsky, 2019-03-04:
@risen unfortunately, I have compared `lsusb -v` command outputs with connected USB-C Adapter and without - they were the same. It looks like unused usb

risen, 2019-03-04:
@ivanrigovsky strange… USB-C audio can pass through analog, and/or use digital (usb audio) streams. The dongle I bought was from the online Apple store (10€, free shipping). Maybe you have another kind (either different model or different brand) that uses the analog path?

jboyens, 2019-03-04:
I see a similar new audio device created when I attach the Google Headphone adapter that I received with my mobile phone:

```
Mar 04 13:37:31 flexo kernel: usb 1-2: new full-speed USB device number 41 using xhci_hcd
Mar 04 13:37:31 flexo kernel: usb 1-2: New USB device found, idVendor=18d1, idProduct=5034, bcdDevice= 0.21
Mar 04 13:37:31 flexo kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 04 13:37:31 flexo kernel: usb 1-2: Product: Headphone adapter
Mar 04 13:37:31 flexo kernel: usb 1-2: Manufacturer: Google Inc
Mar 04 13:37:31 flexo kernel: usb 1-2: SerialNumber: 89NG019BLXTK02VN
Mar 04 13:37:31 flexo kernel: input: Google Inc Headphone adapter as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:18D1:5034.001E/input/input51
Mar 04 13:37:31 flexo kernel: hid-generic 0003:18D1:5034.001E: input,hidraw7: USB HID v1.11 Device [Google Inc Headphone adapter] on usb-0000:00:14.0-2/input3
Mar 04 13:37:31 flexo kernel: usbcore: registered new interface driver snd-usb-audio
```

ivanrigovsky, 2019-03-05:
@risen yep, maybe, I'll try to buy another one. Thanks.

christophgysin, 2019-03-06:
I just picked one of these up (Apple USB-C to headphone jack adapter), and can confirm that it works with both stereo playback and analog capture. It also supports volume/play/pause buttons on two different headsets I tried.

ivanrigovsky, 2019-04-10:
Guys, what about volume? I hear ok only with maximum volume, but often it is not enough in Skype for example. How I can improve volume for that case with Apple USB-C headphones?

monkcs, 2019-04-10:
You can install gnome-tweak-tools and enable overampflication for the audio output in settings menu