Dunedan/mbp-2016-linux #96: USB C dock/cable

mfauvain, 2019-03-11:
Macbookpro14,1 (kernel 4.16) was running fine on a USBc dock (pluggable) up until today when Ethernet connection from dock is not recognized anymore. Or more precisely, it is recognized and then (after a few seconds) drops with below log
`[ 5074.197445] r8152 4-2.3:1.0 enp7s0u2u3: carrier on
[ 5075.895104] r8152 4-2.3:1.0 enp7s0u2u3: Stop submitting intr, status -71
[ 5083.461988] usb 4-2.3: Disable of device-initiated U1 failed.
[ 5083.476025] usb 4-2.3: Disable of device-initiated U2 failed.
[ 5083.490031] usb 4-2.3: usb_reset_and_verify_device Failed to disable LTM
[ 5083.676896] usb 4-2.3: USB disconnect, device number 7
[ 5083.810423] usb 4-2.3: Device not responding to setup address.
[ 5084.023994] usb 4-2.3: Device not responding to setup address.
[ 5084.230311] usb 4-2.3: device not accepting address 9, error -71
[ 5084.543813] usb 4-2.3: new SuperSpeed USB device number 10 using xhci_hcd
[ 5084.561039] usb 4-2.3: New USB device found, idVendor=0bda, idProduct=8153
[ 5084.561045] usb 4-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[ 5084.561049] usb 4-2.3: Product: USB 10/100/1000 LAN
[ 5084.561053] usb 4-2.3: Manufacturer: Realtek
[ 5084.561056] usb 4-2.3: SerialNumber: 000001
[ 5084.570004] usb 4-2.3: Enable of device-initiated U1 failed.
[ 5084.576999] usb 4-2.3: Set SEL for device-initiated U2 failed.
[ 5084.605608] usb 4-2.3: Disable of device-initiated U1 failed.
[ 5084.614191] usb 4-2.3: Disable of device-initiated U2 failed.
[ 5084.621000] usb 4-2.3: Set SEL for device-initiated U1 failed.
[ 5084.627996] usb 4-2.3: Set SEL for device-initiated U2 failed.
[ 5084.635069] r8152 4-2.3:2.0: Unknown version 0x0000
[ 5084.642019] usb 4-2.3: Set SEL for device-initiated U1 failed.
[ 5084.649013] usb 4-2.3: Set SEL for device-initiated U2 failed.
[ 5084.657937] usb 4-2.3: usb_reset_and_verify_device Failed to disable LTM
[ 5085.618803] usb 4-2-port3: Cannot enable. Maybe the USB cable is bad?
[ 5085.619600] usb 4-2.3: USB disconnect, device number 10
`
What is very weird, is that I didn't change the USBc cable (from dock to laptop) and it was working fine before.
Now, changing cable to Apple usbc/usbc cable works! (who knows why) and also, same 'now non-working' cable (the one provided with the pluggable dock) used at work on the same pluggable usbc dock (and same laptop obvioulsly) works....
I am clueless. If cable works at work, why wouldn't it work at home on the same  dock? Also, why apple's cable works (on same pluggable dock). I did experience a very weird behaviour with an apple thunderbolt 3 to 2 adapter (see issue #87 ) but here I am speechless.

mfauvain, 2019-03-14:
I had another shot at googling issue, bad cable, r8152 etc.. still stuck.
- with Pluggable cable, ethernet works booting OSX (and all the other ports as well)
- with Apple cable, ethernet works (as all the other ports) booting linux (4.16)
- with Pluggable cable, ethernet works booting linux at work (same USB dock model, not same dock though)
- all USB port + audio + charging work with Pluggable cable booting linux, but Ethernet won't. logs above.

I've tried unload r8152 mod, then reload. booting with cable plugged, and unplugged. All the same. Ethernet won't work, even though it used to (with this exact same cable)
Any way to override the `usb_reset_and_verify_device Failed to disable LTM`?
It seems very related to same 'security' issue I had with thunderbolt a few months ago.

jboyens, 2019-03-14:
I've had the same problem with of my hubs recently. I'm going to try a much
older kernel to see if it's a userland issue or an issue in the kernel.
I've been able to "wake" the device up by doing a probe with ```lsusb``` or
another tool to probe the entire USB tree. So I'd expect something changed
in the kernel and USB is a bit lazy.

jboyens, 2019-03-19:
(I just had to do this, so I thought I'd update with how it's done)

When I reattached after a meeting, my hub w/ ethernet did not reconnect and was not detected by the system. I then ran `sudo usbhid-dump` and it was found and reattached.

mfauvain, 2019-03-19:
`sudo usbhid-dump` not working for me. Dock is still recognized, with power and everything else, apart from Ethernet. 
With Apple's cable it does though.

jboyens, 2019-03-19:
Wait. I wonder if it's a power issue. I have an cdc_ether/r8152 ethernet ported hub. It won't work unless I pass my laptop charger through the USB-C port on the hub. i.e. power must flow through the hub or ethernet doesn't work. Is that what you mean when you say you must use Apple's cable?

mfauvain, 2019-03-19:
Hmm, not sure I follow. Hub is a powered hub. So in that sense yes, I use the hub to power the laptop and for the ports it has (USB, Ethernet, Audio). When I use the hub/dock (the brand is Pluggable) with the provided Pluggable USBC cable the Ethernet port is not recognized (it used to with exact same cable) and everything else does. But if I use Apple's cable (the one provided with the laptop) and only the cable (not the Apple charger) everything on the Pluiggable hub/dock works.
If it is a power issue (ie, cable is bad) then why would the cable work at my office?
If it is a Realtek driver issue, why simply using another cable (apple's one) works?

mfauvain, 2019-03-25:
Using Apple's cable with the Dock works and gives another Ethernet name for the port : 
`[ 6165.606600] r8152 1-8.3:1.0 enp0s20f0u8u3: carrier on`
instead of the 
`[ 5074.197445] r8152 4-2.3:1.0 enp7s0u2u3: carrier on` when I use the Pluggable cable. Any reason for this? is this normal that port changes name using different cable?

Also tried using usb_modeswitch, not better.

christophgysin, 2019-03-26:
> Any reason for this? is this normal that port changes name using different cable?

Yes, that's expected behaviour, see [Predictable Network Interface Names](https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames).

You can rename a specific network interface to a consistent name using udev rules, by creating a file `/etc/udev/rules.d/10-rename-network.rules` with the contents:
```
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="ff:ff:ff:ff:ff:ff", NAME="eth0"
```
Just replace the address with the MAC address of your interface.

mfauvain, 2019-03-26:
@christophgysin thank you. However reading your link it seems ethernet port should stick to the same name shall I use apple's cable or pluggable's cable.
Or maybe I wasn't clear in my post, I use the same dock (and same plug) and try one cable and then another cable, according to `3. Names incorporating physical/geographical location of the connector of the hardware (example: enp2s0)` it shouldn't change name.. Or you are saying that the kernel determine the name based on the cable not the actual port on the hub?
What I am trying to unpuzzle is if for some reason, with one cable (the bad one) systems is stuck on a non existant interface, whereas with another cable systems switches correctly.

roadrunner2, 2019-03-28:
There are two USB controllers on the *,1 machines, and it looks like a different one is used depending on which cable you use (enp0s20... is pci dev 00:14.0, enp7s0... is pci dev 07:00.0). The only reason for this I can think of offhand that a different mode is being negotiated when using different cables (e.g. direct USB 2/3/3.1 mode vs DP+USB-3.1 vs Thunderbolt), but why that would be the case I don't know either.

mfauvain, 2019-03-29:
@roadrunner2 thank you, makes complete sense. Now, why is it the case I have no clue either.

mfauvain, 2019-04-02:
Some magic, the thunderbolt adapter TB3 to TB2 that was causing issue (issue #87) is now working... ie I have Ethernet, USB, audio and Display working. So it seems port 07:00.0 is working ok. What seems to be the issue is the switch/selection between 00:14:0 and 07:00.0.... not sure. 

Also I did try booting the Arch USB install and issue with USB dock Ethernet port not being recognized is there as well..

Dunedan, 2019-08-23:
@mfauvain Do you got any new insights in the meanwhile?
Somebody else here able to reproduce the problem?

mfauvain, 2019-08-23:
Luckily on holidays so could not try at work. Wifi works anywhere I've been though. I guess I will have to avoid work in the future ha!
Will revert when back, early Sep.

jboyens, 2019-08-29:
I still have a similar problem daily on the 14,3. My consistent fix has been to run `sudo lsusb -v` to force enumerating the whole device chain. Once I do that, everything is fine.

Every great once-and-a-while, it won't work, when that happens, I unplug the dock, unplug the power from the dock, wait a few seconds and then hook it all back up. After that I'm back in business.

mfauvain, 2019-09-02:
My bad, comment above was for issue #109. @Jboyens, I tried your fix again with `sudo lsusb -v`, but ethernet port on dock is still absent when using usb-c cable provided with the dock (was working initially). When using Apple's cable, ethernet port is there.

jboyens, 2019-09-09:
Sorry I wasn't able to help more!

mfauvain, 2019-09-14:
Had another shot at it today, now with kernel 5.12.14 and did what @jboyens suggested.. surprise, ethernet is back.. I powered down the dock, and ran `sudo lsusb -v`, for some reason this time the command took longer, pausing on some usb ports anf then magically the ethernet port appeared.. but this time on the same port as with the apple usb cable :
`[  411.415100] r8152 1-8.3:1.0 enp0s20f0u8u3: carrier on`
and not the one that was used before (with same original dock cable) : enp7s0u2u3...

Dunedan, 2019-09-15:
As you've narrowed down the problem a bit now: Have you considered opening an issue in the kernel bugzilla, so that hopefully somebody with more knowledge of the USB stack can have a look at it?

mfauvain, 2019-09-15:
Still very unclear to me, however I submited the bug in bugzilla : [https://bugzilla.kernel.org/show_bug.cgi?id=204871](url)

mfauvain, 2019-10-15:
back again with the issue. Same as last time. power failure while connected. after power is back up, ethernet port not recognized with original usb-c cable. again, 'sudo lsusb -v' not doing any magic. Reported the issue again to linux-usb@vger.kernel.org as asked on bugzilla.

Dunedan, 2020-12-08:
@mfauvain @jboyens Any updates to the problems you were facing? Are they still present or did they get fixed in the meanwhile?

mfauvain, 2020-12-08:
Still present yes, the only 'workaround' that 'works' is to boot macOS, unplug USBc cable, shutdown and plug USBc Cable, this triggers the poweron. Mac is setup to boot to Linux by default (systemd-boot) and cable/dock is then recognized with all ports working... 
Currently running kernel 5.9.12.