Dunedan/mbp-2016-linux #97: linux-5.x regulator_enable oops prevents booting into X

christophgysin, 2019-03-30:
On MacbookPro13,1, vanilla linux-5.x finally boots again without any patches. Unfortunately, X wont start as mentioned by https://github.com/Dunedan/mbp-2016-linux/issues/62#issuecomment-473831525. The problem seems to stem from:

```
BUG: unable to handle kernel NULL pointer dereference at 0000000000000078
#PF error: [normal kernel read fault]
PGD 0 P4D 0
Oops: 0000 [#1] PREEMPT SMP PTI
CPU: 2 PID: 152 Comm: kworker/u8:2 Tainted: G           OE     5.0.4-arch1-1-ARCH #1
Hardware name: Apple Inc. MacBookPro13,1/Mac-473D31EABEB93F9B, BIOS 228.0.0.0.0 10/30/2018
Workqueue: events_unbound async_run_entry_fn
RIP: 0010:regulator_enable+0xe/0x60
Code: f9 ff ff 41 89 c4 85 c0 0f 89 14 ff ff ff e9 62 ff ff ff e9 28 29 00 00 0f 1f 40 00 0f 1f 44 00 00 55 48 89 fd 53 48 83 ec 20 <48> 8b 5f 78 65 48 8b 04 25 28 00 00 00 48 89 44 24 18 31 c0 48 89
RSP: 0018:ffffbf2b4202fe38 EFLAGS: 00010282
RAX: ffffffff8af48000 RBX: ffffa390197b0da0 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000007 RDI: 0000000000000000
RBP: 0000000000000000 R08: ffffa390189a8000 R09: ffffa3901710ca18
R10: 00000000000002eb R11: 0000000000000000 R12: ffffa3901e43a900
R13: 0000000000000000 R14: ffffa39018a813c0 R15: 0ffffa3901e43a90
FS:  0000000000000000(0000) GS:ffffa3901eb00000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 0000000000000078 CR3: 000000045ae86006 CR4: 00000000003606e0
DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Call Trace:
 ? try_to_wake_up+0x236/0x500
 regulator_bulk_enable_async+0x12/0x20
 async_run_entry_fn+0x37/0x140
 process_one_work+0x1eb/0x410
 worker_thread+0x2d/0x3d0
 ? process_one_work+0x410/0x410
 kthread+0x112/0x130
 ? kthread_park+0x80/0x80
 ret_from_fork+0x35/0x40
Modules linked in: intel_rapl(+) x86_pkg_temp_thermal intel_powerclamp coretemp snd_compress ac97_bus hci_uart snd_pcm_dmaengine kvm_intel kvmgt vfio_mdev mdev vfio_iommu_type1 vfio brcmutil snd_hda_intel kvm btqca btrtl btbcm snd_hda_codec btintel cfg80211 iTCO_wdt iTCO_vendor_support nls_iso8859_1 i2c_algo_bit snd_hda_core nls_cp437 bluetooth snd_hwdep vfat fat drm_kms_helper snd_pcm applesmc irqbypass input_polldev intel_cstate intel_uncore snd_timer intel_rapl_perf pcspkr drm joydev snd mousedev i2c_i801 input_leds mmc_core soundcore mei_me ecdh_generic thunderbolt(+) 8250_dw rfkill intel_gtt sbs intel_xhci_usb_role_switch roles agpgart mei syscopyarea acpi_als sysfillrect kfifo_buf sysimgblt idma64 sbshc fb_sys_fops industrialio evdev mac_hid pcc_cpufreq apple_bl ac facetimehd(OE+) videobuf2_dma_sg videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media ip_tables x_tables btrfs libcrc32c crc32c_generic xor raid6_pq algif_skcipher af_alg dm_crypt dm_mod
 crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 crypto_simd xhci_pci cryptd glue_helper xhci_hcd hid_apple hid applespi(OE) crc16 intel_lpss_pci intel_lpss spi_pxa2xx_platform
CR2: 0000000000000078
---[ end trace 8526d8c246aa145a ]---
```
As mentioned in https://github.com/Dunedan/mbp-2016-linux/issues/29#issuecomment-478151670, commenting out the call to `regulator_bulk_enable()` seems to fix this (and makes bluetooth work too).

Dunedan, 2019-04-02:
This bug got reported by @roadrunner2 upstream in https://bugzilla.kernel.org/show_bug.cgi?id=202963 and Chen-Yu Tsai provided a patch asking for feedback on LKML: https://lkml.org/lkml/2019/3/31/344

Dunedan, 2019-04-06:
Did somebody provide feedback to Chen-Yu? At least on LKML I see no response to his email. @kaloz maybe?

meeuw, 2019-04-13:
I've tested the patch and replied to Chen-Yu that it works and asked if he'd like me to do some specific tests.

jboyens, 2019-04-19:
I tested this patch as well, and while I didn't have problems booting into X, I wasn't able to actually power down my 14,3 on Linux 5.x. The system would hang and never complete shutdown.

This also fixes that issue.

meeuw, 2019-05-04:
Fix is now in bluetooth-next:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=62611abc8f37d00e3b0cff0eb2d72fa92b05fd27

meeuw, 2019-05-25:
Landed in 5.0.15 🎊  https://lwn.net/Articles/788144/

Dunedan, 2019-07-25:
@christophgysin Can you confirm that the commit @meeuw mentioned fixes it for you as well?

christophgysin, 2019-07-25:
Yes, this is fixed and can be closed.

closed 2019-07-25 by Dunedan