Dunedan/mbp-2016-linux #98: Initialization issues on mbp 14,1 wifi and battery status.

hashimda, 2019-04-05:
Most of the time, WiFi and battery status initialize correctly on the MBP 14,1 (Mid 2017 Macbook Pro 13 Inch), however, sometimes they don't. It is seemingly random as to when the WiFi and battery would want to initialize on boot-up. Generally the problem is solved by rebooting until both WiFi and battery status successfully initialize, however, I would like to solve this issue and have them both work successfully on every startup. I tried searching on the internet and this git to find if someone else has this issue, and I did find a vague forum post about how the system may load faster and before the WiFi and battery status successfully initialize. I am facing this issue on the distros i have tried on this laptop, such as Ubuntu, Arch Linux, Debian, Fedora and OpenSUSE. I believe the issue may be related to the system not waiting on the battery sensors and WiFi to work. Is anyone else facing this issue, and is there a solution to make the system wait on WiFi and the battery status to initialize before reaching the desktop? The closest thing I could find to a solution is the fedora systemd-udev-settle service. Is anyone else facing a similar issue?

roadrunner2, 2019-04-06:
While I don't have your system, I've not heard of anyone running into this, let alone ever seen it myself, But a few initial questions to clarify the behaviour you're seeing when they don't initialize properly:
- does `/sys/class/power_supply/` contain only an `ADP1` entry?
- does `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/` exist?
- does `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:8d/PNP0C09:00/ACPI0001:00/ACPI0002:00/` exist?
- what does `dmesg | grep SBS0` show?
- what does `dmesg | grep brcmfmac` show?
- what does `iwconfig` show?
- does `sudo rmmod brcmfmac ; sudo modprobe brcmfmac` cause the wifi to show up? If not, do the outputs of the above two commands change after doing this?

hashimda, 2019-04-06:
@roadrunner2
I will provide you with the output of these now:

- `/sys/class/power_supply/` both contain `ADP1` and `BAT0` (the system is loaded with both correctly initialized)

 - The directory `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/` exists.

 -  The directory `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:8d/PNP0C09:00/ACPI0001:00/ACPI0002:00/` does not exist . In fact there are no directories starting with the name 'device' in the directory `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/`

  - The output for `dmesg | grep SBS0` is `[   11.658784] ACPI: Smart Battery System [SBS0]: Battery Slot [BAT0] (battery present)`

 - The output for `dmesg | grep brcmfmac` is:
```
[   11.024134] usbcore: registered new interface driver brcmfmac
[   11.043029] brcmfmac 0000:02:00.0: enabling device (0000 -> 0002)`
[   11.151822] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5
[   11.154419] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt failed with error -2
[   11.154427] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2`
[   11.524124] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip `BCM4350/5`
[   11.524146] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.clm_blob failed with error -2
[   11.524148] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   11.525277] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015 03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6
[   11.594986] brcmfmac 0000:02:00.0 wlp2s0: renamed from wlan0
```

-  The output for `iwconfig` is: 
```
lo        no wireless extensions.

wlp2s0    IEEE 802.11  ESSID:"network"  
         ` Mode:Managed  Frequency:2.412 GHz  Access Point: 10:0D:7F:D4:C1:AB   `
          Bit Rate=52 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=70/70  Signal level=-36 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
`
- As for `sudo rmmod brcmfmac ; sudo modprobe brcmfmac`, I will need to try to reproduce this issue, I will try and report. 

I am using Debian Unstable GNU/Linux with the latest Linux kernel (5.0.5).

roadrunner2, 2019-04-06:
> @roadrunner2
> I will provide you with the output of these now:

Ok, that gives us a base, thanks. Please also provide them after you've reproduced the issue.

>     * The directory `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:8d/PNP0C09:00/ACPI0001:00/ACPI0002:00/` does not exist . In fact there are no directories starting with the name 'device' in the directory `/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/`

Well, that's certainly odd: according to the dsdt for 14,1 that's supposed to be the PCI root device. What is the output of these two?
```
ls -l /sys/class/power_supply/BAT0
cat /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/path
```

hashimda, 2019-04-06:
@roadrunner2 
The output for these two commands are: 

- The output for `ls -l /sys/class/power_supply/BAT0` is:
`lrwxrwxrwx 1 root root 0 Apr  6 14:21 /sys/class/power_supply/BAT0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:85/PNP0C09:00/ACPI0001:00/ACPI0002:00/power_supply/BAT0
`

- The output for `cat /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/path` is:
`\_SB_.PCI0`

hashimda, 2019-04-06:
@roadrunner2 
The output for these two commands are: 

- The output for `ls -l /sys/class/power_supply/BAT0` is:
lrwxrwxrwx 1 root root 0 Apr  6 14:21 /sys/class/power_supply/BAT0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:85/PNP0C09:00/ACPI0001:00/ACPI0002:00/power_supply/BAT0

- The output for `cat /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/path` is:
`\_SB_.PCI0`

roadrunner2, 2019-04-06:
@hashimda Now I'm even more confused, since BAT0 links to a device you say doesn't exist. So just to confirm: I presume the following gives an error then?
```
ls /sys/class/power_supply/BAT0/
```
(note the trailing slash)

hashimda, 2019-04-06:
@roadrunner2 I believe the issue may be that you wrote that the device was device8d, while it was device85, so it was not found. Also, the output of `ls /sys/class/power_supply/BAT0/` is:
```
alarm               current_avg   model_name  technology          voltage_now
capacity            current_now   power       temp
charge_full         cycle_count   present     type
charge_full_design  device        status      uevent
charge_now          manufacturer  subsystem   voltage_min_design

```
which includes a directory named devices, I do apologize for the inconsistency, I did type ls on that folder before without recalling a devices folder being there. Anyway, I have reset my PRAM/NVRAM before posting this issue on Github, and I haven't been able to reproduce the wifi issue since, however the battery detection issue has come up again once. Thank you for your time, @roadrunner2

roadrunner2, 2019-04-06:
@hashimda Ah, this make a lot more sense. Thanks. Sorry for the confusion on the `device8d` part - I just copied the path as seen on my system, but should've realized those `device<xx>` parts may differ since they are runtime generated names.

Anyway, now we know what to look for next time you can reproduce the issue.

hashimda, 2019-04-07:
@roadrunner2 I have reproduced the issue, and, running `sudo rmmod brcmfmac` then running `sudo modprobe brcmfmac` did not fix the issue, and running them again did not produce any output, none of these commands produced any output.

roadrunner2, 2019-04-07:
@hashimda So the output of `dmesg | grep brcmfmac` is empty? And to clarify, both the wifi and the battery status are not working right now?

What is the output of the following commands now?
```sh
dmesg | grep SBS0
ls -l /sys/class/power_supply/
ls -l /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/
ls -l /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:85/PNP0C09:00/ACPI0001:00/ACPI0002:00/
```
If that last directory does not exist, what parent directory does?

roadrunner2, 2019-04-07:
Oh, and what is the output of
```sh
lspci | grep -i network
```

hashimda, 2019-04-07:
I have restarted my macbook unfortunately, and the wifi is back, however the battery is uninitialized, so, I will send you some outputs (with the battery uninitialized)

Running  `dmesg | grep SBS0` yields no output.
Running `ls -l /sys/class/power_supply/` yields the output:
```
total 0
lrwxrwxrwx 1 root root 0 Apr  7 13:39 ADP1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ADP1

```
Running `ls -l /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/` yields the output:
```
total 0
-r--r--r-- 1 root root 4096 Apr  7 13:44 hid
-r--r--r-- 1 root root 4096 Apr  7 13:44 modalias
-r--r--r-- 1 root root 4096 Apr  7 13:44 path
drwxr-xr-x 2 root root    0 Apr  7 13:44 power
-r--r--r-- 1 root root 4096 Apr  7 13:44 status
lrwxrwxrwx 1 root root    0 Apr  7 13:39 subsystem -> ../../../../bus/acpi
-rw-r--r-- 1 root root 4096 Apr  7 13:39 uevent
-r--r--r-- 1 root root 4096 Apr  7 13:44 uid
```
Running `ls -l /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:85/PNP0C09:00/ACPI0001:00/ACPI0002:00/` yields: 
total 0
```
-r--r--r-- 1 root root 4096 Apr  7 13:44 hid
-r--r--r-- 1 root root 4096 Apr  7 13:44 modalias
-r--r--r-- 1 root root 4096 Apr  7 13:44 path
lrwxrwxrwx 1 root root    0 Apr  7 13:44 physical_node -> ../../../../../../../pci0000:00/0000:00:1f.0/PNP0C09:00/ACPI0001:00/ACPI0002:00
drwxr-xr-x 2 root root    0 Apr  7 13:44 power
lrwxrwxrwx 1 root root    0 Apr  7 13:39 subsystem -> ../../../../../../../../bus/acpi
-rw-r--r-- 1 root root 4096 Apr  7 13:39 uevent

```
Just to clarify, on my last boot, when I had the issue reproduced, when I tested `sudo rmmod brcmfmac` and `sudo modprobe brcmfmac`, both WiFi and battery didn't work, however I have rebooted unfortunately and now I am sending you commands from this session where the battery indicator doesn't work but WiFi does.

The ouput of `lspci | grep -i network` on this session is:
`02:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4350 802.11ac Wireless Network Adapter (rev 05)`

roadrunner2, 2019-04-07:
Are the `sbs` and `sbshc` modules loaded? (I'm assuming these are built as modules in your kernel)
```sh
lsmod | grep sbs
```
If so, does reloading `sbs` (`sudo rmmod sbs ; sudo modprobe sbs`) make the battery appear in `/sys/class/power_supply/`?

hashimda, 2019-04-17:
@roadrunner2 
The sbs and sbshc modules are loaded. The output of `lsmod | grep sbs` is:

```
sbs                    20480  0
sbshc                  16384  1 sbs

```
Reloading `sbs`, by typing `sudo rmmod sbs ; sudo modprobe sbs` actually makes the battery show up again and makes it work perfectly fine again! It works!

roadrunner2, 2019-04-19:
> @roadrunner2
> The sbs isn't loaded, but the sbshc modules is loaded. The output of `lsmod | grep sbs` is:
> 
> ```
> sbs                    20480  0
> sbshc                  16384  1 sbs
> ```

Actually, this indicates `sbs` is in fact loaded.

> Reloading `sbs`, by typing `sudo rmmod sbs ; sudo modprobe sbs` actually makes the battery show up again and makes it work perfectly fine again! It works!

Great. So this appears to indicate some sort of race during boot, though I don't know why.

> Perhaps the solution to this problem is to add sbs to `/etc/initramfs-tools/modules` and then update initramfs?

I doubt it, as it seems to me more like `sbs` is being loaded too early sometimes, but worth a try I guess.

Dunedan, 2019-09-16:
@hashimda Do you still experience this issue with a recent kernel version?

hashimda, 2019-09-17:
@Dunedan There are no issues with initialization or crashing on recent kernel versions so far, it's been working perfectly. I am using the 5.2.14 kernel on Arch Linux, though all the 5.2.x kernels seem to work well. This issue can be closed.

closed 2019-09-17 by hashimda