Dunedan/mbp-2016-linux #99: MacBook 12, 2017

marbetschar, 2019-04-07:
How to determine which of this is applicable for the MacBook 12", 2017...? I already got Linux running on this machine, but only with the bare minimum: Display, Trackpad and Keyboard.

Love to get Bluetooth up & running as a next step. How to figure out which exact hardware device is builtin and where/how to check if there is any driver/firmware available?

Would love to get some pointers to head into the right direction.

FWIW: I'm currently running a [custom spin of Manjaro Linux](https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12), with the `macbook12-apple-spi` driver preinstalled.

leifliddy, 2019-04-07:
@marbetschar there's already an open thread for bluetooth issues, please continue the discussion there
https://github.com/Dunedan/mbp-2016-linux/issues/29

To summarize what the current status is:
bluetooth should work out-of-the-box with on the MacBook10,1 on kernels 4.20 and below.

Just make sure your kernel config includes the following:
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y

For kernel 5.x, you'll need to apply this patch:
https://lkml.org/lkml/2019/3/31/344

**no firmware needs to be loaded for this bluetooth device

--

marbetschar, 2019-04-07:
@leifliddy thanks for the quick response! Love to try the provided kernel config ... but ince I've never done that before I have no clue where to put it :speak_no_evil: 

Any chance you can get me a brief of explanation of the steps need to be taken?

leifliddy, 2019-04-07:
@marbetschar first find which kernel you are currently using. 
Could you post the output of 'uname -r'
You'll find the corresponding kernel config under /boot
Search this file for those kernel config options

ie
[root@black ~]# uname -r
5.0.5-200.fc29.x86_64

[root@black ~]# grep -E 'CONFIG_BT_HCIUART_SERDEV|CONFIG_BT_HCIUART_BCM'  /boot/config-5.0.5-200.fc29.x86_64 
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y

marbetschar, 2019-04-07:
Kernel:

```
# uname -r
4.20.17-1-MANJARO
```
But I can't find any corresponding config file:

```
# ls -l /boot
insgesamt 51668
drwxr-xr-x 5 root root     4096  1. Jan 1970  efi
drwxr-xr-x 5 root root     4096  7. Apr 17:38 grub
-rw-r--r-- 1 root root 32306719  6. Apr 14:13 initramfs-4.20-x86_64-fallback.img
-rw-r--r-- 1 root root 12398380  6. Apr 14:13 initramfs-4.20-x86_64.img
-rw-r--r-- 1 root root  2219520 26. Mär 12:11 intel-ucode.img
-rw-r--r-- 1 root root       22 19. Mär 19:07 linux420-x86_64.kver
drwxr-xr-x 2 root root     4096  6. Apr 00:55 memtest86+
-rw-r--r-- 1 root root      709  6. Apr 15:01 refind_linux.conf
-rw-r--r-- 1 root root  5957504 19. Mär 19:07 vmlinuz-4.20-x86_64
```
Does it work if I just add a new file with the flags in it?

leifliddy, 2019-04-07:
sorry, I'm familiar with Manjaro.
Ok, try this:
```
cat /lib/modules/$(uname -r)/build/.config
```
does this file exist?

marbetschar, 2019-04-07:
Ideed, it does:

```
$ egrep 'CONFIG_BT_HCIUART_SERDEV|CONFIG_BT_HCIUART_BCM' /lib/modules/$(uname -r)/build/.config
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
```

FWIW: In GNOME settings I'm able to switch bluetooth on. But next time I'm going to the very same panel it's again switched off:

[image]

leifliddy, 2019-04-07:
could you post the output of the following two commands?

hciconfig 
dmesg | egrep -i 'bluetooth|hci_uart| dw-apb-uart|brcmfmac|8250|ttys'

marbetschar, 2019-04-07:
```
hciconfig
bash: hciconfig: Command not found.
```
(seems to be deprecated. Seems it's no longer in `bluez-utils` and I was not able to install `bluez-utils-compat` due to a PGP verification error)

```
dmesg | egrep -i 'bluetooth|hci_uart| dw-apb-uart|brcmfmac|8250|ttys'
[    1.006334] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
[    1.015331] sched_clock: Marking stable (1018250819, -2931442)->(1027618675, -12299298)
[    6.692869] Bluetooth: Core ver 2.22
[    6.692887] Bluetooth: HCI device and connection manager initialized
[    6.692892] Bluetooth: HCI socket layer initialized
[    6.692894] Bluetooth: L2CAP socket layer initialized
[    6.692897] Bluetooth: SCO socket layer initialized
[    6.793133] dw-apb-uart.2: ttyS0 at MMIO 0x9282b000 (irq = 21, base_baud = 115200) is a 16550A
[    6.793218] serial serial0: tty port ttyS0 registered
[    6.809703] dw-apb-uart.3: ttyS1 at MMIO 0x9282c000 (irq = 20, base_baud = 3000000) is a 16550A
[    6.809786] serial serial1: tty port ttyS1 registered
[    6.946732] Bluetooth: HCI UART driver ver 2.3
[    6.946734] Bluetooth: HCI UART protocol H4 registered
[    6.946735] Bluetooth: HCI UART protocol BCSP registered
[    6.946751] Bluetooth: HCI UART protocol LL registered
[    6.946752] Bluetooth: HCI UART protocol ATH3K registered
[    6.946765] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    6.946797] Bluetooth: HCI UART protocol Intel registered
[    6.946834] Bluetooth: HCI UART protocol Broadcom registered
[    6.946846] Bluetooth: HCI UART protocol QCA registered
[    6.946847] Bluetooth: HCI UART protocol AG6XX registered
[    6.946848] Bluetooth: HCI UART protocol Marvell registered
[    6.962933] hci_uart_bcm serial1-0: Unexpected ACPI gpio_int_idx: -1
[    6.962936] hci_uart_bcm serial1-0: Unexpected number of ACPI GPIOs: 0
[    6.962940] hci_uart_bcm serial1-0: No reset resource, using default baud rate
[    7.192802] usbcore: registered new interface driver brcmfmac
[    7.325659] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5
[    7.328776] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2
[    7.628600] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5
[    7.628623] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.clm_blob failed with error -2
[    7.628625] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    7.629123] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015 03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6
[    7.662052] brcmfmac 0000:02:00.0 wlp2s0: renamed from wlan0
[    9.176098] Bluetooth: hci0: command 0xfc18 tx timeout
[    9.214055] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.214056] Bluetooth: BNEP filters: protocol multicast
[    9.214060] Bluetooth: BNEP socket layer initialized
[   17.069640] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   17.069645] Bluetooth: hci0: Failed to set baudrate
[   19.202953] Bluetooth: hci0: command 0x0c03 tx timeout
[   27.096214] Bluetooth: hci0: BCM: Reset failed (-110)
```

leifliddy, 2019-04-07:
Damn, I had no idea that hciconfig was deprecated, I normally use bluetoothctl anyway. I was looking for a simple way to see if the bluetooth device was registered. I can see in the dmesg output, that it's not. 
Is your system using grub to boot and are you using an HFS+ ESP partition
could you post the output of the following commands:
gdisk /dev/nvme0n1 -l
efibootmgr -v

Here's what they look like on my system:
The EF00 partition on my system is not actually being used. The next time I reinstall linux on this system, I'll remove it. The AF00 partition in mounted under /boot/efi
```
Number  Start (sector)    End (sector)  Size       Code  Name
   1             256           51455   200.0 MiB   EF00  
   2           51456          102655   200.0 MiB   AF00  Linux HFS+ ESP
   3          102656          364799   1024.0 MiB  8300  
   4          364800         2413823   7.8 GiB     8200  
   5         2413824       122138111   456.7 GiB   8300 
```
```
[root@mac ~]# efibootmgr -v
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0002,0080,0001
Boot0000* Fedora	HD(2,GPT,90ceb047-008b-43be-803c-4b3eb18f0786,0xc900,0xc800)/File(\EFI\fedora\shimx64.efi)
```

marbetschar, 2019-04-07:
It's a dualboot environment with macOS on an encrypted partition. Also installed Linux on an encrypted LUKS partition.

```
$ sudo gdisk /dev/nvme0n1 -l
GPT fdisk (gdisk) version 1.0.4

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
Disk /dev/nvme0n1: 122138133 sectors, 465.9 GiB
Model: APPLE SSD AP0512J                       
Sector size (logical/physical): 4096/4096 bytes
Disk identifier (GUID): E64F8F55-87E0-4C03-98E8-6FD467DA5483
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 5
First usable sector is 6, last usable sector is 122138127
Partitions will be aligned on 1-sector boundaries
Total free space is 19 sectors (76.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1               6           76805   300.0 MiB   EF00  EFI System Partition
   2           76806        73259366   279.2 GiB   AF0A  NoName
   3        73259367       102523388   111.6 GiB   8300  
   4       102523392       122105343   74.7 GiB    0700  
   5       122105360       122138127   128.0 MiB   AB00  Booter
```

```
$ efibootmgr -v
BootCurrent: 0001
BootOrder: 0001,0000
Boot0000* Manjaro	HD(1,GPT,cdd0289c-ec1a-4167-933a-615e4aa98245,0x6,0x12c00)/File(\EFI\Manjaro\grubx64.efi)
Boot0001* rEFInd Boot Manager	HD(1,GPT,cdd0289c-ec1a-4167-933a-615e4aa98245,0x6,0x12c00)/File(\EFI\refind\refind_x64.efi)
Boot0081* Recovery OS	PciRoot(0x0)/Pci(0x1c,0x0)/Pci(0x0,0x0)//HD(2,GPT,24e9ecac-7dc3-4791-af34-981c1481dad6,0x12c06,0x45d964b)/VenMedia(be74fcf7-0b7c-49f3-9147-01f4042e6842,d7b8b164425fe446b9533f249428580f)/File(\836F48FA-091E-45A2-9ED9-DF2891294479\boot.efi)
Boot0082* 	PciRoot(0x0)/Pci(0x1c,0x0)/Pci(0x0,0x0)//HD(2,GPT,24e9ecac-7dc3-4791-af34-981c1481dad6,0x12c06,0x45d964b)/VenMedia(be74fcf7-0b7c-49f3-9147-01f4042e6842,46c84fc81b985742b6a6ef7ac029fbd5)/File(\836F48FA-091E-45A2-9ED9-DF2891294479\System\Library\CoreServices\boot.efi)
BootFFFF* 	PciRoot(0x0)/Pci(0x1c,0x0)/Pci(0x0,0x0)//HD(3,GPT,33af8ae5-0810-4b3f-8c33-98ce5f6617d6,0x5ef2e10,0xef8000)/File(\System\Library\CoreServices\boot.efi)
```

leifliddy, 2019-04-07:
Hmmm...I'm wondering what would happen if you tried changing the boot order to 0000,0001
efibootmgr -o 0000,0001

The use of rEFInd is the one thing that stands out as being different from my system. Although, there could be something different with the Macbook10,1 models vs the Macbook8,1 + 9,1 models.
Just to be sure, you could also try booting Fedora 29 from a usb stick, bluetooth should just work with kernels 4.20 and below.

marbetschar, 2019-04-07:
I can't think of why rEFInd would cause a bluetooth issue. I can see the point of figuring out the system differences, but from my point of view I'd prefer not tinkering with the partitions to get bluetooth working ;)

One point of confusion might be the fact, that my MacBook is refered to as "MacBook 10,1 Retina" - it is not a MacBook Pro, just a regular MacBook 12-inch from 2017. From what I know there seems to be a MacBook Pro out there actually beeing a "MacBookPro 10,1" from mid 2012. Maybe this is what causes the confusion?

marbetschar, 2019-04-07:
@leifliddy I've read through the https://github.com/Dunedan/mbp-2016-linux/issues/29# a bit. From the current point of work, what is the suggested path forward in terms of the kernel version? Shall I upgrade to v5.x, or use another version?

Currently I'm using 4.20, because it was the only one which allowed me to install the system due to the [removal of the cipher aes-xts-plain-64](https://forum.manjaro.org/t/gnome-installer-crashes-on-my-custom-spin/81827/6).

But as 4.20 is EOL anyways it may not make a lot of sense putting any more effort in this kernel version?

leifliddy, 2019-04-07:
I have a macbook9,1 (12 inch 2016), and bluetooth has been working fine for ages. 

If you look at this thread https://github.com/Dunedan/mbp-2016-linux/issues/29
 teancum has a macbook9,1 but couldn't get bluetooth working with Manjaro, but he did get it working with Fedora. It seems that there is something off with the way that Manjaro is setup. 

And the one glaring difference between Manjaro and Fedora is the lack of the AF00 (HFS+ ESP) efi partition and the use of rEFInd vs grub. Although rEFInd would be the key suspect. 

I would try booting from a Fedora 29 usb stick and see if that works. That would let us know whether the issue lies with the Manjaro or whether it's a Macbook10,1 driver issue. 

Or you could try changing the boot order so that grub boots by default (Boot0000). The worst case scenario is that the system doesn't boot (but I doubt that would happen) In any case it's fairly easy to boot from a rescue usb stick and change the boot order back.

leifliddy, 2019-04-07:
@marbetschar For the sake of being thorough could check to see if these kernel options are configured?
```
cat /lib/modules/$(uname -r)/build/.config | egrep 'CONFIG_SERIAL_8250_LPSS|CONFIG_BT_HCIUART_BCM|CONFIG_BT_HCIUART_SERDEV|CONFIG_SERIAL_DEV_BUS|CONFIG_SERIAL_DEV_CTRL_TTYPORT|CONFIG_SERIAL_8250_DW'
 ```

leifliddy, 2019-04-07:
@marbetschar I might have some time this week to install Manjaro on my Macbook9,1 and play around with it.

marbetschar, 2019-04-07:
```
$ cat /lib/modules/$(uname -r)/build/.config | egrep 'CONFIG_SERIAL_8250_LPSS|CONFIG_BT_HCIUART_BCM|CONFIG_BT_HCIUART_SERDEV|CONFIG_SERIAL_DEV_BUS|CONFIG_SERIAL_DEV_CTRL_TTYPORT|CONFIG_SERIAL_8250_DW'
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_SERIAL_8250_DW=m
CONFIG_SERIAL_8250_LPSS=y
CONFIG_SERIAL_DEV_BUS=y
CONFIG_SERIAL_DEV_CTRL_TTYPORT=y
```

leifliddy, 2019-04-07:
@marbetschar That's exactly what I see on my system

marbetschar, 2019-04-07:
@leifliddy if you're able to test Manjaro, this would be awesome! If you like, you can use one of the isos I've built to get a quick install: https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12

Just downloaded Fedora29 and will test it right now - but I suspect the keyboard and trackpad won't work since the apple-spi driver is WIP, and I do not have any usb keyboard or mouse available right now.

leifliddy, 2019-04-07:
Hmmm....some people have stated in that they needed to change 
CONFIG_SERIAL_8250_DW=m to CONFIG_SERIAL_8250_DW=y to get bluetooth to work. 
I need to research this a bit more....

marbetschar, 2019-04-07:
FWIW: I've installed the bcm4350-firmware using `yay` after the initial install. You can find the details here: https://forum.manjaro.org/t/macbook-12-2017-challenge-the-quest-for-drivers/82098/19?u=marbetschar

@leifliddy thank you very much for giving a hand! I highly appreciate your efforts.

leifliddy, 2019-04-07:
@marbetschar so bluetooth is working now?!?

marbetschar, 2019-04-07:
@leifliddy no its not 🤣 just wanted to say thank you for your time 😄

leifliddy, 2019-04-07:
@marbetschar I was going to say...that this is a hci uart bluetooth device and doesn't require any firmware. The functions are derived from acpi methods. Cool, no worries hopefully we can figure this out : )

leifliddy, 2019-04-07:
@marbetschar 
could post the output of the following commands?
lsmod | grep 8250
lsinitrd /boot/initramfs-4.20-x86_64.img | grep 8250

marbetschar, 2019-04-07:
```
$ lsmod | grep 8250
8250_dw                28672  0
```
@leifliddy `lsinitrd` seems not to be installed. Which package does contain this command?

leifliddy, 2019-04-07:
@marbetschar 
dracut

marbetschar, 2019-04-07:
@leifliddy dracut seems not to be available for Arch:

- https://www.reddit.com/r/archlinux/comments/3ejvn0/whats_mkinitcpio_and_is_it_needed_when_installing/ctfna6z/
- https://wiki.archlinux.org/index.php?title=Dracut&redirect=no

leifliddy, 2019-04-07:
hmm...maybe it's called lsinitcpio for Manjaro

marbetschar, 2019-04-07:
`$ lsinitcpio /boot/initramfs-4.20-x86_64.img | grep 8250` does not output anything

leifliddy, 2019-04-07:
Interesting, I would try generating a new initramfs-4.20-x86_64.img and adding the 8250_dw module to it. 
With Manjaro, you would use mkinitcpio to do this. 8250_dw seems to be a pretty important module related to the Macbook's bluetooth device. If that adding the module to initramfs doesn't work, I would try compiling a new kernel and include the 8250_dw module in the kernel (CONFIG_SERIAL_8250_DW=y). I'll play around with Manjaro tomorrow....

marbetschar, 2019-04-08:
@leifliddy just booted Fedora 29 and there Bluetooth seems to work. At least GNOME is reporting the Bluetooth state as "On".

marbetschar, 2019-04-08:
Added the 8250_dw module to the initramfs, but it did not help in any way. Here's what I've done:

Deinstalled the not needed firmware:

```
$ sudo pacman -R bcm4350-firmware
```

Added module to */etc/mkinitcpio.conf*:

```
...
MODULES="8250_dw"
...
```

Built initramfs:

```
$ sudo mkinitcpio -P
==> Building image from preset: /etc/mkinitcpio.d/linux420.preset: 'default'
  -> -k /boot/vmlinuz-4.20-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-4.20-x86_64.img
==> Starting build: 4.20.17-1-MANJARO
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [block]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [encrypt]
  -> Running build hook: [filesystems]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-4.20-x86_64.img
==> Image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux420.preset: 'fallback'
  -> -k /boot/vmlinuz-4.20-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-4.20-x86_64-fallback.img -S autodetect
==> Starting build: 4.20.17-1-MANJARO
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [modconf]
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: wd719x
==> WARNING: Possibly missing firmware for module: aic94xx
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [encrypt]
  -> Running build hook: [filesystems]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-4.20-x86_64-fallback.img
==> Image generation successful
```
Checked if module is in initramfs:

```
$ lsinitcpio /boot/initramfs-4.20-x86_64.img | grep 8250
usr/lib/modules/4.20.17-1-MANJARO/kernel/8250_dw.ko
```

Rebooted the machine and tested Bluetooth in GNOME settings. Also looked at the dmesg output: the baudrate error is still there:

```
$ dmesg |grep Bluetooth
[    7.213117] Bluetooth: Core ver 2.22
[    7.213606] Bluetooth: HCI device and connection manager initialized
[    7.213611] Bluetooth: HCI socket layer initialized
[    7.213613] Bluetooth: L2CAP socket layer initialized
[    7.213618] Bluetooth: SCO socket layer initialized
[    7.366822] Bluetooth: HCI UART driver ver 2.3
[    7.366824] Bluetooth: HCI UART protocol H4 registered
[    7.366825] Bluetooth: HCI UART protocol BCSP registered
[    7.366841] Bluetooth: HCI UART protocol LL registered
[    7.366842] Bluetooth: HCI UART protocol ATH3K registered
[    7.366851] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    7.366885] Bluetooth: HCI UART protocol Intel registered
[    7.366922] Bluetooth: HCI UART protocol Broadcom registered
[    7.366932] Bluetooth: HCI UART protocol QCA registered
[    7.366933] Bluetooth: HCI UART protocol AG6XX registered
[    7.366934] Bluetooth: HCI UART protocol Marvell registered
[    9.130544] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.130546] Bluetooth: BNEP filters: protocol multicast
[    9.130549] Bluetooth: BNEP socket layer initialized
[    9.437501] Bluetooth: hci0: command 0xfc18 tx timeout
[   17.544208] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   17.544213] Bluetooth: hci0: Failed to set baudrate
[   19.677517] Bluetooth: hci0: command 0x0c03 tx timeout
[   27.574141] Bluetooth: hci0: BCM: Reset failed (-110)
```

Will try to build a new Kernel now....

marbetschar, 2019-04-08:
@leifliddy just stumpled accross the `CONFIG_SPI_DW_PCI` Kernel config. Do you think this is needed? Just looked familiar in terms of the whole SPI, DWS stuff :)

As of now it is set like follows in my system - what's the value on your end with working Bluetooth?

```
$ cat /lib/modules/$(uname -r)/build/.config | grep 'CONFIG_SPI_DW_PCI'
CONFIG_SPI_DW_PCI=m
```

BTW: Kernel with the flags `CONFIG_SERIAL_8250_DW=y` and `CONFIG_SERIAL_8250_LPSS=m` is compiling now...

leifliddy, 2019-04-08:
Good work so far, your're moving pretty fast! I don't see 'CONFIG_SPI_DW_PCI' in any of my (Fedora) 4.20 or 5.0 kernel configs.

marbetschar, 2019-04-09:
@leifliddy compiled the kernel and installed it. Unfortunately Bluetooth is still not working :(
After booting I can see the following output:

```
$ uname -a
Linux Shira 4.20.17-1-MANJARO #1 SMP PREEMPT Mon Apr 8 20:28:14 UTC 2019 x86_64 GNU/Linux
```

```
$ cat /lib/modules/$(uname -r)/build/.config | egrep 'CONFIG_SERIAL_8250_LPSS|CONFIG_BT_HCIUART_BCM|CONFIG_BT_HCIUART_SERDEV|CONFIG_SERIAL_DEV_BUS|CONFIG_SERIAL_DEV_CTRL_TTYPORT|CONFIG_SERIAL_8250_DW'
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_SERIAL_8250_DW=y
CONFIG_SERIAL_8250_LPSS=m
CONFIG_SERIAL_DEV_BUS=y
CONFIG_SERIAL_DEV_CTRL_TTYPORT=y
```

```
$ dmesg |grep Blue
[    7.365295] Bluetooth: Core ver 2.22
[    7.365314] Bluetooth: HCI device and connection manager initialized
[    7.365319] Bluetooth: HCI socket layer initialized
[    7.365321] Bluetooth: L2CAP socket layer initialized
[    7.365324] Bluetooth: SCO socket layer initialized
[    7.566516] Bluetooth: HCI UART driver ver 2.3
[    7.566518] Bluetooth: HCI UART protocol H4 registered
[    7.566519] Bluetooth: HCI UART protocol BCSP registered
[    7.566533] Bluetooth: HCI UART protocol LL registered
[    7.566534] Bluetooth: HCI UART protocol ATH3K registered
[    7.566545] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    7.566567] Bluetooth: HCI UART protocol Intel registered
[    7.566607] Bluetooth: HCI UART protocol Broadcom registered
[    7.566626] Bluetooth: HCI UART protocol QCA registered
[    7.566627] Bluetooth: HCI UART protocol AG6XX registered
[    7.566628] Bluetooth: HCI UART protocol Marvell registered
[    9.672212] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.672214] Bluetooth: BNEP filters: protocol multicast
[    9.672218] Bluetooth: BNEP socket layer initialized
[    9.815791] Bluetooth: hci0: command 0xfc18 tx timeout
[   17.709153] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   17.709158] Bluetooth: hci0: Failed to set baudrate
[   19.842570] Bluetooth: hci0: command 0x0c03 tx timeout
[   27.735879] Bluetooth: hci0: BCM: Reset failed (-110)
```

marbetschar, 2019-04-09:
Seems Bluetooth is finally working!!! :tada: :tada: 

For this, I compiled a custom Kernel - based on the [Manjaro `linux420`](https://gitlab.manjaro.org/packages/core/linux420).

I changed set flags in `config.x86_64` like:

```
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_SERIAL_8250_DW=y
CONFIG_SERIAL_8250_LPSS=y
CONFIG_SERIAL_DEV_BUS=y
CONFIG_SERIAL_DEV_CTRL_TTYPORT=y
```

And applied the patch of christophgysin: https://github.com/christophgysin/linux/commit/b8c43916bafe6dc8de1aed5ae1cbc76cf7bad374

After compiling and booting with the new kernel, Bluetooth state was shown as "On" and I was able to connect to my iPhone.

@leifliddy thank you so much for your help!! Need now to figure out a way to get Bluetooth running on the LTS Kernel 4.19....

leifliddy, 2019-04-09:
@marbetschar That is excellent work, really impressed that you got it working!!!
I'm wondering what specific action was responsible for bluetooth to start working. The macbook 10,1 doesn't use the  8250_lpss module (CONFIG_SERIAL_8250_LPSS), so the kernel options are essentially the same as you were using on the previous attempt. 

I'm wondering if the patch alone was responsible. There have been a number of complaints listed in https://github.com/Dunedan/mbp-2016-linux/issues/29
where the bluetooth controller doesn't always come online at boot time on the macbook 12 inch models. 
The dmesg logs are almost identical:

```
[    7.978779] Bluetooth: hci0: command 0xfc18 tx timeout
[   16.042789] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   16.042793] Bluetooth: hci0: Failed to set baudrate
[   18.090886] Bluetooth: hci0: command 0x0c03 tx timeout
[   26.282855] Bluetooth: hci0: BCM: Reset failed (-110)
```

I wondering if applying this patch would help at all. I'm going to suggest that to teancum...

marbetschar, 2019-04-09:
Thank you for the kind words 🙏🏼 

Mhm, those are exactly my thoughts. Wondering if applying the patch would be enough.

Will test these assumptions against a newer Manjaro Kernel (4.19) - but there I think another patch should be applied?

leifliddy, 2019-04-09:
Do you still have the previous kernel installed? Do you know you how to patch hci_bcm.c and then build the hci_uart.ko module?
If not, I can give you clear instructions on how to do it...

marbetschar, 2019-04-09:
@leifliddy which kernel do you mean with "previous"...? I still have the linux420 kernel with the flags and patch applied as described [here](https://github.com/Dunedan/mbp-2016-linux/issues/99#issuecomment-481267741). Do you mean this one?

Not sure if I know how to create a patch, but I know how to apply an existing patch [like this](https://github.com/christophgysin/linux/commit/b8c43916bafe6dc8de1aed5ae1cbc76cf7bad374.patch).

What's the goal?

marbetschar, 2019-04-09:
Got an idea which _may_ get us closer in tracking this down:

**As of now we know:**

- Bluetooth works out of the box in Fedora29
- In Manjaro Bluetooth it doesn't

First conclusion: Either Manjaro applies some configuration/patch/whatever which breaks Bluetooth - or Fedora29 does so to fix Bluetooth.

**Furthermore we assume:**

1. `CONFIG_SERIAL_8250_LPSS` is not needed for Bluetooth
2. `CONFIG_SERIAL_8250_DW` is needed for Bluetooth, but configure it as `m` is probably enough
3. A patch needs to be applied in order to get Bluetooth working

**Based upon this we declare the next actions:**
The assumptions 1. & 2. are easy to test for me. I just need some time to properly compiling and testing the kernels.

@leifliddy for 3. I'm out. Don't know the inner workings well enough to track this down. If you have time to help out on this, this would be awesome!!

My theory here is: If there's a patch which needs to be applied, this patch was applied to Fedora29. To figure out which patch this is, we should be able to compare the repos. As long as they are targeting the same base kernel version we should be able to pinpoint in which patch the magic happens right?

Here are the source codes of the corresponding kernels:

- [Fedora29 Kernel repository (Kernel v4.18.16, according to Wikipedia)](https://src.fedoraproject.org/rpms/kernel/tree/f29)
- [Manjaro linux418](https://gitlab.manjaro.org/packages/core/linux418)

Not quite sure if all of those assumptions I've made are correct. Just a best guess based upon my current knowledge.

leifliddy, 2019-04-09:
Can you try this compiling the hci_uart  kernel module for the 4.20.17-1 kernel?
First, install the linux-headers package for the 4.20.17-1-MANJARO kernel
Then download the kernel source tarball from https://www.kernel.org/
```
wget https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.20.17.tar.xz
```
extract the drivers/bluetooth directory
cd to bluetooth directory
backup the hci_bcm.c file (optional)
```
cp  hci_bcm.c hci_bcm.c.orig
```
replace the Makefile with this one
```
obj-m 		+= hci_uart.o
hci_uart-y	:= hci_ldisc.o
hci_uart-m 	+= hci_serdev.o
hci_uart-m 	+= hci_h4.o
hci_uart-m	+= hci_bcsp.o
hci_uart-m	+= hci_ll.o
hci_uart-m	+= hci_ath.o
hci_uart-m	+= hci_h5.o
hci_uart-m	+= hci_intel.o
hci_uart-m	+= hci_bcm.o
hci_uart-m	+= hci_qca.o
hci_uart-m	+= hci_ag6xx.o
hci_uart-m	+= hci_mrvl.o
hci_uart-objs	:= $(hci_uart-y)

all:
	make -C /lib/modules/$(shell uname -r)/build M=$(PWD) modules

clean:
	make -C /lib/modules/$(shell uname -r)/build M=$(PWD) clean
```
put the contents of this patch in a file called patchfile
https://github.com/christophgysin/linux/commit/b8c43916bafe6dc8de1aed5ae1cbc76cf7bad374.patch
now patch hci_bcm.c

```
patch < patchfile
```
compile hci_uart.ko module
```
[root@mac bluetooth]# make
make -C /lib/modules/4.20.16-200.fc29.x86_64/build M=/root/4.20.bluetooth modules
make[1]: Entering directory '/usr/src/kernels/4.20.16-200.fc29.x86_64'
  CC [M]  /root/4.20.bluetooth/hci_ldisc.o
  CC [M]  /root/4.20.bluetooth/hci_serdev.o
  CC [M]  /root/4.20.bluetooth/hci_h4.o
  CC [M]  /root/4.20.bluetooth/hci_bcsp.o
  CC [M]  /root/4.20.bluetooth/hci_ll.o
  CC [M]  /root/4.20.bluetooth/hci_ath.o
  CC [M]  /root/4.20.bluetooth/hci_h5.o
  CC [M]  /root/4.20.bluetooth/hci_intel.o
  CC [M]  /root/4.20.bluetooth/hci_bcm.o
  CC [M]  /root/4.20.bluetooth/hci_qca.o
  CC [M]  /root/4.20.bluetooth/hci_ag6xx.o
  CC [M]  /root/4.20.bluetooth/hci_mrvl.o
  LD [M]  /root/4.20.bluetooth/hci_uart.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/4.20.bluetooth/hci_uart.mod.o
  LD [M]  /root/4.20.bluetooth/hci_uart.ko
make[1]: Leaving directory '/usr/src/kernels/4.20.16-200.fc29.x86_64'
```

compress module (optional)
```
xz hci_uart.ko
```
copy or move module to drivers/bluetooth
```
cp hci_uart.ko.xz /lib/modules/$(uname -r)/kernel/drivers/bluetooth/
```
reboot...

leifliddy, 2019-04-09:
This patch was never applied to the Fedora kernel...

marbetschar, 2019-04-10:
@leifliddy thanks for the clarification! Was already wondering why everyone seems to patching instead of upstreaming to get the stuff fixed once and for all :stuck_out_tongue_winking_eye: 

In regards of Kernel patching: If I got this straight, the basic idea is to simply apply the [patch](https://github.com/christophgysin/linux/commit/b8c43916bafe6dc8de1aed5ae1cbc76cf7bad374.patch) without changing anything of the default Manjaro Kernel configuration, right?

Just messed up my Manjaro install because of a silly mistake. Will do a fresh install based upon [my previous work](https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12) with Kernel v4.20 so we can take it from there.

leifliddy, 2019-04-10:
@marbetschar 
right, so the idea is to patch the module (aka driver), compile it, install it. 
If you look at your kernel config, all of the modules being built (*=m) are going to physically on your system under /lib/modules/$(uname -r)/kernel
ie
let's say your kernel config contains this:
CONFIG_BT_HCIUART=m
This option will cause the hci_uart module to be compiled outside of the kernel.
This module will physically reside on your system at:
```
/lib/modules/$(uname -r)/kernel/drivers/bluetooth/hci_uart.ko.xz  (could also be called hci_uart.ko)
```
Since this module is not built into the kernel, we can simply use the kernel source code to re-build this module and then overwrite the original hci_uart.ko module. It's really that simple. 

However, if a module is built into the kernel 
ie
```
/boot/vmlinuz-4.20-x86_64
```
And you wanted to patch the driver, you would have to re-compile the entire kernel.
When you run:
```
lsmod
```
these are all of the modules that the system has loaded (that were built outside of the kernel)
if a module was built into the kernel, it wouldn't show up with lsmod.

leifliddy, 2019-04-10:
As to why bluetooth doesn't work on the stock Manjaro kernel, I have no idea. But, the answer is a bit more complex then it initially seems. In any case, there are Fedora users who claim the bluetooth controller doesn't always come online when the system boots. So, it's not working perfectly there either. I'm not a kernel developer, so I can only help in a limited capacity when it comes to diagnosing this issue on a lower level.

marbetschar, 2019-04-10:
@leifliddy I'm currently building the 4.20 kernel with the provided patch applied. Thank you so much for the detailed explanation of how builtin and external modules work!!

Wasn't aware of this, that's why I simply kicked of a full kernel build. However, as I didn't changed any Kernel configuration we should end up with the same result eventually.

Will report back as soon as everythings ready.

marbetschar, 2019-04-10:
@leifliddy mission accomplished :100: :v: :smile: 

The default Manjaro kernel configuration works with simply the patch applied. What's next to get this working in 4.19 or 5.x?

Here are the details:

```
$ uname -r
4.20.17-1-MANJARO

$ cat /lib/modules/$(uname -r)/build/.config | egrep 'CONFIG_SERIAL_8250_LPSS|CONFIG_BT_HCIUART_BCM|CONFIG_BT_HCIUART_SERDEV|CONFIG_SERIAL_DEV_BUS|CONFIG_SERIAL_DEV_CTRL_TTYPORT|CONFIG_SERIAL_8250_DW'
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_SERIAL_8250_DW=m
CONFIG_SERIAL_8250_LPSS=y
CONFIG_SERIAL_DEV_BUS=y
CONFIG_SERIAL_DEV_CTRL_TTYPORT=y
```

marbetschar, 2019-04-10:
@leifliddy out of curiosity:

Which type of patches get "backported" to a Linux kernel LTS version? Bug- and security fixes? I doubt "new features" would make it?

marbetschar, 2019-04-10:
@leifliddy was able to get Bluetooth running under linux 4.19.32 by applying [this patch](https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12/blob/master/patches/linux-4.19.32_drivers-bluetooth-hci_bcm.patch).

Thank you so much for the detailed walk through of how to apply a patch, recompile the module and then replace it - it really helped me a lot!!!

If you find some time I'd love to hear how you came up with the Makefile, as I do not fully understand from where you grabbed the file names for following lines and what the lines actually do:

```
obj-m 		+= hci_uart.o
hci_uart-y	:= hci_ldisc.o
hci_uart-m 	+= hci_serdev.o
hci_uart-m 	+= hci_h4.o
hci_uart-m	+= hci_bcsp.o
hci_uart-m	+= hci_ll.o
hci_uart-m	+= hci_ath.o
hci_uart-m	+= hci_h5.o
hci_uart-m	+= hci_intel.o
hci_uart-m	+= hci_bcm.o
hci_uart-m	+= hci_qca.o
hci_uart-m	+= hci_ag6xx.o
hci_uart-m	+= hci_mrvl.o
hci_uart-objs	:= $(hci_uart-y)
...
```

But this is more out of interest - no necessity at all.
Will close the issue. 🎉

closed 2019-04-10 by marbetschar

leifliddy, 2019-04-11:
For kernel 5.0, you would need to apply both of these patches
https://github.com/christophgysin/linux/commit/b8c43916bafe6dc8de1aed5ae1cbc76cf7bad374.patch
https://lkml.org/lkml/2019/3/31/344

It's probably a good idea to learn how to create a patch file
copy hci_bcm.c to hci_bcm_new.c
```
cp hci_bcm.c hci_bcm_new.c
```
manually make edits to hci_bcm_new.c

create patch
```
diff -up hci_bcm.c hci_bcm_new.c > patchfile
```
to apply patch
```
patch < patchfile
```
it should look something like this
```
--- hci_bcm.c	2019-04-11 03:17:06.039555016 +0200
+++ hci_bcm_new.c	2019-04-11 03:18:27.454567293 +0200
@@ -228,9 +228,15 @@ static int bcm_gpio_set_power(struct bcm
 	int err;
 
 	if (powered && !dev->res_enabled) {
-		err = regulator_bulk_enable(BCM_NUM_SUPPLIES, dev->supplies);
-		if (err)
+		/* Intel Macs use bcm_apple_get_resources() and don't
+		 * have regulator supplies configured.
+		 */
+		if (dev->supplies[0].supply) {
+			err = regulator_bulk_enable(BCM_NUM_SUPPLIES,
+						    dev->supplies);
+			if (err)
 			return err;
+		}
 
 		/* LPO clock needs to be 32.768 kHz */
 		err = clk_set_rate(dev->lpo_clk, 32768);
@@ -252,14 +258,16 @@ static int bcm_gpio_set_power(struct bcm
 	if (err)
 		goto err_txco_clk_disable;
 
-	err = dev->set_device_wakeup(dev, powered);
-	if (err)
-		goto err_revert_shutdown;
-
 	if (!powered && dev->res_enabled) {
 		clk_disable_unprepare(dev->txco_clk);
 		clk_disable_unprepare(dev->lpo_clk);
-		regulator_bulk_disable(BCM_NUM_SUPPLIES, dev->supplies);
+
+		/* Intel Macs use bcm_apple_get_resources() and don't
+		 * have regulator supplies configured.
+		 */
+		if (dev->supplies[0].supply)
+			regulator_bulk_disable(BCM_NUM_SUPPLIES,
+					       dev->supplies);
 	}
 
 	/* wait for device to power on and come out of reset */
@@ -269,8 +277,6 @@ static int bcm_gpio_set_power(struct bcm
 
 	return 0;
 
-err_revert_shutdown:
-	dev->set_shutdown(dev, !powered);
 err_txco_clk_disable:
 	if (powered && !dev->res_enabled)
 		clk_disable_unprepare(dev->txco_clk);
```
It looks like Chen's patch has been sent to bluetooth subsystem maintainer for approval.
https://lore.kernel.org/linux-bluetooth/20190401034312.27475-1-wens@kernel.org/

Sadly, it appears that christophgysin's patch has not been submitted. 
It look like to needs more work though, perhaps conditional logic to check whether device is an affected macbook model
I'm sure other devices benefit from the lines he removed.

leifliddy, 2019-04-11:
@marbetschar 
> Which type of patches get "backported" to a Linux kernel LTS version? Bug- and security fixes? I doubt "new features" would make it?

These patches would be considered bug fixes as they fix something that's broken. If Chen's patch gets approved, it would get back ported to when the issue first occurred, which is kernel 5.0

marbetschar, 2019-04-11:
Once again; thank you so much @leifliddy !!!

leifliddy, 2020-01-13:
@marbetschar 
Since, we had this discussion, I've developed better techniques for compiling and installing a modified ```hci_uart.ko```

For example, you don't have to overwrite the original module, you can just put an updated version of  ```hci_uart.ko``` in ```/lib/modules/$(uname -r)/updates```, and it will supersede the original module. 
I've also created an install method in the ```Makefile``` to install the module. 

I've created this project to patch and install the ```hci_uart.ko```
https://github.com/leifliddy/macbook12-bluetooth-driver