Dunedan/mbp-2016-linux #100: Not a single mainstream distro has support for keyboard/mouse out of the box?

brandonros, 2019-04-18:
I just want to make sure that Ubuntu, Debian, Fedora, Manjaro, and whatever other flavor of the year is out there all *don't* support booting one of the most popular laptops on the market without extensive tweaks/configuration.

I see the link to https://github.com/cb22/macbook12-spi-driver

Not a single distro ships with that by default?

Dunedan, 2019-04-19:
Yes, that's true, but is likely to change at least for the MacBook Pro 2016 and 2017 once the driver is upstreamed (which is currently on its way). For the MacBook Pro 2018 the situation is more dire. See #71 for details.

closed 2019-04-19 by Dunedan

brandonros, 2019-04-19:
Where can I track the driver being upstreamed progress?

Dunedan, 2019-04-19:
The linux-input mailinglist is probably the best place for that: https://www.spinics.net/lists/linux-input/msg61058.html

brandonros, 2019-04-19:
Are you able to provide the GitHub community with a high level overview in this issue with why a super popular 3 year old device has not had patches to enable basic functionality (like being able to use the operating system with keyboard/mouse/support) merged mainstream yet?

jboyens, 2019-04-19:
@brandonros Hi Brandon, thanks for joining us here. You may not realize this, but everyone here is a volunteer who is not paid to do any of the support we do for the community. We aren't Apple engineers (at least no one has said as much).

Super popular or not, no one is beholden to provide you with an "easy" install on major distros. Even when the keyboard driver gets mainlined. I'm afraid you're going to be disappointed either way due to WiFi and Audio issues that aren't likely to ever be fixed.

Daniel and others have worked hard to debug issues and make this work in a way that means I can use it on my 14,3 as a daily driver. Something I didn't think I would ever be able to do. I'm super thankful to this community and that's why I remain here to help others as I can.

Please be kind and refrain from demanding service or explanations.

brandonros, 2019-04-19:
@jboyens I hope that the tone in my issue did not come across as "you guys suck, I'm not grateful" by any means. I also hope that it did not come across as "I feel *I* am owed an explanation" or "I feel *I* am owed an easy installation"

> Super popular or not, no one is beholden to provide you with an "easy" install on major distros.

I don't know the number of weekly/monthly Linux kernel contributors, but I'm kind of surprised to hear that given the worldwide force of developers out there, nobody can spare some extra cycles to help you + Daniel get MacBooks working "better" for the community. Why should it be you two going at it alone, "failing" to deliver so to speak? Who wouldn't benefit from easier installation and better support in mainstream distros?

Why should it be up to random stragglers of the open source community to get MacBook support into Linux? Why don't mainstream distros care about supporting the MacBook?

I'm really not targeting these questions at you/Daniel or demanding answers. I'm trying to understand what part of the open source system is "broken". Obviously, the mantra "well, if you don't like it, fix it" applies. I obviously don't have 20-40 hours a week for 8+ weeks to learn the issues, submit patches, etc. Not many people do/would want to do that, and those who do are "heroes" in the open source community, obviously.

But... why is the hardware so broken/unsupported that tons of effort is required? Apple doesn't care to support Linux, I get that. But... nobody across the... I'm guessing 10k kernel developers out there care about MacBooks either?

jboyens, 2019-04-24:
> nobody can spare some extra cycles to help you + Daniel get MacBooks working "better" for the community. Why should it be you two going at it alone

Right up front, I don't want to leave the impression that I've done anything to move this project forward other than to debug / run Linux daily on a 14,3. It's kind of a "stolen valor" that I'm not worthy of. I'm just a guy who happens to have too much time on his hands.

I understand your frustration and it's valid, but I think it's aimed in the wrong direction. It should be aimed at Apple for creating bespoke hardware that doesn't add anything to the laptop, but makes it harder to use non-Apple operating systems. The T2 chips and Touchbar make things really difficult. 

The difference between reality and "what would be ideal" is that no one is really clambering to run Linux on these laptops. Most folks who get these things are getting them from work and either A) aren't allowed to try and install an "alternative operating system" or B) don't really care to. This coupled with Audio and WiFi issues and less than stellar battery life and you'd be better off using a more compatible laptop. Some contributors have already done just that.

Some of us, however, are stuck in our ways / workflows / with whatever work gives us.

There has been a lot of help from the kernel developers in getting things running and working, and many patches have been upstreamed. Those developers don't always have access to the hardware and rely on smart folks doing debugging and trying out weird things. It's up to us, because we have the hardware.

> Why should it be up to random stragglers of the open source community to get MacBook support into Linux? Why don't mainstream distros care about supporting the MacBook?

I don't think it's about not caring, I think it's about resources. Why should mainstream Linux distros care about a niche market like people who want to run Linux on Macbooks (specifically newer Macbooks)? It's a niche of a niche. If it's in the kernel or it's an easy fix, then they'll probably add the patches or hwdb updates or whatever to make it work, but they aren't likely to go buy a laptop and pay someone to go figure it all out.

> But... why is the hardware so broken/unsupported that tons of effort is required? Apple doesn't care to support Linux, I get that. But... nobody across the... I'm guessing 10k kernel/distro maintainers/developers out there care about MacBooks either?

Apple does things that almost seem aggressive to Linux. They support Windows w/ Bootcamp (even that support is faltering), but I'm afraid being able to run Linux on a Mac is likely going to be rather impossible soon. The T2 chip means that unless Apple gives us a way to unlock it, we aren't going to be booting. It's not just Apple either, Broadcom could release the firmware for the WiFi card, but, it isn't profitable for them (came directly from a Broadcom employee).

This is why Libre hardware and software is important. This vendor lock-in is the pits. I know that I won't ever buy a personal Mac again. I'm going to buy from System76 or Purism and support companies that don't try to stop me from doing things with my hardware.

I'm sorry for the rant, but I think these things are important. I'm incredibly thankful to all the folks here who know anything about how to debug hardware. It's honestly amazing.

brandonros, 2019-04-24:
I guess my biggest question is... what can you do to get the work in this repo merged upstream in some of the biggest repos so people can actually type/use their mouse when their laptop boots? What's holding you back from upstreaming the goods?

roadrunner2, 2019-04-25:
@jboyens Just want to note: testing is hugely important too, as without it the quality of the drivers/apps/fixes/hacks etc would be much lower, and as such everybody who takes the time to test things out contributes significantly IMO. So no need to denigrate your contributions.

@brandonros Upstreaming takes a lot of time and work. E.g. getting the keyboard and trackpad driver in has so far taken over 2 months of discussions, reviews, and changes (it looks like it's in decent shape now, though). And that's spare time and energy that's taken away from other things (such as debugging/fixing other problems), with limited gains from doing so (at least for me personally).

P.S. Just to be clear: I'm not complaining about the upstreaming process - the reviews and resulting changes are necessary and good. But it does require work.