Dunedan/mbp-2016-linux #101: Random crashing on Linux on the mbp 14,1

hashimda, 2019-04-20:
I have an issue when using Linux on my Mid 2017 13 inch Macbook Pro (14,1, no touchbar). No matter which distribution I try, or desktop environment, when I use Linux, after 30 minutes to an hour of normal usage, the entire desktop crashes, logs out, and I lose all my unsaved data in each session. I was using Debian Sid with Gnome and Linux 5.0.5 with the bluetooth patch last, however vanilla kernels also give thisi issue,  and when I checked the gnome logs, it said "Cannot access system logs", and on Fedora when it crashed it gave me a vague error message, something along the lines of the Gnome process being broken. Does anyone else have this issue? No matter whether I am using Ubuntu, Debian, Arch Linux, Fedora or OpenSUSE with MATE, KDE or Gnome this issue persists. I tried creating a 16GB swap for good measure to check whether this issue was caused by running out of RAM, however this doesn't work. I did set vm.swappiness=1 though, in most cases, however even with leaving it default this issue still persists. Resetting PRAM and SMC doesn't help, nor does using Refind. Does anyone else face this issue? Perhaps I should try an Arch Linux installation with the latest kernel and everything to see how that goes. How can I go resolving this issue and trying to provide adequate information to help solve this? Also the crashing occurs on both Wayland and Xorg, and seems random and unavoidable. It just randomly occurs no matter the load, I would just be working on my laptop and it crashes. Thank you.

hashimda, 2019-04-20:
By the way, this has occurred on every kernel version I have tested, though I have mostly tested the 4.18.x, 4.19.x and 5.0.x kernels.

turenar, 2019-04-24:
I have a similar issue with 4.19.27-gentoo-r1 on MacBook Pro 14,3. It seems to occasionally occur after 30min-3hrs(?) of normal usage. With older kernels (4.16.x?), I guess I didn't encounter the issue.

hashimda, 2019-04-24:
@turenar 
So what you are saying is that if I switched to a 4.16.x kernel then I wouldn't get the issue anymore? Problem solved hopefully? Stick to 4.16.x until the next major kernel update is released then try it out? Hopefully going back would solve the problem. Thank you!

roadrunner2, 2019-04-24:
Is this just the desktop crashing then? I.e. you can log back in without needing to reboot? Does the systemd journal (`sudo journalctl ...`) show anything useful?

hashimda, 2019-04-24:
@roadrunner2 
Yes it is just the desktop crashing, I can log back in without needing to reboot.
As I said, I was on Debian Sid, and when it crashed, when I checked Gnome System Logs, it said unable to read system logs, and on other distros, it gives me a vague error message (something like broken pipe on Gnome shell). I didn't use `sudo journalctl`, however, this issue does also occur on KDE and MATE, it is not exclusively a Gnome issue. I am downloading Ubuntu 18.04 LTS though and am going to try out the regular Gnome desktop with the 4.16.x kernel. Thank you.

turenar, 2019-04-24:
I had checked syslog once (after crash/force-shutdown/boot) and seen file padded with '^@' in 'less'. And, when I got crash after 'git commit' completed, HEAD commit object was corrupted. IMO, my issue should be crash in kernel, not DE.

roadrunner2, 2019-04-24:
Hmm, if git got corrupted, that could certainly indicate a kernel fs/disk-driver issue. But then dmesg should show something - is there nothing (crash related) in there?

turenar, 2019-04-24:
I suspect my data corruption was caused by crash between the time fs metadata synced and the time file contents would be synced.
btw, I have any other information, such as dmesg, syslog or even panic message, other than something crashed and files (sometimes?) corrupted.

hashimda, 2019-04-24:
@turenar @roadrunner2 
I have now installed Ubuntu 18.04.2 LTS and so far there have been no crashes, not at all, I am using the Linux 4.16.3 kernel, though I think the default 4.15 kernel should work fine (and should probably be used for security updates), running for a few hours, it seems like this issue is non existent so far.
Thanks!

roadrunner2, 2019-04-27:
@hashimda That is good to know. However, if you want to get this fixed in newer kernels, then we're going to need to find the cause. If you could boot a newer kernel and then provide the dmesg output (or least the parts when the crash occurs) after the crash, that might help noticeably (since this appears to be a kernel issue, I'm assuming there should be something relevant in dmesg).

hashimda, 2019-04-27:
Yes @roadrunner2, I just installed the latest kernel (5.0.9), and I am waiting for it to crash. When it crashes, I will provide dmesg output and whatever I can that may be relevant to the issue. Thank you for your help.

Dunedan, 2019-07-25:
@hashimda @turenar: Any updates regarding this issue?

hashimda, 2019-08-12:
@Dunedan I haven't been trying to run Linux on my MBP lately, but I'm fairly sure that it wouldn't be crashing with the newest kernel patches and things, I may give trying to run Linux another go soon, but have been busy and have been sticking to macOS for now. Thanks!

Dunedan, 2019-08-23:
I'm closing this for now, as this doesn't seem to be an issue anymore. If it reappears again and is linked to the hardware, we can reopen it again of course.

closed 2019-08-23 by Dunedan