Dunedan/mbp-2016-linux #102: Sound on MacBook 10,1 Retina (12-inch)

marbetschar, 2019-04-27:
Audio basically works, but the speakers don‘t.

This issue originated from https://github.com/Dunedan/mbp-2016-linux/issues/17#issuecomment-487260247

Dunedan, 2019-05-11:
When you say "audio basically works", what do you mean by that? Does the headphone jack work?

likefiftyninjas, 2019-05-11:
No, as of right now, no audio works on a MacBook. Jack nor speakers.

leifliddy, 2019-05-19:
The headphone jack works on the Macbook8,1, 9,1, and 10,1 models. The speakers are the issue.

Dunedan, 2019-05-29:
Somebody recently opened a bugzilla issue regarding this topic for the MacBook9,1: https://bugzilla.kernel.org/show_bug.cgi?id=203613

leifliddy, 2019-08-05:
@dvulricht has managed to get the speakers working on my macbook 9,1!!!!
I'll post updates on this in the coming days.

leifliddy, 2019-08-15:
I created a github project for the macbook12 audio driver (my first github project)
https://github.com/leifliddy/macbook12-audio-driver

It's definitely a bit rough around the edges and a bit hacky, but it works. 
Please test out this driver and give me your feedback.

Dunedan, 2019-08-20:
@leifliddy Congrats to getting it to work, but I believe you closed the bug in the kernel Bugzilla too early, as this isn't fixed in the kernel yet, isn't it? It'd be great if code to make the speakers work would make its way into the mainline kernel sooner or later as well.

leifliddy, 2019-08-20:
I had created a bounty for issue:
https://www.bountysource.com/issues/56233031-macbook8-1-12-inch-early-2015-no-speaker-sound-output
I had to close the bug report in order to award the bounty to @dvulricht for his work in developing the patch. I can reopen the bug report once the bounty is paid out.

leifliddy, 2019-08-28:
I reopened the bug report. But, due to the series of vendor node commands that need to be 'replayed' in order for the speakers to work, I don't think it's likely that this driver will ever make its way into the kernel. But who knows, stranger things have happened. In any case, we still need to get the HP audio working with this driver.

xaiki, 2021-01-02:
i can confirm that HP work on a Macbook10,1 with https://github.com/leifliddy/macbook12-audio-driver,
that said:
 - input stops working (expected ?)
 - hearphone jack doesn't work (expected)
 - the machine seems to come back even less from suspend with that module loaded.

gc070653, 2024-01-01:
I'm trying to install your driver on a macbook9.1 2016 with Manjaro on it with kernel 5.15.144-1-Manjaro but the sound is not exiting from the internal speakers. Can you confirm that the speakers are working normally, so I'll investigate on what is wrong with my installation? Earphones were working on the initial installation . I already tried with other distros, Ubuntu Debian Fedora but no success.  
Thank you

nnylip, 2024-03-09:
The speaker output is working on a macbook 12inch 2017  using .../leifliddy/macbook12-audio-driver  (thank you!). However, microphone is not work. Does anyone know a way to get this to work? Or where would one start to do this (I am willing to try even though its been awhile for device drivers)?

gc070653, 2024-03-10:
I solve the problem using the earphone with microphone: I reboot with the original drivers that allow the usage of earphoneLuca

nnylip, 2024-03-10:
I tried using a jabra 65t and it wans't recognized. I have a Harmon Kardon speaker with mike that is 
recognized. What product do you use? thanks

gc070653, 2024-03-11:
I'm using a simple and old NOKIA earphone with mike
It is only for testing, not using it normally
But did you install the bluetooth driver from https://github.com/leifliddy/macbook12-bluetooth-driver
Ciao
Luca

nnylip, 2024-03-11:
Thanks for info. I got a tiny usb microphone that works. I did install the bluetooth driver you 
mention, it works quite nicely. The jabra 65t, though recognized, must require the a cs408 driver as 
it appears in the speaker icon in the system try but does not get sound. Another headset (motorola) 
does work.
It just seem more elegant if the internal mike would work. I am thinking of trying to adapt 
microphone driver in https://github.com/davidjo/snd_hda_macbookpro for the macbook 10,1. Not a 
strong C developer but could always learn.

gc070653, 2024-03-11:
Keep me updated on the internal mike because I'm interested 
It would be the best solution.
Did you try to use the facetimehd camera? On my mac it is working

nnylip, 2024-03-11:
It is working, used the tips in this page:

https://github.com/patjak/facetimehd/issues/196

gc070653, 2024-03-12:
It is the same tip I followed

https://github.com/patjak/facetimehd/issues/196

nnylip, 2024-03-12:
Do you have to reinstall drivers when a big upgrade arrives? yesterday, apt upgrade, caused the need 
to reinstall the facetimehd and bluetooth drivers. Interestingly, the sound driver was unaffected.

gc070653, 2024-03-13:
I'm running LMDE 6, installed few days ago, and so far everything is running properly.
If the update requires a new kernel probably you need to reinstall the drivers because it changes the name of some directories.

6isito-spec, 2026-02-15:
Does this have a solution (today 15th Feb 26)?
I've tried everything I've found online including the suggestions from this post, but the audio is still not working.

I’m dual-booting Ubuntu 24.04.4 LTS on MacBook10,1, audio is not working and I’ve already digged all the possible solutions from online resources, but no luck so far.

What I’ve Tried:
Using Alsamixer, I’ve made sure all sound level are >0 and auto-mute is disabled

I’ve installed Pulseaudio and made sure Profile is Analog Stereo Output and Output device is set to Speakers.

Here’s the

“**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CS4208 Analog [CS4208 Analog]
Subdevices: 0/1
Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
Subdevices: 1/1
Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
Subdevices: 1/1
Subdevice #0: subdevice #0”

also

“00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
Subsystem: Intel Corporation Sunrise Point-LP HD Audio
Flags: bus master, fast devsel, latency 64, IRQ 39, IOMMU group 9
Memory at 92820000 (64-bit, non-prefetchable) [size=16K]
Memory at 4000000000 (64-bit, non-prefetchable) [size=64K]
Capabilities:
Kernel driver in use: snd_hda_intel
Kernel modules: snd_soc_avs, snd_sof_pci_intel_skl, snd_hda_intel”

I’ve really tried many things and I’m not able to list them all here, but with no success

Any help will be greatly appreciated and I hope a solution is possible as I love this combo Linux on Mac

help help help 😀