Dunedan/mbp-2016-linux #105: Is there an alternative repository for the iMac?

CuriousTommy, 2019-07-05:
I apologize if I am intruding here, but I was wondering if anyone here knows if there is a repository that is structured similar to this repo but for the iMac.

I would love to report and discuss some of the issues that occur when I run Fedora 30 on my 2019 iMac.

If this discussion is not okay, you may close this issue.

Dunedan, 2019-07-25:
I'm not aware of a similar repository for other Apple hardware. Feel free to open issues here for iMacs as well. While I'm not sure if it makes sense to include the status of hardware support for iMacs in the README, discussing iMac related issues here makes perfect sense, as most problematic hardware is shared with the MacBook Pro's anyway.

CuriousTommy, 2019-07-29:
@Dunedan Awesome! If I have time in the future, I will make a new report about my parent's 2019 iMac.

closed 2019-07-29 by CuriousTommy