Dunedan/mbp-2016-linux #106: MBP 13.3 2019

GoncaloHit, 2019-07-16:
I just wanted to know if anyone got it working, i have a MBP 13.3 i7 with touch bar, i tried the keyboard & touchpad but without success, wanted to know if anyone got something working ?
thanks

roadrunner2, 2019-07-19:
Yes, I have FC30 running on MBP13,3. I presume you built and loaded the keyboard, touchpad, and touchbar drivers, as described in https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7?

Dunedan, 2019-07-20:
@GoncaloHit As you mentioned that your MacBook Pro is from 2019, the "13.3" you mentioned is the display size, right?
A MacBookPro13,3 would be a 15" model from 2016. A MacBook Pro with 13" from 2019 would either be an MacBookPro15,2 or an MacBookPro15,4 (check the link for figuring out which model you're using at the top of the README). For these 2018 and newer MacBook Pro's support is still very experimental. There is some encouraging development going on in #71, but the instructions linked by @roadrunner2 won't work for these newer models.

GoncaloHit, 2019-07-20:
@Dunedan Thanks il will follow up that issue, and yes its the new 13.3 macbook pro with the 8thgen i7, i had arch running on macbook air 2017, my company upgraded me to this, so yes il wait and see :)

Dunedan, 2019-07-25:
Great. I'm closing this one now, as there is the already mentioned #71. Also feel free to open additional issues for specific problems.

closed 2019-07-25 by Dunedan